zs6buj/AntTracker #8: Naze32

SwiftFlying, 2018-08-17:
HI,
(I'll order a Naze32 to take a look at. Problem is that our postal service is super slow. Maybe I can find it on Amazon. Their shipping is way faster)
I can send you NAZE32, and if you like, I can redesign a PCB for you and make a prototype so you can have more time to debug the software.

zs6buj, 2018-08-17:
Hi SwiftFlying

Thank you for the offer of the Naze32, but it comes through the same dysfunctional postal service. :(

The PCB idea sounds interesting. I'm assuming you will integrate the naze. I did some research and discovered there are various naze boards, evolving from stm32f1 to stmf4 I think. The oldest/cheapest would probably do provided that they are still available like the eBay one you linked to. I tried to buy one of those but they don't ship here.

Regards
Eric
zs6buj

zs6buj, 2018-08-17:
Ok, I just bought one of these from Amazon. Amazon Shipping items usually take about a week, 'though last time they sent the totally wrong thing from their distribution centre in Seattle.

[https://www.amazon.com/gp/product/B078WRVWPN/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1](url)

SwiftFlying, 2018-08-17:
I am very familiar with all NAZE32 versions, because I have experience in designing flight control boards. For example, F1 F3 F4 F7, I can design.Can you send me your contact? We can talk about it.

zs6buj, 2018-08-19:
Hey SF

I'm wondering what you would put on the board? The entire "FC" and PSU, or perhaps a bespoke tracker board STM32, compass and PSU? 

You can get me at eric at zs6buj dot com, or eric stockenstrom on whatsapp. What is your time zone?

regards
Eric

SwiftFlying, 2018-08-19:
Hi zs6buj
My WhatsApp account is +86 18818594513.I work in Shenzhen, China.

zs6buj, 2018-08-19:
My WhatsApp number is +27 72 566 3325.

Sounds like you could create a popular product. :)

Are you familiar with EZ-Wifibroadcast?

SwiftFlying, 2018-08-20:
I have already added your number.

SwiftFlying, 2018-08-21:
That’s great, thank you!

| |
xiaohong dong
邮箱：swiftflying@163.com
|

Signature is customized by Netease Mail Master

On 08/21/2018 16:22, Eric Stockenstrom wrote:

Hello my friend SwiftFlying

Sorry I missed you on WhatsApp.

Remarkably, the Naze32 that I ordered on Amazon has arrived after 4 days! Now that is good service. I'll play around with it today.

closed 2018-11-07 by zs6buj