zs6buj/AntTracker #10: telemetry USB, request

marcogarzotti, 2019-01-30:
Hi, it's me again.
I hope not to have missed another software option, but would it be feasible to use the micro usb connection on the bluepill to receive telemetry data? Most of the 433/915mHz telemetry kits i see online have one module with plain serial output and the other one with usb output. Having USB support would make the them compatible with this antenna tracker and this could also simplify the wiring.

zs6buj, 2019-01-30:
Hi Marco
I see what you are getting at. As far as I know the usb port on the bluepill can only be configured for bootloading. But I'll look a little closer when I get a chance. There is also the Maple Mini, which is a little easier to use.

marcogarzotti, 2019-01-30:
My blue-pill understanding so far is that there's only USB issue: the resistor (D+ wire) that usually is not the right one, and this could block the USB comunication. Some PC can manage the mismatching value some can't, but in any case this issue can be solved with a soldering iron.  
Beside that the blue pill can be flashed with the maple mini bootloader, for instance, and then the usb can be used at will, also as a virtual COM. What i don't know is how the COM on the telemetry module is supposed to work with the virtual COM on the blue pill. 
[Here ](https://www.stm32duino.com/viewtopic.php?t=3240) a reputable STM32 user wrote "STLink V2.0 is only for uploading and/or debugging, and not for serial com!" and "Connect a USB cable from PC to the micro USB slot of the board and you will have USB serial." therefore there's some hope for my request.
I have a bluepill and a USB telemetry module (I'm lucky, aren't I? :) so i can do some tests if this is helpful!

zs6buj, 2019-01-30:
ok, I got it working with the maple DFU interface, but like you point out, only some boards work because they shipped with the wrong pull-up resistor. No, that's not a big issue.

So USB is serial (0), which we currently use to flash, and for debugging. We could use serial 1 for debugging, with a TTL/USB converter, or alternatively a small OLED screen (which I was just discussing with one of the other guys).

So, yes, I guess it could be done. I could switch the serial ports around pretty easily.

zs6buj, 2019-01-30:
Ok, done.

You can try it if you like. Remember, it is as yet untested.

AntTrackMavlinkUSBv0.36 on my GitHub.

closed 2019-12-10 by zs6buj