zs6buj/AntTracker #11: Cannot set home annttrackwifi

Ediher0504, 2019-02-02:
I tried to build the AntiWifi and LED status according to the instructions, except when I pulled up the GPio5, the LED kept blinking fast, does that mean I haven't made a home determination, is there a wrong step I made?

zs6buj, 2019-02-02:
Hi Ediher0504
Ok,  pin 5 is pulled high in the Setup function. The push button grounds it momentarily.  The LED blinks slowly when it gets good Mavlink telemetry, and fat when it get a good GPS lock. Then, and only then, if the button is pushed (pin 5 grounded, the LED goes on solid and home location is remembered.

Ediher0504, 2019-02-03:
Hi, thank you for your quick reply, cool it works well, there are more that I will ask, this is out of the topic I made, what kind of servo should I use for azimuth, 90 °, 180 ° or 360 °, so too for elevators, I'm afraid to make the mistake of choosing the type of servo, so that it provides an inaccurate range of motion.
Sorry, my English is a little messed up, I only rely on google translate

zs6buj, 2019-02-03:
Your English is fine, even if it is with Google. :)

I suggest you get two 180 degree servos. They will cover 360 degrees because the  tracker flips over and behind when it needs to.

Ediher0504, 2019-02-03:
Alright, thank you for all your information and direction, I really like your project, develop it

Ediher0504, 2019-02-03:
- This code makes me hesitate to use a servo 180 °, sorry I am not too familiar with sketch 
Link picture https://prnt.sc/mg0db7

zs6buj, 2019-02-03:
Hi Ediher0504

Yes, you are correct. Well spotted! That is one of the very old versions of Servos.ini, before I implemented improvements to flip the antenna over and behind. It will still work, but only uses 90 degrees of the 180 degrees range of the elevation servo.

I have updated the wifi version for you to v0.21 with the latest Servo.ini, which does the flip.

Ediher0504, 2019-02-03:
Thank you for doing that, gerenerik.bin version, this means uploading flash mode, this will be easier.
Btw telemetry can only be connected with antrackwifi, it cannot be connected at once with a mission planner

closed 2019-12-10 by zs6buj