zs6buj/AntTracker #12: General questions and version discrepancies

skaman82, 2019-02-21:
Hi, I am very interested in building this tracker (LTM and Mavlink-BT options). But first I have noticed some version discrepancies. The LTM version is v0.22 and mavlink is 0.37. Why they are not at the same stage? And do I need to decide witch protocol I have to use or it determines the protocol automatically? That would be great because I probably use Mavlink AND LTM depending on Aircraft. 

Also I've seen on RC groups some code supporting an OLED I2C display. Is this already part of your code - this would be great if we could have some configuration options (like protocol selection) and/or for some status data.

zs6buj, 2019-02-22:
Hi skaman82

The LTM and Mavlink versions diverged when Mavlink specific changes were made, for example to support Mavlink 2 using the Ardupilot libraries. The essential functionality is the same. 

Right now I don't detect protocol automatically, but I can see your point that you might need such a feature. Maybe I'll do that when I get some time (in short supply right now). :)

The same goes for OLED I2C. I have some immature test code up and running, but need to tidy it up before publishing. My OLED display is the popular 0.96 inch two colour one, but I find it a bit small, so I was looking at slightly bigger screens, but other things intervened. Also one of the guys has produced a version/fork supporting OLED if you can find it.

skaman82, 2019-02-22:
@zs6buj Thanks for you quick answers. I will give it a go and build this. Really looking forward to multi-protocol version! :D

closed 2019-12-10 by zs6buj