zs6buj/AntTracker #14: Arduino Uno/Mega and Bluetooth

arjrj16, 2019-07-14:
Hi,
Would this work with the Arduino Uno/Mega 2560? Also, I saw the code written for bluetooth - does it work with the hc05? Also, is it possible to bind the hc05 with the tbs crossfire since the crossfire requires you to accept a passkey? From my understanding, LCDs are not yet supported?
Thanks in advance,
Arjun M

arjrj16, 2019-07-14:
Also, would I need to comment out Blue Pill or Maple Mini for

arjrj16, 2019-07-14:
Sorry didn't mean to submit that or close it - but what I was wondering if I need to comment out Blue Pill or Maple Mini to make the teensy work? I found the rc groups forum which helped a bunch.

zs6buj, 2019-07-15:
No problem at all. Hope you answered all your questions.

arjrj16, 2019-07-15:
Hi,
I am still a little bit confused - I would like to build it with the teensy and I still cannot figure out if I need to comment out maple mini or the blue pill on the code? Also, is there any way to increase the refresh rate for the antenna tracker, I read on the forum it refreshes every 2-3 seconds? Lastly, I understand you used the teensy because of its multiple serial debugging ports, but would it still work with an Arduino mega 2560? If I use it, would it have a higher processing speed, allowing me to decrease the refresh rate?
Thanks in advance,
Arjun M

zs6buj, 2019-07-16:
Hi Arjun,

Yes you must comment out the blue pill and maple mini declarations.

The tracker is faster than the telemetry rate. Increase the telemetry rate in Mission Planner and location will update more often. 2 to 3 seconds is not correct.

No, the 2560 is much slower than a Teensy 3.2 or STM32F103C

arjrj16, 2019-07-18:
Hi,
My parts are slowly coming in, but my teensy has already arrived. When I tried to upload it onto the teensy I received the following error message:
Arduino: 1.8.8 (Windows 10), TD: 1.46, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz (overclock), Faster, US English"

C:\Users\arjun\Desktop\Ghetto Station\Other Antenna Tracker\AntTracker-master\AntTrack_v2.06\AntTrack_v2.06.ino:163:36: fatal error: Adafruit_SSD1306_STM32.h: No such file or directory

Multiple libraries were found for "Servo.h"
compilation terminated.

 Used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Servo
 Not used: C:\Program Files (x86)\Arduino\libraries\Servo
Error compiling for board Teensy 3.2 / 3.1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Do you know how I would fix this?

arjrj16, 2019-07-18:
Sorry, I think I missed some steps earlier. But I am still getting an error message:
Arduino: 1.8.8 (Windows 10), TD: 1.46, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz (overclock), Faster, US English"

In file included from C:\Users\arjun\Documents\Arduino\AntTracker-master\AntTrack_v2.06\AntTrack_v2.06.ino:158:0:

Multiple libraries were found for "Servo.h"
C:\Users\arjun\Documents\Arduino\libraries\Servo\src/Servo.h:75:2: error: #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."

 Used: C:\Users\arjun\Documents\Arduino\libraries\Servo
 #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."

 Not used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Servo
  ^

 Not used: C:\Program Files (x86)\Arduino\libraries\Servo
C:\Users\arjun\Documents\Arduino\AntTracker-master\AntTrack_v2.06\AntTrack_v2.06.ino:163:36: fatal error: Adafruit_SSD1306_STM32.h: No such file or directory

compilation terminated.

Error compiling for board Teensy 3.2 / 3.1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Thankyou so much,
Arjun M

zs6buj, 2019-07-18:
arjrj16 

The Teensy 3.x is not supported right now.

You can ignore the warnings. There are different libraries for different boards, and you have them all on your library path. It will choose the correct one for the Teensy.

The error causing your compile failure is it can't find the Adafruit_SSD1306_STM32.h in your library, but you have select board Teensy 3x

arjrj16, 2019-07-18:
How would I fix this? Is there a way I can use the Adafruit_SSD1306_STM32.h with the teensy? I understand that people have gotten it to work with teensy 3.2, but how did they do this?
Thank you so much for your help... I am very new to this so any help is very much appreciated.

arjrj16, 2019-07-18:
In my errors, it also seems the servo.h library isn't working? (See below)
Thankyou so much.
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\arjun\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\arjun\Documents\Arduino\libraries -fqbn=teensy:avr:teensy31:usb=serial,speed=96,opt=o2std,keys=en-us -ide-version=10808 -build-path C:\Users\arjun\AppData\Local\Temp\arduino_build_44539 -warnings=none -build-cache C:\Users\arjun\AppData\Local\Temp\arduino_cache_171069 -verbose C:\Users\arjun\Documents\Arduino\AntTracker-master\AntTrack_v2.06\AntTrack_v2.06.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\arjun\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\arjun\Documents\Arduino\libraries -fqbn=teensy:avr:teensy31:usb=serial,speed=96,opt=o2std,keys=en-us -ide-version=10808 -build-path C:\Users\arjun\AppData\Local\Temp\arduino_build_44539 -warnings=none -build-cache C:\Users\arjun\AppData\Local\Temp\arduino_cache_171069 -verbose C:\Users\arjun\Documents\Arduino\AntTracker-master\AntTrack_v2.06\AntTrack_v2.06.ino
Using board 'teensy31' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr
Using core 'teensy3' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=146 -DARDUINO=10808 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy3" "C:\\Users\\arjun\\AppData\\Local\\Temp\\arduino_build_44539\\sketch\\AntTrack_v2.06.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=146 -DARDUINO=10808 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy3" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\EEPROM" "C:\\Users\\arjun\\AppData\\Local\\Temp\\arduino_build_44539\\sketch\\AntTrack_v2.06.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=146 -DARDUINO=10808 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy3" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\EEPROM" "-IC:\\Users\\arjun\\Documents\\Arduino\\libraries\\Servo\\src" "C:\\Users\\arjun\\AppData\\Local\\Temp\\arduino_build_44539\\sketch\\AntTrack_v2.06.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=146 -DARDUINO=10808 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy3" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\EEPROM" "-IC:\\Users\\arjun\\Documents\\Arduino\\libraries\\Servo\\src" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\SPI" "C:\\Users\\arjun\\AppData\\Local\\Temp\\arduino_build_44539\\sketch\\AntTrack_v2.06.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=146 -DARDUINO=10808 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy3" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\EEPROM" "-IC:\\Users\\arjun\\Documents\\Arduino\\libraries\\Servo\\src" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\SPI" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\Wire" "C:\\Users\\arjun\\AppData\\Local\\Temp\\arduino_build_44539\\sketch\\AntTrack_v2.06.ino.cpp" -o nul
Multiple libraries were found for "Servo.h"
[... cut ...]
compilation terminated.

Using library EEPROM at version 2.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\EEPROM 
Using library Servo at version 1.1.4 in folder: C:\Users\arjun\Documents\Arduino\libraries\Servo 
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI 
Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire 
Error compiling for board Teensy 3.2 / 3.1.

closed 2019-12-10 by zs6buj