zs6buj/AntTracker #15: Waiting for GPS lock

arjrj16, 2019-07-21:
Hi,
I have successfully compiled the code onto the stm32 board. Everything seems to be working - the OLED says:
AntTracker by zs6buj
Setting up Bluetooth
Hdg source = craft's GPS
Found Telem 57600b/s
MavLink 1 found
Waiting for GPS lock
MavLink telemetry good!
 The only thing is - it is stuck at the Waiting for GPS lock, even when I have 11 satellites. But, when I power off my drone, it continues saying walk out 10 meters then push the button(this still doesn't work.)
I have tried to debug it by connecting my USB to TTL's Rx and tx pins to the board, but when I open the serial port(after uncommenting //debug all) nothing is there. I have it set to new line, 9600 baud. Do you know how I could fix this? Thank you so much for all the help, I really appreciate it.
Thanks,
Arjun M

zs6buj, 2019-07-21:
Hi arjrj16

Well, it looks like you are very close!

The fact that it said " Mavlink telemetry is good" means you are nearly there. The if later you have un-commented Debug_All and you get no serial monitor messages, means you have gone backwards. Did you change something in-between?

I noticed the "Setting up Bluetooth". So, do you have a BT adapter in the input. I seem to recall you mentioning that before. Before you try to debug the BT side, get the tracker working with direct serial telemetry at 57600 b/s.

arjrj16, 2019-07-23:
Hi,
Thanks for getting back to me.
I recently haven't been able to get my oled to work. It only works when I
supply the stm32 with 3.3v from an arduino or my usb to ttl's 3.3v output.
When I try to use it with the linear voltage regulator the oled doesn't
work. I used my meter and from the linear voltage regulator I get 3.307v vs
the usb to ttl i get 3.395v or the arduino gives 3.319v. Is there something
wrong with my board or is this common? I still don't know if the tracker
works. Before the oled stopped working, I was able to get the debug to
work... i forgot to connect the ttl's power ports to the stm32, but it
worked after. Only shortly later I burnt out my oled after my servos went
crazy(I dont know why) but when I received the new oled display, it only
works the stm32 receiving 3.3v.
Do you have a number I can possibly Skype you with? I really appreciate all
the help.
Thanks,
Arjun M

arjrj16, 2019-07-24:
Hi Eric,
So I have tried completely switching out the stm32 board, and have
reinstalled the program from online, only changing the //debug all to debug
all. I have wired up everything, but unfortunately, the Oled still does not
work. I think it is the software as I tried uploading an adafruit OLED test
program, and it worked perfectly, so I doubt the hardware could be the
issue. Are there any restraints set in the program before allowing the OLED
to work? Thanks for your help.
Arjun M

arjrj16, 2019-07-24:
[image]
Here is a picture of my wirng

zs6buj, 2019-07-25:
Hi Arjun

Yet you had the OLED working before, I assume with the same firmware and libraries?
You posted " the OLED says:
AntTracker by zs6buj
Setting up Bluetooth
Hdg source = craft's GPS
Found Telem 57600b/s
MavLink 1 found
Waiting for GPS lock
MavLink telemetry good!"

When you start up now, does the serial monitor work, and display the same things as above? If not, disconnect the OLED, and try again to make sure the board and firmware work, then try again with the OLED.

Disconnect the servos  for now.

closed 2019-12-10 by zs6buj