zs6buj/AntTracker #16: Frsky r9m telemetry questions

Fpvmatt, 2019-10-16:
Hi I have been searching for an antenna tracker I could use with Inav and the frsky r9m system. I have gps telemetry coming into my radio from my FC. Can I hook up this antenna tracker to the smart port at the back of the module with wires or a bluetooth link? I'm hoping I'm correct with my assumption that this is how it works.

Thanks in advance.

macfly1202, 2019-10-16:
I could say without any hesitation it’s perfectly working and many way to achieve what you want. zs6buj have a converter from
Sbus to mavlink if necessary. 
Go the the right documentation and read carefully the first Post : https://www.rcgroups.com/forums/showthread.php?2942118-Build-a-Reliable-and-Cheap-Antenna-Tracker-this-Weekend-%C2%96-by-zs6buj

Regards.

Fpvmatt, 2019-10-16:
Thanks what I would like to do is feed the smart port from the back of the transmitting module into a hc05 or 6 and have another hc05 or 6 output the smart port telemetry into the Arduino controlling the antenna tracker.  After re reading multiple times I have yet to wrap my head around if this is possible or how this works.

Im well versed with bluetooth modules and arduino i'm just not very sure what all is involved with frsky's smart port.

Thanks again

closed 2019-12-10 by zs6buj