zs6buj/AntTracker #24: WIFI CONNECT ISSUE

lipk888, 2020-09-25:
HI zs6buj:
first many thanks for create this nice project and great work.
i try to use ESP32 with the main board and try to use my cross fire mirco V2'S wifi connect to tracker and i try to use STA mode and UDP port to connect.
i set remortIP address to 192.168.4.101 and port is 14550. which show below and my crossfire mavlink output UDP port also is 14550. so i think this should be work but i failed. 
the OLED show connect correctly, but no more information show and no mavlink data feedback, i checked AntTrack_v2.14.ino file and it show if right connect OLED will show "client connected" so i think my connect is failed.
actully i just have poor knowledge about code, so i'm not sure if there have some mistake for my setting.
so my question is how to set local IP/PORT and remort IP/Port for STA or AP mode?

below is my information:
crossfire wifi AP IP: 192.168.4.1 
crossire wifi mavlink UDP port:14550

[image]
[image]

zs6buj, 2020-09-26:
Hi lipk888

From what I can see your settings should work. So we expect your crossfire access point to send udp packets to 192.168.4.101 port 14550. Maybe try to connect on your phone or tablet. Download and install UDP Monitor, and try to connect and see what you get.

lipk888, 2020-09-28:
Hi Zs6buj:
thanks for feedback, 
yes i try to found mavlink output from my crossfire TX V2.
seem no mavlink massage out put, no matter i use PC Qgroundcontrol or use UDP monitor.
so for now i just feedback this issue for TBS team hope they have feedback.

kamik2, 2020-09-28:
there are 2 thing you need to do:
1. (also 1. bug) you need to enable the mavlink option in the crossfire wifi/ udp settings every time you power up the transmitter 
2. (bug 2) atm the wifi fw needs to be connected to a acces point in order to send the data

both issues are adressed on the crossfire side

lipk888, 2020-09-28:
> there are 3 thing you need to do:
> 
> 1. (also 1. bug) you need to enable the mavlink option in the crossfire wifi/ udp settings every time you power up the transmitter
> 2. (bug 2) atm the wifi fw needs to be connected to a acces point in order to send the data
> 3. the standard port in the crossfire is 8888. you can change it to any other port you like in the wifi/ udp setings as well
> 
> both issues are adressed on the crossfire side

hi Kamik2:
thanks for you feedback.
1. yes i checked this issue on crossfire manual it notice known bug for your mention
2. you mean ATT(ESP32) need set to AP mode, then crossfire connect to ESP32 AP? 
3. I try  change crossfire port to 14550, then ATT with STA mode connect, still not work

kamik2, 2020-09-28:
i meant that you nee to connec the crossfire to a AP and the trcker as well (not sure if the last one is possible?) - otherwise the crossfire does not start to send mav packages.

lipk888, 2020-09-28:
> i meant that you nee to connec the crossfire to a AP and the trcker as well (not sure if the last one is possible?) - otherwise the crossfire does not start to send mav packages.

hi Kamik2:
ok, i will try later, so seems i need 3th wifi AP, both crossfire and ATT link to this 3th AP right?

kamik2, 2020-09-28:
correct

lipk888, 2020-09-28:
hi Kamik2:
i try for below step:
1. use iphone build AP point.
2. ATT wifi and crossfire wifi link to iphone's AP point.
3.open crossfire mavlink output switch and set port to 8888

but as you can see below i steel can not get mavlink massage.
[image]
[image]
[image]

lipk888, 2020-09-30:
hi team:
finally i set mavlink output from crossfire TX V2, seems crossfire have some bugs as kamik2 said. crossfire need link to 3th AP then it can send out mavlink massage and then ATT link to crossfire's AP

now i face other issue, ATT feedback heartbeat good and i checked my X9D screen and INVA both show 3D fixed.
but i wait 5 mins LED still with slowly flash.OLED screen no massage feedback GPS lock  as ATT notice say slowly LED means mavlink ok, fast LED means GPS lock.
so i'm not sure what i mistake or what i missing?

[image]
[image]
[image]

zs6buj, 2020-09-30:
You could try 

`#define Debug_Mav_GPS   `

in config.h

This should show you GPS information coming in from Mavlink

lipk888, 2020-10-01:
> You could try
> 
> `#define Debug_Mav_GPS `
> 
> in config.h
> 
> This should show you GPS information coming in from Mavlink

hi Zs6buj:
thanks for quickly reply.
i try to follow beow set, but still no massage back. and no GPS massage back is any mistake i do 
[image]

zs6buj, 2020-10-01:
lipk888

What is coming out the serial monitor?

[image]

lipk888, 2020-10-01:
hi zs6buj:
i get log below, it seems no GPS massage feedback
16:14:51.095 -> Starting up......
16:14:51.162 -> Target Board is ESP32 Dev Module
16:14:51.195 -> Mavlink WiFi In
16:14:51.229 -> Az = 90.00	 El = 0.00	 hmHdg = 90.00
16:14:51.229 -> pntAz = 90 El = 0.00 LastGoodpntAz = 90 LastGoodEl = 0 hmHdg = 90.00 azPWM = 1500 elPWM = 2300
16:14:51.229 -> Heading_Source = Tracker's Own Compass
16:14:51.263 -> ----- Compass Found -----
16:14:51.263 -> Sensor:       HMC5883
16:14:51.263 -> Driver Ver:   1
16:14:51.263 -> Unique ID:    12345
16:14:51.263 -> Max Value:    800.00 uT
16:14:51.263 -> Min Value:    -800.00 uT
16:14:51.263 -> Resolution:   0.20 uT
16:14:51.297 -> --------------------------
16:14:51.297 -> 
16:14:51.297 -> Display tracker heading for 20 seconds
16:14:51.297 -> Heading = 276
16:14:52.279 -> Heading = 276
16:14:53.293 -> Heading = 276
16:14:54.279 -> Heading = 275
16:14:55.266 -> Heading = 276
16:14:56.280 -> Heading = 276
16:14:57.295 -> Heading = 276
16:14:58.270 -> Heading = 276
16:14:59.281 -> Heading = 276
16:15:00.290 -> Heading = 276
16:15:01.269 -> Heading = 276
16:15:02.279 -> Heading = 276
16:15:03.294 -> Heading = 276
16:15:04.272 -> Heading = 276
16:15:05.284 -> Heading = 275
16:15:06.302 -> Heading = 276
16:15:07.309 -> Heading = 276
16:15:08.289 -> Heading = 275
16:15:09.303 -> Heading = 276
16:15:10.283 -> Heading = 276
16:15:11.292 -> Trying to connect to tbs_crossfire_D8BFC0FE921E.
16:15:12.908 -> WiFi connected!
16:15:12.908 -> Local IP address: 192.168.4.100
16:15:12.908 -> WiFi RSSI:-51 dBm
16:15:12.943 -> UDP started, listening on IP 192.168.4.100, UDP port 8800
16:15:15.032 -> Client connected: Remote UDP IP: 192.168.4.1  Remote  UDP port: 8800
16:15:15.336 ->  UDP WiFi record read:
16:15:15.336 -> 04 FE 09 3E 01 FA 00 05 00 00 00 01 00 51 03 03 DE 04 
16:15:15.370 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  

ap_mavlink_version=3
16:15:15.370 ->  hb_count=1
16:15:15.370 ->  UDP WiFi record read:
16:15:15.370 -> EC FE 09 58 01 FA 00 05 00 00 00 01 00 51 03 03 63 EC 
16:15:15.370 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  

ap_mavlink_version=3
16:15:15.370 ->  hb_count=2
16:15:15.606 ->  UDP WiFi record read:
16:15:15.606 -> 5B FE 09 72 01 FA 00 05 00 00 00 01 00 51 03 03 6B 5B 
16:15:15.606 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  

ap_mavlink_version=3
16:15:15.641 ->  hb_count=3
16:15:15.641 -> hbGood=true
16:15:16.010 ->  UDP WiFi record read:
16:15:16.044 -> 47 FE 09 7E 01 FA 00 05 00 00 00 01 00 51 03 03 B1 47 
16:15:16.044 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  

ap_mavlink_version=3
16:15:16.753 ->  UDP WiFi record read:
16:15:16.753 -> 7C FE 09 8B 01 FA 00 05 00 00 00 01 00 51 03 03 3B 7C 
16:15:16.753 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  

ap_mavlink_version=3
16:15:18.475 ->  UDP WiFi record read:
16:15:18.475 -> 82 FE 09 BE 01 FA 00 05 00 00 00 01 00 51 03 03 00 82 
16:15:18.475 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  

ap_mavlink_version=3
16:15:19.725 ->  UDP WiFi record read:
16:15:19.725 -> F5 FE 09 D7 01 FA 00 05 00 00 00 01 00 51 03 03 59 F5 
16:15:19.725 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  

ap_mavlink_version=3
16:15:22.154 ->  UDP WiFi record read:
16:15:22.189 -> 84 FE 09 0A 01 FA 00 05 00 00 00 01 00 51 03 03 0F 84 
16:15:22.189 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  

ap_mavlink_version=3
16:15:24.719 ->  UDP WiFi record read:
16:15:24.719 -> 44 FE 09 49 01 FA 00 05 00 00 00 01 00 51 03 03 5E 44 
16:15:24.755 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  

ap_mavlink_version=3
16:15:25.767 ->  UDP WiFi record read:
16:15:25.767 -> 0D FE 09 56 01 FA 00 05 00 00 00 01 00 51 03 03 6D 0D 
16:15:25.767 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  

ap_mavlink_version=3

lipk888, 2020-10-01:
> hi zs6buj:
> i get log below, it seems no GPS massage feedback
> 16:14:51.095 -> Starting up......
> 16:14:51.162 -> Target Board is ESP32 Dev Module
> 16:14:51.195 -> Mavlink WiFi In
> 16:14:51.229 -> Az = [...]

lipk888, 2020-10-01:
sorry for mis opration for close issue.....i just reopen it ......sorry.
i just check debug mode, seems even i set GPS debug it only can output heartbeat debug

zs6buj, 2020-10-01:
Could you comment out all debugs except the GPS debug and run it again, but longer. Remember your GPS must have clear access to the sky. :)

lipk888, 2020-10-01:
Hi zs6buj:
i try set below , but still only have #0 massage out
Starting up......
22:16:28.746 -> Target Board is ESP32 Dev Module
22:16:28.781 -> Mavlink WiFi In
22:16:28.815 -> Az = 90.00	 El = 0.00	 hmHdg = 90.00
22:16:28.815 -> pntAz = 90 El = 0.00 LastGoodpntAz = 90 LastGoodEl = 0 hmHdg = 90.00 azPWM = 1500 elPWM = 2300
22:16:28.815 -> Heading_Source = Tracker's Own Compass
22:16:28.850 -> ----- Compass Found -----
22:16:28.850 -> Sensor:       HMC5883
22:16:28.850 -> Driver Ver:   1
22:16:28.850 -> Unique ID:    12345
22:16:28.850 -> Max Value:    800.00 uT
22:16:28.850 -> Min Value:    -800.00 uT
22:16:28.850 -> Resolution:   0.20 uT
22:16:28.850 -> --------------------------
22:16:28.850 -> 
22:16:28.850 -> Display tracker heading for 20 seconds
22:16:28.850 -> Heading = 27
22:16:29.866 -> Heading = 27
22:16:30.848 -> Heading = 28
22:16:31.869 -> Heading = 28
22:16:32.854 -> Heading = 27
22:16:33.878 -> Heading = 27
22:16:34.857 -> Heading = 27
22:16:35.872 -> Heading = 27
22:16:36.858 -> Heading = 27
22:16:37.871 -> Heading = 28
22:16:38.882 -> Heading = 27
22:16:39.861 -> Heading = 28
22:16:40.873 -> Heading = 27
22:16:41.855 -> Heading = 27
22:16:42.870 -> Heading = 27
22:16:43.885 -> Heading = 27
22:16:44.867 -> Heading = 27
22:16:45.883 -> Heading = 27
22:16:46.869 -> Heading = 28
22:16:47.883 -> Heading = 28
22:16:48.864 -> Trying to connect to tbs_crossfire_D8BFC0FE921E.
22:16:50.492 -> WiFi connected!
22:16:50.492 -> Local IP address: 192.168.4.100
22:16:50.492 -> WiFi RSSI:-75 dBm
22:16:50.525 -> UDP started, listening on IP 192.168.4.100, UDP port 8800
22:17:12.020 -> Client connected: Remote UDP IP: 192.168.4.1  Remote  UDP port: 8800
22:17:12.308 ->  UDP WiFi record read:
22:17:12.308 -> 41 FE 09 CA 01 FA 00 05 00 00 00 01 00 51 03 03 BE 41 
22:17:12.344 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
22:17:12.344 ->  hb_count=1
22:17:13.677 ->  UDP WiFi record read:
22:17:13.677 -> 20 FE 09 F0 01 FA 00 05 00 00 00 01 00 51 03 03 61 20 
22:17:13.677 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
22:17:13.677 ->  hb_count=2
22:17:19.288 ->  UDP WiFi record read:
22:17:19.288 -> EF FE 09 6F 01 FA 00 05 00 00 00 01 00 51 03 03 8C EF 
22:17:19.288 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
22:17:19.324 ->  hb_count=3
22:17:19.324 -> hbGood=true
22:17:19.612 ->  UDP WiFi record read:
22:17:19.612 -> FF FE 09 88 01 FA 00 05 00 00 00 01 00 51 03 03 05 FF 
22:17:19.612 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
22:17:20.332 ->  UDP WiFi record read:
22:17:20.332 -> 48 FE 09 A2 01 FA 00 05 00 00 00 01 00 51 03 03 0D 48 
22:17:20.332 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
22:17:21.448 ->  UDP WiFi record read:
22:17:21.448 -> 0F FE 09 BB 01 FA 00 05 00 00 00 01 00 51 03 03 53 0F 
22:17:21.448 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
22:17:22.672 ->  UDP WiFi record read:
22:17:22.672 -> 76 FE 09 D4 01 FA 00 05 00 00 00 01 00 51 03 03 67 76 
22:17:22.672 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
22:17:24.630 ->  UDP WiFi record read:
22:17:24.630 -> E6 FE 09 07 01 FA 00 05 00 00 00 01 00 51 03 03 3F E6 
22:17:24.630 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
22:17:25.439 ->  UDP WiFi record read:
22:17:25.439 -> B3 FE 09 14 01 FA 00 05 00 00 00 01 00 51 03 03 D6 B3 
22:17:25.439 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
22:17:27.272 ->  UDP WiFi record read:
22:17:27.272 -> F7 FE 09 3A 01 FA 00 05 00 00 00 01 00 51 03 03 67 F7 
22:17:27.272 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
22:17:27.883 ->  UDP WiFi record read:
22:17:27.883 -> 80 FE 09 53 01 FA 00 05 00 00 00 01 00 51 03 03 3E 80 
22:17:27.883 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
22:17:28.497 ->  UDP WiFi record read:
22:17:28.497 -> 70 FE 09 60 01 FA 00 05 00 00 00 01 00 51 03 03 68 70 
22:17:28.497 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
22:17:28.907 ->  UDP WiFi record read:
22:17:28.907 -> 12 FE 09 6D 01 FA 00 05 00 00 00 01 00 51 03 03 58 12 
22:17:28.907 -> Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=0  ap_base_mode=81 ap_custom_mode=5  ap_system_status=3  ap_mavlink_version=3
[image]

zs6buj, 2020-10-02:
Yes, it looks like you are getting only heartbeats down from crossfire

lipk888, 2020-10-02:
Hi zs6buj:
today i have try other way to make sure mavlink massage send out from INAV and i confirm massage send out from INAV.
i just change crossfire Nano output from
(CRFS+MAVLINK-->CH1/2 for CRSF CH3/4 for mavlink ) to (SBUS+MAVLINK  CH1--> SBUS  CH3/4--> MAVLINK), 
due to SBUS only output massage for RC and mavlink.

so for result, i still only can get #0 heartbeat massage and no GPS, but on transmitter if show GPS location.

lipk888, 2020-10-02:
HI Zs6buj:
i just change code below to "GCS_Mavlink.h", so i'm sure this why i only get #0 heartbeat?
[image]

lipk888, 2020-11-04:
Hi Zs6buj:
finally i did it, all system run good base on V2.15.

closed 2020-11-04 by lipk888