zs6buj/AntTracker #25: compilation

secretman28, 2020-10-02:
no matching function for call to 'Adafruit_SSD1306::Adafruit_SSD1306(int, int, TwoWire*, int)'
what is wrong, 

Heltec Wifi Kit 32
i cant find the error ;(

zs6buj, 2020-10-02:
Please use the Adafruit libraries in the library folder inside AmtTracker folder

[image]

secretman28, 2020-10-03:
will give it a try today

secretman28, 2020-10-03:
better but no
'SetHomePin' was not declared in this scope

secretman28, 2020-10-03:
uint8_t SetHomeState = digitalRead(SetHomePin);            // Check if home button is pushed = 0

secretman28, 2020-10-05:
cant get it to work even if i try other board kind same error
dont know what i do wrong, im using the  includ library

zs6buj, 2020-10-05:
I just compiled AntTrackv2.14 clean again here. Checked all the libraries. Sorry, I don't know how to help you.  Is your Arduino IDE working for other compiles? Maybe try some examples.

secretman28, 2020-10-05:
yes my arduino work for other thnig ;(
can you build an  #define ESP32_Variant     4    //  Heltec Wifi Kit 32 - contributed by Noircogi ?
that i can test  its a week i try to find what dont work here :(

zs6buj, 2020-10-05:
Yes sure.

zs6buj, 2020-10-05:
The Heltec binary for 2.14 is on github now.

I had to add the pin info for the Heltec board, and guessed the Servo pins.

I need to tidy up this version because I did it in a hurry and never got around to cleaning it up. My bad. I'll try to do that tomorrow.

secretman28, 2020-10-05:
tanks a lot I give it a try tonight :)

secretman28, 2020-10-08:
i cant find the bin the only one that was there is the one it was when i download can you point me at it ?
tanks

zs6buj, 2020-10-12:
Hi secretman28 

I updated features and published v2.15. You will find a Heltec WiFi kit 32 binary there.

secretman28, 2020-10-12:
tanks a lot !

mello73, 2021-01-24:
Hello,
Please, what is the best version of the arduino IDE for compile for the ESP32
Regards and Happy New Year

secretman28, 2021-01-25:
> Hello,
> Please, what is the best version of the arduino IDE for compile for the ESP32
> Regards and Happy New Year

I never got it to work I don't have time to get in to it probably a stupidity but never find it

zs6buj, 2021-01-25:
Hi @secretman28 

I'm using v1.8.13 of Arduino IDE.

https://randomnerdtutorials.com/installing-the-esp32-board-in-arduino-ide-windows-instructions/

mello73, 2021-01-25:
Hello evreybody,
Thanks @secretman28
Thanks @zs6buj

I proced to instal, program and test the chain.

I'm very very iteresting  to see the reaction of the tracker. I never see a tracker works ,,,

mello73, 2021-01-25:
Compilation AntTrack_v2.15.4 OK 
- Arduino 1.8.12
- copy all the librery from git & add PixhawkArduinoMAVLink 1.0.1

[image]
[image]
[image]

mello73, 2021-01-25:
Great :)

mello73, 2021-01-25:
Little two questions please ( I don't know if here is the good place) :

1. With this tracker I only need one gps and in this case, the tracker record the gps coordinate when I push the "Set_Home button" ?
2. Can I use for the tracker the compass integer in M8Q-5883 ( the QMC5883L ) ?

Thanks a lot

zs6buj, 2021-01-26:
> 1.The tracker can use only one gps in the aircraft and in this case, the tracker record the gps coordinate when I push the "Set_Home button" ?

Yes, the tracker gets the GPS do-ordinates from the GPS on the aircraft. But it also needs to know the direction the tracker is facing at rest (when you switch on). From that it can extrapolate where to point when the aircraft moves.

The tracker can take the direction from the aircraft at rest if the aircraft has a compass, or you can "set home", walk with the aircraft in the direction the tracker is facing (usually the field in front of you), and push the button again. These are options in the firmware macros.

Some guys have added a compass onto the tracker box to use as a reference for the assembly at rest.

zs6buj, 2021-03-28:
V2.17.0 Beta published with compass and GPS on the tracker box. Moving Tracker box supported.

closed 2021-03-28 by zs6buj