zs6buj/AntTracker #26: Compilation ESP32

Almischalmi, 2020-10-16:
I get the following error which seems to point to a STM32 library:

In file included from C:\Users\Elmar\Documents\GitHub\AntTracker\AntTrack_v2.15\EEPROM.ino:6:0:
C:\Users\Elmar\Documents\Arduino\libraries\EEPROM/EEPROM.h:4:20: fatal error: wirish.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.

zs6buj, 2020-10-17:
If you are compiling for the Dev Board, make sure the ESP32 EEPROM lib is used. Try ticking verbose compiling in preferences.

I did a trial recompile now and found a problem with the dev board and no display support. Fixed in v2.15.1.

Almischalmi, 2020-10-17:
Thank you, that helped. Now I only have compilation problems when trying to compile with SSD1306:

Arduino: 1.8.13 (Windows 10), Board: "Node32s, 80MHz, 921600"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Elmar\AppData\Local\Arduino15\packages -hardware C:\Users\Elmar\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Elmar\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Elmar\Documents\Arduino\libraries -fqbn=esp32:esp32:node32s:FlashFreq=80,UploadSpeed=921600 -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\Elmar\AppData\Local\Temp\arduino_build_925207 -warnings=none -build-cache C:\Users\Elmar\AppData\Local\Temp\arduino_cache_323315 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\Elmar\Documents\GitHub\AntTracker\AntTrack_v2.15\AntTrack_v2.15.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Elmar\AppData\Local\Arduino15\packages -hardware C:\Users\Elmar\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Elmar\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Elmar\Documents\Arduino\libraries -fqbn=esp32:esp32:node32s:FlashFreq=80,UploadSpeed=921600 -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\Elmar\AppData\Local\Temp\arduino_build_925207 -warnings=none -build-cache C:\Users\Elmar\AppData\Local\Temp\arduino_cache_323315 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\Elmar\Documents\GitHub\AntTracker\AntTrack_v2.15\AntTrack_v2.15.ino

Using board 'node32s' from platform in folder: C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4

Using core 'esp32' from platform in folder: C:\Users\Elmar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4

Detecting libraries used...

"C:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\Elmar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32 [... cut ...]
Using library Adafruit_Sensor-master at version 1.0.2 in folder: C:\Users\Elmar\Documents\Arduino\libraries\Adafruit_Sensor-master 

Using library Adafruit_HMC5883_Unified at version 1.2.0 in folder: C:\Users\Elmar\Documents\Arduino\libraries\Adafruit_HMC5883_Unified 

Using library EEPROM at version 1.0.3 in folder: C:\Users\Elmar\Documents\Arduino\libraries\EEPROM 

Using library TroykaGPS at version 1.0.0 in folder: C:\Users\Elmar\Documents\Arduino\libraries\TroykaGPS 

exit status 1

no matching function for call to 'Adafruit_SSD1306::Adafruit_SSD1306(int, int, TwoWire*, int)'

zs6buj, 2020-10-17:
Yeah, we are using an older SSD1306 library. Have you used all the libraries from here 

https://github.com/zs6buj/AntTracker/tree/master/libraries

Almischalmi, 2020-10-17:
Yes, I have, that's why I'm wondering. Your building environment seems pretty custom...

zs6buj, 2020-10-18:
I took a careful look, and I'm using the latest Adafruit SSD1306 library.  My development machine crashed a month or two back and I rebuilt everything and must have switched libraries. Sorry for the waste of time. I'll update the lib folder.

[image]

Almischalmi, 2020-10-18:
Thank you for creating that project in the first place. 

Now it's actually compiling with display support. 

But when I compile it with UDP in STA mode, I get the following error:

C:\Users\Elmar\Documents\GitHub\AntTracker\AntTrack_v2.15.1\Mavlink.ino: In function 'bool Send_UDP(mavlink_message_t*)':
Mavlink:520:35: error: 'set' was not declared in this scope
     UDP.beginPacket(UDP_remoteIP, set.udp_remotePort);
                                   ^
exit status 1
'set' was not declared in this scope

Sorry, I'm just not used to programming anymore, especially with OO languages.

zs6buj, 2020-10-18:
Ok, fixed. v2.15.2

Thanks for the testing :)

Almischalmi, 2020-10-19:
perfect, thank you for the quick fix

closed 2020-10-19 by Almischalmi