zs6buj/AntTracker #27: R9M 2019 ACCESS

MrCoopa, 2020-10-19:
Hello, 
does it work with the R9M 2019 ACCESS protocol? The U360UTS has trouble with it maybe this could be an alternative.
Thank you!

zs6buj, 2020-10-19:
Hi Mr Coopa, yes it should work now if you use SPort on ACCESS. I'm testing FPort right now and should publish in a few days

MrCoopa, 2020-10-21:
Thank you. Can you write a comment if it works?
Have a great day!

zs6buj, 2020-10-21:
Unfortunately my R9M/R9 is the pre ACCESS model so I have no way of testing ACCESS on it.

I'm using a Taranis lite and Archer receiver to test ACCESS + FPort. I also have an older non-access Horus and Taranis. I guess they are now ornaments.:(

zs6buj, 2021-03-16:
Ok, you can try the latest V2.16.3 with R9M/R9 and S.Port. It tests good here, but I'm still waiting for kit to test F.Port.

I've also added onto my GitHub FrSkyTo UDP, so you can add an ESP32/8266 on the RM9 and communicate with the tracker using WiFi/UDP. Feedback please. :)

zs6buj, 2021-03-28:
My RM9 kit arrived, and I'll do some testing soon.

zs6buj, 2021-03-31:
It has become apparent that 2019 (ACCESS enabled) RM9 / R9, when inserted into the older Horus X10, even with the ACCESS update kit installed, will not support ACCESS. Just another scam by FrSky to force us to throw out all our  expensive 1 or 2 year old, but now obsolete, kit and buy their new models. Time to give the FrSky proposition a re-think and look at Jumper.

MrCoopa, 2021-04-04:
Is it  possible to convert the R9M Access telemetry protocol into Mavlink and then "pipe" it to the Anttracker as Mavlink?

zs6buj, 2021-04-05:
Yes it is possible,  but not necessary. You can feed ACCESS/Fport telemetry directly from the port on the R9M into the tracker. I have tested with ACCESS / Fport out of a Archer receiver and it works fine. I have not tested ACCESS out of the R9M because my Horus is a pre 2019 version. However, I do  expect it to work or maybe require a minor tweak or two. Perhaps you could give it a try for me and report back.

I have also written a little module you can flash onto a small ESP32 board that reads the R9M port and sends WiFi UDP that the tracker can read. It's on my github.

MrCoopa, 2021-04-11:
I ordered some components now and want to give it a try. I will use an ESP32 Dev board and build the 180° version because I don't want to go with a slip ring. 
Part List:

- Electronic box
- ESP32 Dev board
- 2x MG995
- Pan & Tilt Mechanism
- AKK 5.8 GHz diversity receiver
- 150mm Triple Feed Patch
- 5V / 3A UBEC
- a HC-05 with 10kOhm and 2N7000 for Taranis telemetry
- 0.96 OLED display
- button, led, connectors and that stuff

Wires and tools for crimping are at home.

Is anything missing?

zs6buj, 2021-04-11:
MrCoopa:

I would like to understand how you plan to use the HC05 for Taranis telemetry. Could you perhaps draw a rough sketch?

MrCoopa, 2021-04-11:
Oh sorry its not the telemetry from the Taranis but the telemetry out of the R9M ACCESS.

There are several videos: 
https://www.youtube.com/watch?v=lmSMt6JLWL8
https://www.youtube.com/watch?v=0-AyP5Y7pCI

zs6buj, 2021-04-11:
Ah, OK, I get what you want to do. Should be fine.

MrCoopa, 2021-04-14:
I'm using a F405 Wing in the aircraft and have an r9mm using sbus and s port separately. Is it ok for your code or is there a need to change to fport?

MrCoopa, 2021-04-14:
I've problems with the compiler:

In file included from D:\Dokumente\Arduino\AntTracker-master\Stable\AntTrack\EEPROM.ino:6:0:
D:\Dokumente\Arduino\libraries\EEPROM/EEPROM.h:4:20: fatal error: wirish.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.

I used all your libraries but it doesn't fix it.

zs6buj, 2021-04-14:
Either S.Port or F Port. I'd like to test F.Port 2.

MrCoopa, 2021-04-14:
Actually it runs on S Port. If I get the Tracker to work I can change to F Port and try it for you. 

The problem is still current, seems to be a fault with the libraries.

zs6buj, 2021-04-14:
Strange that it can't find that header. I usually use the latest esp32 libs from the espressif web site

zs6buj, 2021-04-14:
https://github.com/espressif/arduino-esp32/tree/master/libraries

MrCoopa, 2021-04-14:
Its showing different faults:

"C:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_trace" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_update" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/asio" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bootloader_support" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bt" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/coap" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/console" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/driver" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/efuse" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp-tls" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp32" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_event" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_client" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_server" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_ota" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_server" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_websocket_client" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/espcoredump" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/ethernet" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/expat" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/fatfs" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freemodbus" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freertos" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/heap" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/idf_test" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/jsmn" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/json" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/libsodium" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/log" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/lwip" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mbedtls" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mdns" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/micro-ecc" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mqtt" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/newlib" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/nghttp" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/nvs_flash" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardw [... cut ...]
Using library Adafruit_Unified_Sensor at version 1.1.4 in folder: D:\Dokumente\AntTracker-master\Stable\AntTrack\libraries\Adafruit_Unified_Sensor 
Using library Adafruit_HMC5883_Unified at version 1.2.0 in folder: D:\Dokumente\AntTracker-master\Stable\AntTrack\libraries\Adafruit_HMC5883_Unified 
Using library EEPROM at version 1.0.3 in folder: C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\EEPROM 
exit status 1
Error compiling for board ESP32 Dev Module.

zs6buj, 2021-04-14:
Did you install the esp32 kit:

https://randomnerdtutorials.com/installing-the-esp32-board-in-arduino-ide-windows-instructions/

MrCoopa, 2021-04-14:
Yes, I do.

zs6buj, 2021-04-14:
I'm not at home right now, and tomorrow is bad for me, but I'll look at it as soon as I can.

MrCoopa, 2021-04-14:
Thank you. I'm trying further but it would be good to have help. 
It seems to be with the EEPROM like I said in the beginning

MrCoopa, 2021-04-14:
I've tried the lastest Beta, but the result are the same. I deactivated all default libraries by moving them away and use only your custom:

Arduino: 1.8.13 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

D:\Programme\Arduino 1.8.13\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Programme\Arduino 1.8.13\Arduino\hardware -hardware C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages -tools D:\Programme\Arduino 1.8.13\Arduino\tools-builder -tools D:\Programme\Arduino 1.8.13\Arduino\hardware\tools\avr -tools C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages -libraries D:\Dokumente\AntTracker-master\Stable\AntTrack\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=115200,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\MILENA~1\AppData\Local\Temp\arduino_build_670860 -warnings=default -build-cache C:\Users\MILENA~1\AppData\Local\Temp\arduino_cache_934214 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -verbose D:\Dokumente\latest Beta\AntTrack\AntTrack.ino

D:\Programme\Arduino 1.8.13\Arduino\arduino-builder -compile -logger=machine -hardware D:\Programme\Arduino 1.8.13\Arduino\hardware -hardware C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages -tools D:\Programme\Arduino 1.8.13\Arduino\tools-builder -tools D:\Programme\Arduino 1.8.13\Arduino\hardware\tools\avr -tools C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages -libraries D:\Dokumente\AntTracker-master\Stable\AntTrack\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=115200,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\MILENA~1\AppData\Local\Temp\arduino_build_670860 -warnings=default -build-cache C:\Users\MILENA~1\AppData\Local\Temp\arduino_cache_934214 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -verbose D:\Dokumente\latest Beta\AntTrack\AntTrack.ino

Using board 'esp32' from platform in folder: C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

Using core 'esp32' from platform in folder: C:\Users\Milena und Niklas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

cmd /c if exist "D:\\Dokumente\\latest Beta\\AntTrack\\partitions.csv" copy /y "D:\\Dokumente\\latest Beta\\AntTrack\\partitions.csv" "C:\\Users\\MILENA~1\\AppData\\Local\\Temp\\arduino_build_670860\\partitions.csv"

cmd /c if not exist "C:\\Users\\MILENA~1\\AppData\\Local\\Temp\\arduino_build_670860\\partitions.csv" copy "C:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\partitions\\default.csv" "C:\\Users\\MILENA~1\\AppData\\Local\\Temp\\arduino_build_670860\\partitions.csv"

Detecting libraries used...

"C:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_trace" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_update" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/asio" "-IC:\\Users\\Milena und Niklas\\AppData\\Local\\Arduino [... cut ...]
Using library ESP32_Servo at version 1.0 in folder: D:\Dokumente\AntTracker-master\Stable\AntTrack\libraries\ESP32_Servo 

Using library SPI at version 1.0 in folder: D:\Dokumente\AntTracker-master\Stable\AntTrack\libraries\SPI 

Using library Wire at version 1.0 in folder: D:\Dokumente\AntTracker-master\Stable\AntTrack\libraries\Wire 

Using library EEPROM in folder: D:\Dokumente\AntTracker-master\Stable\AntTrack\libraries\EEPROM (legacy)

exit status 1

Error compiling for board ESP32 Dev Module.

zs6buj, 2021-04-15:
I'm travelling today, so using my phone.

May I suggest you remove all eeprom libraries, including from  my library.
The Arduino IDE should then select the one from the the ESP32 IDF.

MrCoopa, 2021-04-15:
Thank you very much for this advise. I was able to compile the sketch with the option : #define Telemetry_In  0    // Serial Port (default) - all protocols  
If I change it to: #define Telemetry_In  1   with the bluetooth input im getting the error :

In file included from D:\Dokumente\AntTracker-master\Stable\AntTrack_new\AntTrack_new.ino:179:0:
config.h:576:8: error: #error Bluetooth only available on ESP32
       #error Bluetooth only available on ESP32
        ^

My setting for the ESP32 is: #define ESP32_Variant     1    //  ESP32 Dev Module - there are several sub-variants that work
and its definitely a module with bluetooth.
Maybe you can take a look at it I want to feed the tracker over bluetooth.

zs6buj, 2021-04-16:
Yes, thank you for the feedback. I'll fix this today.

zs6buj, 2021-04-16:
Ok fixed. There was a typo in line 680

 ` #if (defined ESP323)  should be   #if (defined ESP32) `

MrCoopa, 2021-04-16:
It works thank you.
I found a bug in the Anttrack.ino:

The libraries are not in your libraries folder:
#include <mavlink_types.h> 
#include <ardupilotmega/mavlink.h>
#include <ardupilotmega/ardupilotmega.h>

I fixed this error by including the #include <GCS_MAVLink.h> library. 

Without this change there was an error that the  <mavlink_types.h> is missing.

MrCoopa, 2021-04-16:
My tracker shows now that the input is input "mavlink bt in". I want to inport R9M S Port telemetry via bluetooth. Is it possible aswell?

zs6buj, 2021-04-16:
Yes, but I'll have to add some code.  Wifi udp is available for FrSky already, but not BT yet.

MrCoopa, 2021-04-16:
Is it possible that you do the changes next time?

zs6buj, 2021-04-17:
I will try to do it in the next few days, free time allowing.

In the meantime, if you have an extra ESP32 or ESP8266, you could try it on the Taranis side using WiFi UDP with this firmware  https://github.com/zs6buj/FrSky-Passthru-To-WiFi-and-or-Bluetooth

zs6buj, 2021-04-17:
> It works thank you.
> I found a bug in the Anttrack.ino:
>
> The libraries are not in your libraries folder:
> #include <mavlink_types.h>
> #include <ardupilotmega/mavlink.h>
> #include [...]

The Mavlink libraries are rather large.

So I put this link is in the library folder instead.

https://github.com/mavlink/c_library_v2

MrCoopa, 2021-04-17:
Ok. I'm aktually trying with the serial in instead of the bluetooth input. The FRSky Telemetry was found in my latest test.

I will compile the latest version when you have done the adaptation and take it as the final version for me.
The tracker will be stationary with the compass module without a GPS.

zs6buj, 2021-04-17:
Ok, I banged together the FrSky BT option. It's in the Alpha folder, and completely untested.

https://github.com/zs6buj/AntTracker/tree/master/Alpha

MrCoopa, 2021-04-26:
Thank you for the changes. I will test it in the next days. 
I had to wait for any parts until now.

zs6buj, 2021-11-04:
stale

closed 2021-11-04 by zs6buj