zs6buj/AntTracker #28: R9M ACCST S.PORT connexion / lavlink detection

mello73, 2021-01-30:
Hello,

I have 
- R9M ACCST version ( the older version of the ACCESS protocol )
- Aduplane V4.1.0 
[image]

- SERIAL PROTOCOL 23

I dont't know if my process or connexion wirring it's ok or not

1. connct Sport from R9M to pin 27 ESP32
2. [image]
3. powerup plane, wait fix 3D
4. powerup radio and recive = telemetry OK
5. check logs 
 
[image]

zs6buj, 2021-02-01:
Hi @mello73 

It's interesting that we could detect the baud rate, but still not find Frsky telemetry.  To find Frsky telem we look for 0x7E like this:

`      if (chr==0x7E) FSky++;                     // Found candidate FrSky`

S.Port telem is inverted, meaning that it is "idle low", while "normal" telemetry is idle high. So either we tell the esp32 to expect inverted, or you need a hardware inverter.

Let's rather tell the esp32 to expect inverted telemetry

In tab GetProtocol, line 60 replace

`  inSerial.begin(baud);`

with this:

```
    #if ( (defined ESP8266) || (defined ESP32) ) 
      inSerial.begin(baud, SERIAL_8N1, rxPin, txPin, frInvert); 
    #else
      inSerial.begin(baud);
    #endif 
```   

Now in the main tab, called AntTrack..., line 486, similarly  replace

`    inSerial.begin(baud); `

with this:

```
    #if ( (defined ESP8266) || (defined ESP32) ) 
      inSerial.begin(baud, SERIAL_8N1, rxPin, txPin, frInvert); 
    #else
      inSerial.begin(baud);
    #endif 
```

Now, in the config.h tab, in the configuration for ESP32 Dev Module, approximately line 198, insert 

`  #define frInvert          true `

I will make these changes and publish v2.15.5.

Could you let me know the outcome?

Regards
Eric

mello73, 2021-02-01:
THANKS Eric for your reactivity !

I apply all the modify, but I can't test because : 
<<frInvert' was not declared in this scope >>
[image]

Thanks
,,,

zs6buj, 2021-02-02:
Ok, missed that. Put a hash # in front of define frInvert true

#define frInvert true

mello73, 2021-02-02:
Hello Eric,

Thanks !

A new little error : 

[image]
[image]

Have a nice day

zs6buj, 2021-02-02:
Strange. Did you select Dev Board ?

```
//*********************************************************************************************
//**********************   S E L E C T   E S P   B O A R D   V A R I A N T   ******************

#define ESP32_Variant     1    //  ESP32 Dev Module - there are several sub-variants that work
```

mello73, 2021-02-02:
**Merci Eric !
Great ! Great ! Great !**

 hbGood = 1   gpsGood = 1   homeInitialised = 1
  0x5002 Num sats=13 gpsStatus=3 HDOP=8 fr_vdop=0 gpsAlt=22.8 neg=0
 hbGood = 1   gpsGood = 1   homeInitialised = 1
 hom.lat = 45.5XXX hom.lon = 6.XXX hom.alt = -3 cur.lat = 45.XXX cur.lon = 6.XXX cur.alt = 23
   hc_vector.az= 283.66  hc_vector.el= 62.76  hc_vector.dist= 12  Elapsed= 273.836
 Az = 283.66	 El = 62.76	 hmHdg = 241.47
 pntAz = 132 El = 62.76 LastGoodpntAz = 132 LastGoodEl = 62 hmHdg = 241.47 azPWM = 1080 elPWM = 1780
 hbGood = 1   gpsGood = 1   homeInitialised = 1

**Yes !!!!**

You have reason

At the start, I test different configuration and only one esp configuratyon active the display for my cheap ESP TTGO : 
- Variant     4    //  Heltec Wifi Kit 32

For momentanly solve my issue I add in the main tab, called "AntTrack.." at line 187 : 

bool frInvert = true;

I think it's not the best pactice ,,,, 

**Thanks a lot**

zs6buj, 2021-02-02:
un plaisir mon ami

I plan to upgrade support for more boards and more displays, but time escapes me.

mello73, 2021-02-02:
**C'est toi mon AMI -> Merci pour ton Grand Aide !**
I never see in the past a tracking antenna working .
I want to see and check if the algorithm works good ,,,
I'm very very impatiently !!!
I wont to see an check if the algoritms works good ,,, 

I'm very very impatiently !!!

mello73, 2021-02-04:
All Works good
Merci

zs6buj, 2021-02-27:
Bonjour mon ami 

I came back here because the other site was getting a bit jumbled.

I recommend this way forward now that your servos are working:

1. Clone [this firmware](https://github.com/zs6buj/FrSky-Passthru-To-WiFi-and-or-Bluetooth), configure for yourself and UDP out, compile and flash to ESP32 number 1. Connect wires from RM9 S.Port to rx pin of your ESP32, and ground to ground. Start the board and view the log output to the monitor. If you have a display, wait and then see HUD on display.
2. Clone new [Tracker](https://github.com/zs6buj/AntTracker) firmware, configure for FrSky UDP in, flash to ESP32 number 2, start it and view log on serial monitor. If you have a display, wait and HUD info will appear on display.
3. Compare the two HUD's info, should be the same.

I have also added FrSky UDP out into Mav2PT.

Cordialement, mon ami. Nous partons pour le centre commercial.

zs6buj, 2021-02-27:
If you PM me I will give you my e-mail address

mello73, 2021-02-27:
Hello Eric, 

my adress is   
[image]

Merci

closed 2021-03-28 by zs6buj