zs6buj/AntTracker #29: tracking box add Local GPS

wsqzdz, 2021-02-21:
This is another very good project of antenna tracker. I've used ardupilot AAT; u360gts, and I'm looking forward to adding more features，In future versions, will the tracking box add its own GPS to realize tracking on the move

wsqzdz, 2021-02-21:
Another problem if I want to connect the DIY stepper motor control board. Whether the servo drive PWM data can be sent out through the serial port does not need two separate ports to output PWM,

wsqzdz, 2021-02-21:
Another question if I want to connect the DIY stepper motor control board. Whether the servo drive PWM data can be sent out through the serial port does not need two separate ports to output PWM,

zs6buj, 2021-02-21:
Hi @wsqzdz 

`"In future versions, will the tracking box add its own GPS to realize tracking on the move"`

Um, I'm quite busy right now, and also moving house shortly, so not in the next month or two. The change is fairly easy, so we could get to it after we move. **If anyone wants an interesting challenge, I will happily entertain PRs** .

"DIY stepper motor control board".  Please could you post more info or links. One tx serial port pin is enough for PWM data.

wsqzdz, 2021-02-21:
Similar to this project [](https://github.com/kolins-cz/antenna-tracker-mechanics)， the use of stepper motor driven antenna tracker. Now I've used atmege328p + a4988 to run it with accelstepper library. The PWM measurement of pan and tilt drives two stepping motors to make them suitable for different tracking antenna projects, but he still has some small problems

wsqzdz, 2021-02-21:
https://github.com/kolins-cz/antenna-tracker-mechanics

zs6buj, 2021-03-16:
I'm looking into adding a GPS on the tracking box right now. I will need someone to test it please as all my equipment is being packed up for a house move.

zs6buj, 2021-03-28:
Done. V.2.17.0 Beta. Please help with testing.

closed 2021-03-28 by zs6buj