zs6buj/AntTracker #30: esp32 dev board - not working

pan-sattan, 2021-02-25:
anyone can help me?

I am using:

betaflight + fport R9mm

X9D + R9M ACCESS - S.PORT + GND pins used

esp32 dev board

#define ESP32_Variant 1

OLED display GND, 5V, SDA - PIN 21, SCL - PIN 22

#define Display_Support

#define SSD1306_Display

Serial port on PIN27, GND

#define Telemetry_In 0 

GY-271 GND, 5LV, SDA - PIN 21, SCL - PIN 22 

const uint8_t Heading_Source = 2;

nothing else is connected or changed in config...

oled display is working, but in debug I can see that no telemetry is going inside... I tried inverter and the same... on the screen is that telemetry is not going into esp32 board

Any idea what should be wrong? could be, that I dont have in my flat any gps sats found? zero values comes from taranis..

Is there any possibility to switch in config FAKE VALUES?

[image]

[image]

Almischalmi, 2021-02-25:
I have a similar issue. Baud detection works, but no mavlink detection (Mavlink 1/2, 115k/57k)

pan-sattan, 2021-02-26:
all infos are there... latest betaflight, fport, access protocol, latest opentx... I am able to decode s.port from R9m with another arduino and code... but want to uae this one for tracker

zs6buj, 2021-02-26:
Hi Guys, 

Merci d'être venu Marius.

I will look at this today. One thing you have not given me is firmware version number.

Remember f.port is 115200 b/s.

pan-sattan, 2021-02-26:
firmware of betaflight is 4.2.2 I checked it right now... or do you mean another firmware?

pan-sattan, 2021-02-26:
and if I use bluetooth mod for my R9m together with app telemetry viewer, it works normally. My HC-05 is set to baud 57600, even that works with arduino decoder from Pawelsky

zs6buj, 2021-02-26:
No, I meant AntTrack version number. So are you using Mavlink, and serial or BT

Almischalmi, 2021-02-26:
I've tried several of the newer versions. Mavlink on 57k serial.

pan-sattan, 2021-02-26:
I used v2.15.4 and frsky protocol, serial input 
Now I tried v2.15.5 and it moves on! 
I will keep testing... 

[image]

[image]

zs6buj, 2021-02-26:
Oops, it's my bad!

I added frInvert for FrSky telemetry input (for mello73), but left it so it is also inverting Mavlink telemetry.

Please, in the meantime, go to your ESP32 variant setup, and change frInvert line like this:

  `  bool frInvert = false;               // DON'T INVERT FOR MAVLINK`

I am about to publish a new version, with FrSky UDP input and a few other improvements. I will fix this issue there.

Sorry for the inconvenience. No excuses for this!

pan-sattan, 2021-02-26:
I am glad, that I can test it, or help you in development...
I change it and thats the result:

[image]

pan-sattan, 2021-02-26:
[image]

[image]

[image]

zs6buj, 2021-02-26:
Yeah, it looks correct now. Thanks for the feedback.

Now maybe activate these macros to see some telemetry:

```
#define Debug_Mav_Heartbeat  
#define Debug_Mav_GPS   
```

pan-sattan, 2021-02-26:
I tried, but it stops with Mavlink 2 found...
nothing else is comming

pan-sattan, 2021-02-26:
could be the betaflight issue? or maybe non GPS signal in my flat

pan-sattan, 2021-02-26:
[image]

zs6buj, 2021-02-26:
I'm not sure what protocol Betaflight outputs these days

pan-sattan, 2021-02-26:
I think the same like INAV

[image]

pan-sattan, 2021-02-26:
this works for me with decoder from Pawelsky

zs6buj, 2021-02-26:
Can your fc connect to a Mission Planner. I'm afraid I don't know Pavelka. Can you post a link?

pan-sattan, 2021-02-26:
I will try later mission planner.

https://www.rcgroups.com/forums/showthread.php?2245978-FrSky-S-Port-telemetry-library-easy-to-use-and-configurable

pan-sattan, 2021-02-26:
I am not able to connect my FC in MP

[image]

[image]

zs6buj, 2021-02-27:
Could you do some research and find out which protocol you are using.

pan-sattan, 2021-02-27:
its Frsky  protocol

zs6buj, 2021-02-27:
Ok pan-sattan, we should be able to solve this. I just tested FrSky S.Port again and it works here, but now I see above you mention F.Port. I have not added F.Port support yet.

pan-sattan, 2021-02-27:
ok, is there something what I can do or help you somehow?

zs6buj, 2021-02-27:
Thanks for offering, but I already have written the F.Port code for other projects. I just need to adapt it here.

zs6buj, 2021-02-28:
@pan-sattan:

From your images I can't see where you pick up the F.Port telemetry. Could you show me were your wiring brings in the P.Port FrSky telemetry,

pan-sattan, 2021-02-28:
source for esp32 is here: from top s.port,+5V,GND

[image]

zs6buj, 2021-02-28:
ok, got it. Thanks.

zs6buj, 2021-03-03:
I have posted a Beta of V2.16.1 which adds support for F.Port (1 & 2).

I can only test it properly once I get me new ACCESS Rm9/R9

pan-sattan, 2021-03-03:
[image]

zs6buj, 2021-03-03:
Thanks. I see you activated some of the debugging. Try only this one:

#define Debug_FrSky

Make sure your flight computer is powered up, and it would be good if your GPS could see the sky without obstruction.

pan-sattan, 2021-03-03:
[image]

zs6buj, 2021-03-04:
I was able to debug S.Port.

Please could you try V2.16.2 (Beta). You should get something like this:

```
Starting AntTrack_F version:2.16.02
Display support activated: Landscape 
 version:2.16.02
Target Board is ESP32 / Variant is LILYGO® TTGO T-Display ESP32 1.14 inch ST7789 Colour LCD
Mavlink Serial In
Heading_Source = Flight Computer
Serial port rx pin 13 is IDLE_LOW, inverting rx polarity
AutoBaud - sensing rxPin 13
Baud rate detected is 57600 b/s
FrSky S.Port found
FrSky read good!
 FrSky 0x5004 Dist to home=0.0 Rel Alt=30.0
Passthru dialect detected
 FrSky 0x5002 Num sats=12 gpsStatus=3 HDOP=8 gpsAlt=15.2 hdopGood=1 neg=0
 FrSky 0x800 latitude=-25.9462109
 FrSky 0x800 longitude=28.0435505
 FrSky 0x5005 Heading=129.2
 FrSky 0x5004 Dist to home=0.0 Rel Alt=30.0
 FrSky 0x5002 Num sats=12 gpsStatus=3 HDOP=8 gpsAlt=15.2 hdopGood=1 neg=0
 FrSky 0x800 latitude=-25.9462109
GPS good
GPS lock good! Push set-home button anytime to start tracking.
```

pan-sattan, 2021-03-04:
yes, its finally work, I am able to see Heading, which is moving...

[image]

closed 2021-03-28 by zs6buj