zs6buj/AntTracker #31: v2.16.3 not working display

pan-sattan, 2021-03-17:
Hi,

I tested latest stable release and there is not implemented Display option

zs6buj, 2021-03-17:
Hi PS, the Display option now lives with each variant.

pan-sattan, 2021-03-17:
Hi, yes I saw that... but after uploading is my OLED without any pixel with esp32 dev board
Then I tried 2.15.5 and there is display OK

zs6buj, 2021-03-18:
There are several display types. If you are using the SSD1306 insert #define SSD1306_Display, and above the test add #define displaySupport

So around line 250 in config.h  it should look like this:

    #define displaySupport
    #if (defined displaySupport)   // Display type defined with # define displaySupport   
      #define SSD1306_Display         // OLED display type    
      /* Below please choose either Touch pin-pair or Digital pin-pair for display scrolling

I will leave these macros in the code in future, and try to make configuration clearer. Thanks for the feedback.

pan-sattan, 2021-03-18:
in my board definition is that:

[image]

and after modification:

[image]

zs6buj, 2021-03-18:
Can you please modify it the way I showed you above.

pan-sattan, 2021-03-18:
yes, last screen is with your modification, row 243 replaced with:
#define displaySupport #if (defined displaySupport) // Display type defined with # define displaySupport #define SSD1306_Display // OLED display

zs6buj, 2021-03-18:
Ah, I see.

Pan you must still include the lines below

      /* Below please choose either Touch pin-pair or Digital pin-pair for display scrolling
       *  Pin == 99 means the pin-pair is not used
       */ 
      #define Pinfo         99        //    Digital pin to toggle information/log page              
      #define Pup           99        // 35 Board Button 1 to scroll the display up
      #define Pdn           99        //  0 Board Button 2 to scroll the display down  
      #define Tinfo         15        // 15 Touch pin to toggle information/log page       
      #define Tup           33        // 33 Touch pin to scroll the display up
      #define Tdn           32        // 32 Touch pin to scroll the display down   
      #define SDA           21        // I2C OLED board
      #define SCL           22        // I2C OLED board
      #define i2cAddr      0x3C       // I2C OLED board
    #endif   
    /*  
      SPI/CS                       Pin 05   For optional TF/SD Card Adapter
      SPI/MOSI                     Pin 23   For optional TF/SD Card Adapter
      SPI/MISO                     Pin 19   For optional TF/SD Card Adapter
      SPI/SCK                      Pin 18   For optional TF/SD Card Adapter  
    */

  #endif

pan-sattan, 2021-03-18:
I had to modify line #252  Tinfo 99
and now it works

[image]

closed 2021-03-28 by zs6buj