zs6buj/AntTracker #32: bugs in code

pan-sattan, 2021-04-19:
Hi,

I found some bugs in 2.17.6:

1/ revers elevation
 if   #define ReverseElevation is used, then this in Servos.ino is wrong - row #81 elt = 180 - el;

2/ for #define HMC5883L and undefine //#define QMC5883L that message will appear:

[image]

looks like there is not rule for HMC5883L compass (I am using it)

zs6buj, 2021-04-19:
Thanks for the feedback pan-sattan.

Fixed. V 2.17.7 

Please try it.

pan-sattan, 2021-04-19:
thx, almost perfect... for debug:

[image]

closed 2021-05-10 by zs6buj