zs6buj/AntTracker #33: wifi

pan-sattan, 2021-04-20:
on my esp32 dev board, when I'm debugging wifi, system will not recognize command at 2.17.7 AntTracker.ino #337 WiFi.onEvent(WiFiEventHandler);

and is there some wiki how it should work? because I cannot see any wifi broadcasting if I am using wifi mode 1 - AP even mode 3
maybe some settings...?

zs6buj, 2021-04-21:
Pan, rather use the debug settings in the IDE:

[image]

and try this app on you phone or tablet:

https://play.google.com/store/apps/details?id=com.sandersoft.udpmonitor&gl=ZA

The trick with UDP is the sending port number must match to the receiving port number. II guess you know for UDP broadcast you put 0x255 in the last byte of the sending IP and the message will go to every host on the segment. **But match the ports.**

Unfortunately I will be pretty busy for the next few days as my big house move is on Friday.

K1rdro, 2021-05-10:
I think this may be the same issue I'm seeing. If you #define Debug_WiFi, then the compiler throws an error.
````
error: 'WiFiEventHandler' was not declared in this scope
     WiFi.onEvent(WiFiEventHandler);

...(multiple lines of  arduino doing its thing)...

exit status 1
'WiFiEventHandler' was not declared in this scope
````