zs6buj/AntTracker #34: tracker is not working

pan-sattan, 2021-06-03:
finally had little time for this project and go out... but, connect all, wat for GPS fix and tracker just sitting on the same place and dont turning around... duno why...

this is from debug from my bench at home
[Tracker_LOG.txt](https://github.com/zs6buj/AntTracker/files/6591858/Tracker_LOG.txt)
and my config
[config.txt](https://github.com/zs6buj/AntTracker/files/6591866/config.txt)
using R9 + R9mm reciever, F.Port fw; here on R9 module is hooked up HC05 master bluetooth module (baud 57600)
X9D+ with latest OpenTX fw
KWAD has latest betaflight version 
and in Tracker I am using version 2.17.8; on pin27 is hooked up HC06 slave bluetooth module with same baud rate like HC05

looks like all telemetry data are comming in between RX and tracker,
compass debug is able to show me proper angle of tracker, servos are moving properly with servo debug

zs6buj, 2021-06-03:
Hi pan-sattan

Something is wrong with the data. For example, numsats can never be 71.  I assume you know S.Port telemetry is inverted, so the HC05 will nor simply read it correctly. You must invert the signal before your feed it into the HC05, and then the same when it comes out of the HC06.

If you have another ESP32 or ESP8266, I have a utility on my Github which will read S.Port out of RM9 and send UDP telemetry. Then setup Tracker to read FrSky UDP.

pan-sattan, 2021-06-03:
sure I forgot to mention it... on hc05 there is an inverter, but I am not sure if I already implement it to the hc06 side... I have to check it later, thanks for hint

zs6buj, 2021-06-03:
Actually it's not really necessary on the tracker side because it will accept either regular or inverted. I see in the log it found regular.

But we still seem to have a data quality problem.

pan-sattan, 2021-06-03:
ok, maybe it will not recognize f.port protocol?

zs6buj, 2021-06-03:
That is interesting! It detected S.Port, but you say your setting is F.Port?

zs6buj, 2021-06-03:
We can try turn off auto detect. I'll look at it tomorrow.

pan-sattan, 2021-06-03:
I think, that the reason is bluetooth data transfer, because before when it was wired from R9 s.port it detect f.port properly

zs6buj, 2021-06-04:
Hmm. OK. I suggest you try the ESP32 BT input.

pan-sattan, 2021-06-19:
another esp32 is on the way...
but maybe... is possible to disable auto check for protocol and set it manualy?

zs6buj, 2021-06-20:
Done. Could you try v2.17.10 please.

pan-sattan, 2021-06-20:
I can see only some debug data if I activate 
#define Debug_FrPort_Stream

[F_Port_debug.txt](https://github.com/zs6buj/AntTracker/files/6682224/F_Port_debug.txt)

pan-sattan, 2021-06-20:
even baud I cant debug... it works only with auto protocol mode

zs6buj, 2021-06-20:
Yes, if you define protocol,  then I already know the baud, so I skip autobaud.

zs6buj, 2021-06-20:
But I could retain autobaud if you wish.

pan-sattan, 2021-06-20:
I cannot decide if it works now... on my bench
maybe if fport messages could be translated like frsky...?

zs6buj, 2021-06-20:
Have you implemented the status LED?

pan-sattan, 2021-06-20:
no, but there is still some room to implement it... and I can connect OLED display too

zs6buj, 2021-06-21:
I'm going to close this issue if that is ok with you. You can open a new ticket with pleasure if you need more help. :)

closed 2021-06-21 by zs6buj