zs6buj/AntTracker #35: Compilation issues while haivng heading source 4=Trackerbox_GPS_And_Compass set

esterlinpol, 2021-06-12:
Hello, I'm getting the following compilation issue when I set the :

`#define Heading_Source  4     // 1=Flight_GPS, 2=Flight_Computer, 3=Trackerbox_Compass  4=Trackerbox_GPS_And_Compass`

Getting the following error message in the arduino ide:

`C:\Users\Esterlin\Downloads\AntTracker-master 1\AntTracker-master\Beta\AntTrack\GPS.ino: In function 'void getTrackerboxLocation()':
GPS:147:9: error: 'tboxGPS' was not declared in this scope
     if (tboxGPS.encode(gpsSerial.read())) {
         ^
GPS:168:9: error: 'dt' was not declared in this scope
         dt.yyyy = yyyy - 1900;  // epoch year
         ^
GPS:175:32: error: 'getEpoch' was not declared in this scope
         epochSync = getEpoch(dt);
                                ^
exit status 1
'getBaud' was not declared in this scope`

If I set the heading source to 3, Tracker box compass, the program compiles and works properly, 

This is happening in both stable and beta versions. 

I'm I missing something?

zs6buj, 2021-06-12:
Hi esterlinpol

It looks like you might have a version or library problem. I just clean compiled 2.17.8 with Heading_Source 4.

PreviouslyI changed to the TinyGPSPlus library to improve a few things, and tboxGPS is the instantiated GPS object.

Can you confirm your version, and also make sure you have copied the TinyGPSPlus folder from my library folder on GitHub to your active library folder.

esterlinpol, 2021-06-12:
Hello Sir, I missed the libraries files included in the source code,

So now, 

I have removed all the libraries I had installed, and copied yours to my libraries folder, I am getting the following error now:

`Arduino: 1.8.13 (Windows 10), Board: "FireBeetle-ESP32, 80MHz, 921600"

C:\Users\Esterlin\Documents\Arduino\AntTrack\AntTrack.ino: In function 'void setup()':

AntTrack:533:34: error: 'getBaud' was not declared in this scope

       gpsBaud = getBaud(gps_rxPin);

                                  ^

C:\Users\Esterlin\Documents\Arduino\AntTrack\GPS.ino: At global scope:

GPS:138:1: error: 'TinyGPSPlus' does not name a type

 TinyGPSPlus tboxGPS;

 ^

C:\Users\Esterlin\Documents\Arduino\AntTrack\GPS.ino: In function 'void getTrackerboxLocation()':

GPS:147:9: error: 'tboxGPS' was not declared in this scope

     if (tboxGPS.encode(gpsSerial.read())) {

         ^

GPS:168:9: error: 'dt' was not declared in this scope

         dt.yyyy = yyyy - 1900;  // epoch year

         ^

GPS:175:32: error: 'getEpoch' was not declared in this scope

         epochSync = getEpoch(dt);

                                ^

exit status 1

'getBaud' was not declared in this scop`

More information:

- Using the latest version 2.17.8, also tried the beta.
- #define Telemetry_In  4
- #define Heading_Source  4
- #define HMC5883L
- #define ESP32_Variant     1
- #define displaySupport 
-

zs6buj, 2021-06-12:
If you open the GetProtocol tab, do you see getBaud(   on line 182?

[image]

esterlinpol, 2021-06-12:
Yes, it looks like:

// **********************************************************
uint32_t getBaud(uint8_t pin) {
  Log.print("autoBaud - sensing pin "); Log.println(pin);
 // Log.printf("autoBaud - sensing pin %2d \n", pin );
  uint8_t i = 0;
  uint8_t col = 0;
  pinMode(pin, INPUT);       
  digitalWrite (pin, HIGH); // for noise reduction ?

zs6buj, 2021-06-12:
Could you please post your config.h

esterlinpol, 2021-06-12:
Here it is
[config.h.txt](https://github.com/zs6buj/AntTracker/files/6642339/config.h.txt)

zs6buj, 2021-06-12:
Ok, with your config.h I can replicate the compiler error. I'll be able to post a fix in the morning.

zs6buj, 2021-06-13:
I've fix a few syntax error that cropped up with your config.h.  |Please try v2.17.9. My fault. I seem to have lost some work somewhere while moving house. I've also renewed Adafruit_HMC5883_Unified in my libraries folder, so you might want to copy that over.

esterlinpol, 2021-06-13:
Thanks for the amazing help and fast response, everything is working now.

I'll continue working on building the tracker and post my results in the rcforums thread.

closed 2021-06-13 by esterlinpol