zs6buj/AntTracker #36: frsky D

mariusdp, 2021-08-04:
Sorry 1st if I am in the wrong.
How do I set to use Frsky D? The setup I have is matek f411 (inav3.0.1) with Frsky D and TX16s. The plan would be to forward the telemetry from tx16s through one of the uarts (telemetry mirror). Any help would be appreciated. Thank you.

zs6buj, 2021-08-05:
Hey Marius

Happy to help. I originally programmed for AntTraver to include FrSkyD, but I have not checked it in a long time. If you are serious about using it I could review the code. I'm not sure how to test it, other than maybe try the Mavlink FrSky D serial output.

mariusdp, 2021-08-05:
Hi Eric,

At the moment (and foreseeable future) I have only Frsky D (D4R-II). Inav is set up to send telemetry to tx16s with inav telemetry lua installed and works great. The idea is to use telemetry mirror on one of the UART ports available on tx16s to connect to antenna tracker. Or somehow, if even possible, to use and esp32 or something else to convert frsky D to S port. I would gladly experiment and assist in any way I can. This would be my first tracker since I always was on budget and can go into buying one.  I do have plenty dev bards (esp, stm, arduino, raspi...) and 3d printer to print, compile and test... Also, I would problably test on a X9D plus first... don't want to chance damaging tx16s. Thank you.

zs6buj, 2021-08-09:
Hi Marius,

For a tracker, all you need is home and current (gps co-ordinates and altitude), and the orientation of the tracker box relative to physical world compass points. This could come from box compass or gps, or from aligning the box with the plane (and its compass).

So FkSky D is enough, but if you want more, like for yaapu's lua gui, then you could convert from D to passthru if you choose. Maybe I could cut and paste some code for you, and you could handle the debugging?

mariusdp, 2021-08-09:
Hi,

Just tracking... Yaapu works fine on tx16s so no issue there. The issue is that in current code can't relay D protocol from th16s to tracker. So any help to make that would be grate. I do not intend to relay anything out of the tracker. The only thing I'm looking is to have the antenna tracker working and maybe see when lock in good on oled so makes life easier. Have oleds, i2c compasses and gps... Thanks

zs6buj, 2021-08-09:
I have not looked too closely at the TX16, but you sound confident about an extra uart being available. Does it present the  uart tx  pin to the outside world? You could add an ESP32 board and stream UDP or BT to the tracker.

And I assume you are using opentx?

mariusdp, 2021-08-09:
Yes. It has 2 UART ports available to outside world and the option to output the received telemetry to them. Same like X9D plus... Diference is that on TX16s Uart 1 has "- 5V TX RX" and UART 2 same pls RST pin.

zs6buj, 2021-08-09:
Ah, ok. You could clone and flash this firmware for ESP32 

[](https://github.com/zs6buj/FrSky-Passthru-To-WiFi-and-or-Bluetooth)

It will forward bt (or udp). Remember to cross over uart tx/rx to esp32 rx/tx.

Then in Tracker, 

// Choose one only of these input channels 
// How does telemetry enter the tracker?
//#define Telemetry_In  0    // Serial Port (default) - all protocols        
//#define Telemetry_In  1    // Mavlink BlueTooth Classic- ESP32 
//#define Telemetry_In  2    // Mavlink WiFi - ESP only
//#define Telemetry_In  3    // FrSky UDP - ESP only
#define Telemetry_In  4    // FrSky BT classic - ESP32 only

mariusdp, 2021-08-09:
ok. 10x. Will give it a go when i get home....

mariusdp, 2021-08-09:
Hi, 
so select "FrSky_Port_Type 3" for this to work with D yes?
and "#define FrSky_IO     1     // BT" for I/O

Sorry for so many Q i just want it to get it right.... 

// Choose only one setting for FrSky Port Type 
//#define FrSky_Port_Type 0   // No FrSky Port support needed. Now I'm a "Mavlink Switch"
//#define FrSky_Port_Type 1   // F.Port v1
//#define FrSky_Port_Type 2   // F.Port v2 FrSky ISRM/ACCESS capable transmitters and receivers only
#define FrSky_Port_Type 3   // S.Port / legacy
//#define FrSky_Port_Type 4   // AUTO detect

zs6buj, 2021-08-09:
Yes, just select sport

zs6buj, 2021-11-04:
Please open again if I can help

closed 2021-11-04 by zs6buj