zs6buj/AntTracker #38: Compilation issues for STM32F108C8 on ArduinoIDE

TTN-, 2021-12-18:
1. I installed the libraries by copying them into user\Documents\Arduino\libraries
2. installed STM32DUINO library as per here https://github.com/stm32duino/wiki/wiki/Getting-Started
3. set my configuration in `config.h` as required, servo uS, no compass, bluepill STM32F103C8 etc..
4. set arduino ide board type:
[image]

5. get "no board defined" error: `AntTrack.ino:126:
config.h:210:4: error: #error "No board type defined!"` (Was I supposed to declare that or does the IDE do that bit?)
6. manually add `#define __BluePill_F103C8__` in config.h line 178
7. error during compilation:
```
D:\User_HDD\Google Drive\Projects\RC gear\STM32 antenna tracker\AntTracker-master\Source\AntTrack\EEPROM.ino: In function 'int32_t EEPROMReadlong(uint16_t)':
EEPROM:156:27: error: no matching function for call to 'EEPROMClass::read(uint16_t&, uint16_t*)'
  156 |     EEPROM.read(addr, &two);
      |                           ^
In file included from D:\User_HDD\Google Drive\Projects\RC gear\STM32 antenna tracker\AntTracker-master\Source\AntTrack\EEPROM.ino:94:
C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\libraries\EEPROM\src/EEPROM.h:201:11: note: candidate: 'uint8_t EEPROMClass::read(int)'
  201 |   uint8_t read(int idx)
      |           ^~~~
C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\libraries\EEPROM\src/EEPROM.h:201:11: note:   candidate expects 1 argument, 2 provided
EEPROM:157:29: error: no matching function for call to 'EEPROMClass::read(int, uint16_t*)'
  157 |     EEPROM.read(addr+1, &one);
      |                             ^
In file included from D:\User_HDD\Google Drive\Projects\RC gear\STM32 antenna tracker\AntTracker-master\Source\AntTrack\EEPROM.ino:94:
C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\libraries\EEPROM\src/EEPROM.h:201:11: note: candidate: 'uint8_t EEPROMClass::read(int)'
  201 |   uint8_t read(int idx)
      |           ^~~~
C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\libraries\EEPROM\src/EEPROM.h:201:11: note:   candidate expects 1 argument, 2 provided
Multiple libraries were found for "Wire.h"
 Used: C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\libraries\Wire
 Not used: D:\User_HDD\Documents\Arduino\libraries\Wire
Multiple libraries were found for "EEPROM.h"
 Used: C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\libraries\EEPROM
 Not used: D:\User_HDD\Documents\Arduino\libraries\EEPROM
Multiple libraries were found for "Servo.h"
 Used: C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\libraries\Servo
 Not used: D:\Programs\Arduino\libraries\Servo
 Not used: D:\User_HDD\Documents\Arduino\libraries\Servo
Multiple libraries were found for "SPI.h"
 Used: C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.1.0\libraries\SPI
 Not used: D:\User_HDD\Documents\Arduino\libraries\SPI
exit status 1
no matching function for call to 'EEPROMClass::read(uint16_t&, uint16_t*)'
```

Windows 10 x64
Arduino IDE 1.8.18

Has anyone seen this before?

zs6buj, 2021-12-30:
Sorry for the slow response. I've been away for Christmas.

May I suggest you follow [these instructions](https://www.instructables.com/Getting-Started-With-Stm32-Using-Arduino-IDE/) for now, with this link in Preference:

"http://dan.drown.org/stm32duino/package_STM32duino_index.json"

What's happening here is that my auto board identification code looks for the "__BluePill_F103C8__" or "MCU_STM32F103RB" macro, and the newer development environment is using a different identifier.

```
#elif defined (__BluePill_F103C8__) ||  defined (MCU_STM32F103RB)
  #define Target_Board   1      // Blue Pill STM32F103C8  
  #define STM32F103C8
```

zs6buj, 2021-12-30:
Alternatively, temporarily insert

`#define STM32F103C8`

in place of 

`  #error "No board type defined!"`

TTN-, 2021-12-31:
Thanks Eric. I've installed that and set the board accordingly.

Btw also:
```#include <ardupilotmega/mavlink.h>
#include <ardupilotmega/ardupilotmega.h>```
on line 128 at `AntTracker.ino` isn't finding the files so I've just set them manually, forgot to mention that earlier.

I'm getting a few compiler issues related to those includes now, so I might be including the wrong files. I manually set them to:
```#include "D:\NetworkDrive\Documents\Arduino\libraries\GCS_MAVLink\include\mavlink\v2.0\ardupilotmega\mavlink.h"
#include "D:\NetworkDrive\Documents\Arduino\libraries\GCS_MAVLink\include\mavlink\v2.0\ardupilotmega\ardupilotmega.h"```

Are these the files that are intended to be included here?

TTN-, 2021-12-31:
Do I copy all the library files to `User/Document/Arduino/libraries` ?

TTN-, 2021-12-31:
I've managed to get it to compile by doing the following:
Move all arduino libraries to somewhere else temporarily so we don't get conflicts

in Documents/Arduino/libraries/ have these:
```
TinyGPSPLUS/TinyGPS++.cpp
TinyGPSPLUS/TinyGPS++.h
``
And copy everything from `PixhawkArduinoMAVLink\src\` to a folder in your library, give it a name, in my case I just put it in a folder called `mavlink2`

Now compiles :)

zs6buj, 2021-12-31:
Good to hear. 

Regarding Ardupilot libraries, you will find a link to the latest libs in the tracker library folder. 

APM libs may be found [here](https://github.com/mavlink/c_library_v2)

zs6buj, 2023-03-28:
Stale

closed 2023-03-28 by zs6buj