zs6buj/AntTracker #39: Porting project to PlatformIO project, as opposed to Arduino IDE

TTN-, 2021-12-18:
Would you be open to having this project converted into a PlatformIO project? I'm thinking of doing a fork with exactly that.

https://platformio.org/

It's an extension for Visual Studio Code which makes life a lot easier as a programmer, while still utilizing the arduino framework. Also supports proper debugging on most boards.

TTN-, 2021-12-19:
Ok so I've made a start, and I can see there has been a LOT of work gone into making this project what it is today.
There's also a fair bit of work in tidying it up splitting the headers into .cpp and .h files and just some general house keeping.

TTN-, 2021-12-19:
https://github.com/TTN-/AntTracker

zs6buj, 2021-12-19:
Good morning TTN.  Yes of course, that would be a nice step forward. Let me know if you have any questions. I'm working from my mobile for the next two weeks.

TTN-, 2021-12-19:
Good morning to you too ZS6buj,

Haha, I know the feeling, got side tracked into this! Alright well I'm working on it, might be a few days till I come out with something functional, did you get anywhere with what you had? I'm part way through splitting it all up into headers/cpp files. Is it worth sharing?

zs6buj, 2021-12-19:
Yeah, definitely worth sharing. PIO with included libs is much easier for others to build and flash IMO, and easier to debug.

I got quite far, but then hit some snags with dependencies. Then sidetracked, :)

zs6buj, 2023-03-28:
no progress on this. Sorry I've been quite busy elsewhere

closed 2023-03-28 by zs6buj