zs6buj/AntTracker #40: Support for Stepper Motors

devanesanandrews, 2021-12-28:
Hi, 

The antenna tracker is amazing and works well.

Will there be an support for Stepper motors as they have better holding torque for bigger antennas.

Thank you

zs6buj, 2021-12-28:
Hello and thank you for the feedback. I don't have much experience with steppers, but I could give it a try. In your experience are they fast/responsive enough?

andykirk50, 2021-12-28:
im also very interested in this,
yes they are faster and more responsive.

zs6buj, 2021-12-28:
Ok, cool. Could you recommend a stepper I can get for testing? 

I'm also planning to auto-tune a _remote_ magnetic loop antenna with a small stepper.  (Remote, as in on the roof or somewhere like that). It needs to be geared to _very finely_ adjust a butterfly variable capacitor. I will sample VSWR while transmitting a low power signal into the antenna, and then adjust for lowest reflected power.  Any ideas where to find the geared stepper I need?

andykirk50, 2021-12-28:
nema 23 steppers are the best, used in 3d printers, however you will need a stepper driver
https://www.youtube.com/watch?v=1RcEx5rpNrY

wsqzdz, 2021-12-28:
I have achieved PWM by using 328p Drive two closed-loop stepper motors, but there are still some problems. The system becomes complex. I tried to modify the code of anttracker, but I didn't test it.

zs6buj, 2021-12-28:
Lol. I did a quick stepper refresher after the previous posts. Thanks for the video. This should be doable. Maybe I can scrounge some older printer parts.

zs6buj, 2021-12-28:
> I have achieved PWM by using 328p Drive two closed-loop stepper motors, but there are still some problems. The system becomes complex. I tried to modify the code of anttracker, but I didn't test it.

May I ask which library you used?

andykirk50, 2021-12-28:
id happily donate parts if there was a way to get them to you.

wsqzdz, 2021-12-28:
> 

I use accelstepper library, using 328p to convert PWM into driving pulse. In order to avoid losing steps, I used a closed-loop driver

wsqzdz, 2021-12-28:
I use  library   accelstepper , using 328p to convert PWM into driving pulse. In order to avoid losing steps, I used a closed-loop driver

devanesanandrews, 2021-12-28:
> Hello and thank you for the feedback. I don't have much experience with steppers, but I could give it a try. In your experience are they fast/responsive enough?

Yes they are pretty good also its based on the Micro stepping and the torque used.

I would recommend to go with a proper stepper motor driver Like Tb6600  (https://www.amazon.co.uk/ARCELI-TB6600-Stepper-Controller-controller/dp/B07J2S7198/ref=sr_1_4?crid=1YOWVM2GIK4YK&keywords=tb6600&qid=1640707688&sprefix=tb660%2Caps%2C192&sr=8-4)or 

the ones used in the 3d printers A4988 (https://www.amazon.co.uk/Longruner-Stepstick-Stepper-Driver-Printer/dp/B071P41ZBW/ref=sr_1_1_sspa?crid=31Y1TK8CO2UG8&keywords=3d+printer+stepper+driver&qid=1640707732&sprefix=3d+printer+stepper+dribe%2Caps%2C197&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEzOFJDQjRFN1MwWjVFJmVuY3J5cHRlZElkPUEwMjc4Nzk3MjBFM09VNzdPNlBHNiZlbmNyeXB0ZWRBZElkPUEwMzI0MDU2T1BYNzhVWlFFRU9aJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==)

I have been using tb6600 with my diy cnc and i should say its works pretty good.

Regarding stepper motor antenna tracker here is a 3d printed version from @kolins-cz

https://www.youtube.com/watch?v=I_QwMFngyWA

STL Files : https://github.com/kolins-cz/antenna-tracker-mechanics

zs6buj, 2021-12-28:
All good information.  Also thanks for the offer of parts. First let me look around on the web and see if I can find a suitable "mule" stepper and driver for testing, with reasonably quick shipping.

zs6buj, 2021-12-30:
So I ordered two of [these](https://www.aliexpress.com/snapshot/0.html?spm=a2g0o.9042647.0.0.11ac4c4dWUbgil&orderId=3014871451948155&productId=1005003172428274) and two of [these](https://www.aliexpress.com/snapshot/0.html?spm=a2g0o.9042647.0.0.33cd4c4dxC1mJ8&orderId=3014871451968155&productId=1000001730872), and I'll start looking at the code while waiting for delivery. Shipping through our post office can be quite slow, unfortunately.

devanesanandrews, 2022-01-14:
@zs6buj  That was nice very curios to see the antenna tracker with stepper motors.

I believe these will be of some help.
Controlling stepper motors with RC PWM signal.

https://www.youtube.com/watch?v=0kXlILWAYKg

https://www.youtube.com/watch?v=L84smchueYM

https://gitlab.com/cswiger/AccelStepper_EasyDriver_RC/

zs6buj, 2022-01-14:
Thank you. I'm waiting impatiently for the hardware. :(

zs6buj, 2023-03-28:
Hey Deva, I've been busy elsewhere as you know. I have the steppers now, so when I get some time I can take this further. Closing for now.

closed 2023-03-28 by zs6buj