zs6buj/AntTracker #41: 'HUD_ARROW_OFFSET' was not declared in this scope

Cyberdemon-23, 2022-01-07:
S E L E C T   E S P   B O A R D   V A R I A N T   ===================

#define ESP32_Variant     7

Utilities:1440:11: error: 'HUD_ARROW_OFFSET' was not declared in this scope

       if (HUD_ARROW_OFFSET == 999) {

           ^

zs6buj, 2022-01-08:
Hmm..  I just clean compiled with 

`#define ESP32_Variant     7    // ESP32 Dev Module with ILI9341 2.8" colour TFT SPI 240x320`

Could you try with my config.h, and then compare with yours?

https://drive.google.com/file/d/12Qgt1DqaaUvSx4ZDKbQR8JpJOxqkhh_P/view?usp=sharing

It would be best if you could use the latest source, so we are on the same page.

Cyberdemon-23, 2022-01-08:
Downloaded the latest source, your config file - is the same error. Where is HUD_ARROW_OFFSET written? Maybe the wrong library is being used?

zs6buj, 2022-01-09:
It would be best if you could use the latest source, so we are on the same page. **V2.19.4**.

HUD is only possible on the large  ILI9341_Display. So in the latest version, Utilities tab, line1249 you will find a block of code to populate/paint the ILI9341_Display pages

` #if  (defined ILI9341_Display)`

and HUD_ARROW_OFFSET is used in this conditional block.

So, its use is predicated on this condition in config.h line 508

`  #if (ESP32_Variant == 7)          // ESP32 Dev Module with ILI9341 2.8" colour TFT SPI 240x320`

In that conditional block we define

`  #define ILI9341_Display         // ILI9341 2.8" COLOUR TFT SPI 240x320 V1.2  `

and then in Utilities, line 1249 we have the conditional block mentioned above

` #if  (defined ILI9341_Display)`

zs6buj, 2022-01-09:
I see I added that code in V2.18.0. I can check it out a little later.

zs6buj, 2022-01-09:
When I checked it this morning, it did not compile,  my apologies.  (told myself to go sit in the corner)

It's fixed in V2.19.05.

Cyberdemon-23, 2022-01-09:
is actually compiling now, but the screen is not displaying anything. See if everything is properly connected
[image]

zs6buj, 2022-01-10:
Hmm. I see you have marked the LED pin with 5v, so you know that both Vcc and LED must be powered.

Could you try these pins:

    #define ILI9341_Display         // ILI9341 2.8" COLOUR TFT SPI 240x320 V1.2  
    #define SCLK          18        // blue wire on my test setup
    #define MOSI          23        // yellow wire
    #define CS            25        // white wire
    #define DC            22        // green wire
    #define RST           21        // brown wire

zs6buj, 2022-01-10:
I just tested it with the original pins, and that also works:

    #define ILI9341_Display         // ILI9341 2.8" COLOUR TFT SPI 240x320 V1.2  
    #define SCLK          18        // blue wire on my test setup
    #define MOSI          23        // yellow wire
    #define CS            25        // white wire
    #define DC            26        // green wire  

    #define RST           27        // brown wire
    // LED=3.3V,  Vcc=5v,  Gnd 
    // MISO                not used by Adafruit     

[image]

Could you double check your wiring?

zs6buj, 2022-01-10:
[image]

zs6buj, 2022-01-10:
If you could post the serial monitor log it might be helpful to see if there are any startup anomalies.

Cyberdemon-23, 2022-01-10:
`13:03:11.540 -> 
13:03:11.540 -> Starting AntTrack version:2.19.05
13:03:12.254 -> Display support activated: Landscape 
13:03:12.254 -> 240x320  text_size=2  char_w_px=12  char_h_px=16  scr_h_ch=15  scr_w_ch=26
13:03:12.348 -> Target Board is Target Board = 3  ESP32 / Variant is Dev Module with ILI9341 2.8in COLOUR TFT SPI
13:03:12.348 -> Serial Telemetry In
13:03:12.348 -> Selected protocol is Mavlink 1
13:03:12.348 -> headingSource = 2 FC Compass
13:03:12.348 -> Servo slowdown factor is 10ms per degree of rotation
13:03:13.275 -> Testing Servos
`

Cyberdemon-23, 2022-01-10:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
E (147) psram: PSRAM ID read error: 0xffffffff

Starting AntTrack version:2.19.05
Display support activated: Landscape 
240x320  text_size=2  char_w_px=12  char_h_px=16  scr_h_ch=15  scr_w_ch=26
Target Board is Target Board = 3  ESP32 / Variant is Dev Module with ILI9341 2.8in COLOUR TFT SPI
Serial Telemetry In
Selected protocol is Mavlink 1
headingSource = 2 FC Compass
Servo slowdown factor is 10ms per degree of rotation
Testing Servos

```

Cyberdemon-23, 2022-01-10:
[image]

zs6buj, 2022-01-10:
"E (147) psram: PSRAM ID read error: 0xffffffff"

This looks a bit suspect. I googled it and got this:

"
One of my ESP32-Cam boards has this issue too.
Is the most likely explanation hardware failure of the PSRAM chip?"

The mpu should not hang during "Testing Servos", but you can disable the servo test in config.h

config.h line 109 

`//#define Test_Servos      // Move servos through their limits`

zs6buj, 2022-01-10:
My bootup looks like this:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

Starting AntTrack version:2.19.05
Display support activated: Landscape 
240x320  text_size=2  char_w_px=12  char_h_px=16  scr_h_ch=15  scr_w_ch=26
Target Board is Target Board = 3  ESP32 / Variant is Dev Module with ILI9341 2.8in COLOUR TFT SPI
Serial Telemetry In
Selected protocol is Mavlink 1
headingSource = 2 FC Compass
Servo slowdown factor is 10ms per degree of rotation
Testing Servos

Cyberdemon-23, 2022-01-10:
Uploaded to another board - this error disappeared, but the screen remained white.
I change
`Adafruit_ILI9341 display = Adafruit_ILI9341 (CS, DC, RST); // LED = 3.3V, Vcc = 5v, Gnd`
on the
`Adafruit_ILI9341 display = Adafruit_ILI9341 (CS, DC, MOSI, SCLK, RST);`
now everything works!

zs6buj, 2022-01-11:
I'm guessing that I'm one version behind of the Adafruit ILI9341 library.

[image]

zs6buj, 2022-01-11:
I updated to 1.5.10, but observing the two versions of the object, 

` Adafruit_ILI9341 display = Adafruit_ILI9341(CS, DC, RST);` is far faster. Refresh of the display is very fast, but with 

`Adafruit_ILI9341 display = Adafruit_ILI9341 (CS, DC, MOSI, SCLK, RST);`

refresh is slow. Too slow to refresh to HUD fast enough. I think maybe this version renders the display in main RAM (on the stack), and it is slower. Guessing here.

zs6buj, 2022-01-11:
First is hardware SPI, and the second is software SPI

closed 2022-01-11 by Cyberdemon-23

zs6buj, 2022-01-11:
Cool! My pleasure CyberD.