zs6buj/AntTracker #42: compilation and missing smartport telemetry sensors

shro70, 2022-01-18:
Hello.

When i try to set  #define Heading_Source  3     // 3=Trackerbox_Compass 
i get a compilre error"'uint_16' was not declared in this scope at line 92 in compass :  fHeading = (float)wrap360((uint_16)val);"

The other issue is i only get a few telemetry sensors .
I use Inav + radiomatster TX16s +R9M ACCST in Fport with an inverted HC05 pluged in the back of the R9M. Connection is ok . the esp32 receive telemtry but only GPS, Heading and RSSI.
This setup work great on another tracker (U360gts)
   
[image]

Can you help me ?

zs6buj, 2022-01-19:
Hello, yes I fixed the typo and uploaded v2.19.5b

The tracker only every needs the location of the craft and its altitude, so I don't bother decoding most of the other message types.

shro70, 2022-01-19:
Thanks for the quick answer and fix.
Which altitude sensor is used ? Baro or gps alt because I only get 0m .

zs6buj, 2022-01-19:
0x820 for alritude. You are showing alt = 233 above?

shro70, 2022-01-19:
Ah you're right. That's my gps alt but the screen only show 0.

zs6buj, 2022-01-19:
Ah, OK. I'll check that when I get home.

zs6buj, 2022-01-19:
Ok, I'm looking at the code.  I determine altitude above ground/field/home by saving absolute altitude (above mean sea level) when "home" is set. Home can be set when motors are armed if this option is set

`#if defined SET_HOME_AT_ARM_TIME`  

or by button push. From you log I can see that home has not yet been established.

[image]

So altitude above field will be 0.

As for rssi. number of sats, climb and ground speed, I've added some code for  iNav hud display, but might need a little more work.  I need iNav sensor numbers for speed and climb.

zs6buj, 2022-01-19:
I don't have any iNav kit here unfortunately, so I need links to documents that list the sensors. I have some docs, but they seem out of date.

zs6buj, 2022-01-19:
I found this

```
#define GPS_SPEED_FIRST_ID        0x0830
#define GPS_SPEED_LAST_ID         0x083f

```

shro70, 2022-01-19:
https://github.com/iNavFlight/inav/blob/master/src/main/telemetry/smartport.c

zs6buj, 2022-01-19:
Thanks, I need to do a bit more coding. You could try v 2.19.06 in the meantime. I suggest you use 

#idefine SET_HOME_AT_ARM_TIME

shro70, 2022-01-19:
I tested the 2.19.06 and it solved my issues.
I see you added sat number too .

I think I noticed a minor bug : 
#Define servo slowdown seem to only affect azimuth not elevation.

zs6buj, 2022-01-20:
```
"I think I noticed a minor bug :
#Define servo slowdown seem to only affect azimuth not elevation."
```
Could you double check. It looks ok here.

zs6buj, 2022-01-20:
v2.19.07 added speed, pitch and roll.  Does iNav have a climb (rate of)  message

shro70, 2022-01-20:
Nice. Vertical speed is 
FSSP_DATAID_VARIO = 0x0110

zs6buj, 2022-01-20:
Done. v2.19.07b

I've assumed unit = m *10, so /10 to get m/s

shro70, 2022-01-20:
> ```
> "I think I noticed a minor bug :
> #Define servo slowdown seem to only affect azimuth not elevation."
> ```
> 
> Could you double check. It looks ok here.

I only tested my servo with the #test servo and only azimuth is slowed down.
I tried 5 10 15 and elevation servo still move at full speed while azimut is effectively slowed down.
I only did a bench test no real flying

zs6buj, 2023-03-28:
@shro70: Sorry, I've been away for a while. If you still need help with this please reopen the issue.

closed 2023-03-28 by zs6buj