zs6buj/AntTracker #43: 360 servo

andykirk50, 2022-01-21:
can anyone help please
ive set up my tracker on V2.29.05, everything seems to work fine other than the AZ movement.
i have a 360 servo but it only moves 180. ive checked the output of the PWM and ran servo test to see the output.
when the tracker first loads its set to point 180 (PWM value of 3736. then in the servo test i run it as below, when it tries to test 360 it doesnt move and reads the same as 180 then when it tests 0 degrees it moves down to pwn 769.
please correct me if im wrong but setting the start point at 180 it should have a mid point on the pwm then move left and right from that when moving to 0 and 360

when setting the servo movement to 180 i get pwm 3719 max, 2253 mid and 802 low, so seems to be an issue when 360 is configured

void TestServos() {
delay(8000);
moveServos(180 (pwm 3736)
delay(8000);
moveServos(180 (pwm 3736)
delay(8000);
moveServos(359 (pwm 3736)
delay(8000);
moveServos(359 (pwm 3736)
delay(8000);
moveServos(0 (pwm 769 )
delay(8000);
moveServos(0 (pwm 769)
delay(8000);
moveServos(azStart, elStart);

#define Test_Servos // Move servos through their limits, then try box_hdg every 45 degrees of circle

#define Az_Servo_360 // Means the azimuth servo can point in a 360 deg circle, elevation servo 90 deg
// Default (comment out #define above) is 180 deg azimuth and flip over 180 deg elevation

#define azStart 180 // 0 deg = left, 90 deg = straight ahead, 180 deg = right, 270 deg = behind
#define elStart 45 // 0 = horizontal, 90 = vertical

#define Servo_Slowdown 5 // Default 0 - Try 5, 10.. ms/degree to limit angular velocity

// Set the degree range of the servos here. Do not adjust servo mechanical limits here.
#if defined Az_Servo_360 // 1 x 360, 1 x 90 (or 180) servos
int16_t minAz = 0; // Az lower limit in degrees, left of tracker facing flying field
int16_t maxAz = 359; // Az upper limit in degrees
int16_t minEl = 0; // El lower limit in degrees, horizontal
int16_t maxEl = 90; // El upper limit in degrees, straight up
[AntTrack_360AZ.zip](https://github.com/zs6buj/AntTracker/files/7914544/AntTrack_360AZ.zip)

zs6buj, 2022-01-22:
Hi Andy

[Here](https://drive.google.com/file/d/12RTa0NoTNfYy--MgXbfH6GACVM3lm2d3/view?usp=sharing) is a hack of one of the ESP32_Servo examples. I've added your az configuration. You could add a few print statements if you like, and use it for testing your setup. Your range of 400 - 4000 seems a bit high, but not impossible.

andykirk50, 2022-01-22:
Thank you
Ill give it a try.

Have your parts arrived yet for the stepper motors?

zs6buj, 2022-01-23:
The stepper motors have not yet arrived, and unfortunately I have a deadline of 9 February for an overseas trip.

andykirk50, 2022-01-23:
No problem.

Ive ordered a Pololu T825 pwm to step converter which should be here Monday so ill have a play with that and let you know how I get on.

Hope you have a good trip.

sebas1984x, 2022-05-08:
how can i add the 360 hack?

zs6buj, 2022-05-09:
@andykirk50:

I'm back home, but there is no sign of the stepper motors shipped through the SA post office I'm afraid. Maybe someone decided they looked like interesting paper weights.

@sebas1984x:  

Carefully go through the config.h, especially the SERVO SETTINGS section, and configure according to your perceived needs. As questions if you need clarification. Once you have configured, you can test the outcome using selected debug macros:

#define Debug_AzEl
#define Debug_Servos 
#define Test_Servos 

```
//=============================================================================================
//===================================  S E R V O   S E T T I N G S  ===========================
//=============================================================================================

  #define Test_Servos      // Move servos through their limits, then try box_hdg every 45 degrees of circle

  //#define Az_Servo_360   // Means the azimuth servo can point in a 360 deg circle, elevation servo 90 deg
                           // Default (comment out #define above) if 180 deg azimuth and flip over 180 deg elevation 
                           
  #define azStart  90       // 0 deg = left, 90 deg = straight ahead, 180 deg = right, 270 deg = behind
  #define elStart   0       // 0 = horizontal, 90 = vertical

  #define Servo_Slowdown 10  // Default 0 - Try 5, 10..  ms/degree to limit angular velocity

  // Set the degree range of the servos here. Do not adjust servo mechanical limits here.                         
  #if defined Az_Servo_360   // 1 x 360, 1 x 90 (or 180) servos  
    int16_t minAz = 0;          // Az lower limit in degrees, left of tracker facing flying field
    int16_t maxAz = 359;        // Az upper limit in degrees
    int16_t minEl = 0;          // El lower limit in degrees, horizontal 
    int16_t maxEl = 90;         // El upper limit in degrees, straight up
  #else                      // 2 x 180 deg servos
    int16_t minAz = 0;          // Az lower limit in degrees, left of tracker facing flying field
    int16_t maxAz = 180;        // Az upper limit in degrees, right of tracker facing flying field
    int16_t minEl = 0;          // El lower limit in degrees, horizontal and forward
    int16_t maxEl = 180;        // El upper limit in degrees, horizontal and rearward
  #endif 

  // Sometimes the mechanical movement of a servo is reversed due to the orientation of its mounting
  // Its movement may be reversed here to compensate
  #define ReverseAzimuth          // my azimuth servo has a reversed action
  //#define ReverseElevation

  // Default values for SG90 servos; 500 and 2400
  // My 180 deg servos have a PWM range of 700 through 2300 microseconds. Your's may differ. 
  // ADJUST THE MECHANICAL LIMITS OF MOVEMENT OF YOUR SERVOS HERE BELOW

  #if defined STM32F1xx      // my STM32 based tracker has different servos
    uint16_t minAzPWM = 600;   // right (because mine is reversed)
    uint16_t maxAzPWM = 2300;  // left   
    uint16_t minElPWM = 700;   // front 
    uint16_t maxElPWM = 2300;  // back
  #else
    uint16_t minAzPWM = 625;   // right (because mine is reversed)
    uint16_t maxAzPWM = 2235;  // left 
    uint16_t minElPWM = 600;   // front
    uint16_t maxElPWM = 2257;  // back  
  #endif
```

sebas1984x, 2022-05-11:
Thnx, unfortunately i got the wrong 360 servo (continues rotation)

Can i use a 180° servo with 2:1 gear ratio, if so what do i need to change?

Ps. Thnx for your work, love it 👍🏻

zs6buj, 2022-05-12:
Thanks for your kind feedback. The 180 deg azimuth servo plus 180 deg elevation servo will effectively achieve 360 degree all 'round coverage, because the 180 degree elevation servo flips over and behind when need be. You need to imagine the scenario in your mind's eye, or set it up and use the #define Test_Servos

sebas1984x, 2022-05-12:
Yes i know, the thing is it cant flip backwards to much because of a hdzero reciever on the back. Is there a way to use a 1:2 gear ratio? That would be great

sebas1984x, 2022-05-12:
[image]

zs6buj, 2023-03-28:
I assume this issue is now moot. Sorry, I have been away. Please open it again if necessary. Thanks.

closed 2023-03-28 by zs6buj