zs6buj/AntTracker #44: [Feature Request] Change telemetry protocol using external/onboard button

bsh314, 2022-02-17:
Hello!

I'm trying to build an RubyHD/OpenHD ground station that includes an antenna tracker based on this repo and it's great work. 

As RubyHD/OpenHD is an digital system and a lot of my models are still using analog video system, I want to be able use ATT part of the station independently from RubyHD/OpenHD system that provides Mavlink2 telemetry directly to ATT. In case of analog models I usually take an LTM telemetry from my tx16's internal bluetooth module to my phone/ATT. 

As ESP32 has multiple spare serial ports and general propose pins, additional bluetooth and a button/switch can be added. The idea is: When PINX is LOW, uart0 and Mavlink should be used  . When PINX is HIGH, uart1 and LTM should be used.

I'll try to fork this repo and add this feature.

Requeremests:
- Multiple input UARTs 
- Set per-UART telemetry type and speed
- Current UART port change on STARTUP
- [Optional] Soft serial port
- [Optional] On-the-fly telemetry type change for current UART

If someone has some input or help on this matter, please leave a message here.
Thank you @zs6buj for a great code!

zs6buj, 2022-02-18:
Hello Bohdan, your project sounds interesting. I still use ez-wifibroadcast for hd video on my tl-650. It has worked well for me, and remained unchanged now for a few years. 

Regarding your requirements:

Since the esp32 boards (and AntTracker) support BT internally,  if i understand you correctly, 
you won't necessarly need the extra UART or softwareserial for ltm from your tx16? 

It should be easy to sense a digital pin, and change protocol on-the-fly, as long as the necessary i/o are initialized.

I'm away from home until May, but have some limited resources at my disposal and can try to help.

bsh314, 2022-02-18:
> 

Thanks! I'll keep you updated!

bsh314, 2022-03-05:
Seems like i'm having issues toeven make it work with default configuration.
I have internal GPS + compass on the tracker, it is detected properly and bauds set to 9600, my gps is M8Q from matek.
Serial telemetry should be ok, i have "Serial Telem In" message.
But oled screen always shows "starting" and no "waiting for gps lock" of any kind. Compass is ready.

I've tried to go outside and run it from where gps is available, but no gps sats was reported, oled shows correct values from FC if there is a gps lock, but AntTracker keeps on "starting".

Debugging returns only this:
```
Starting AntTrack version:2.19.07
Setting up wire I2C   SDA:21  SCL:22
Display support activated: Landscape 
64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
Target Board is Target Board = 3  ESP32 / Variant is Dev Module
Serial Telemetry In
Selected protocol is Mavlink 2
Dynamic heading source = 4 Tracker Box Compass
Compass type QMC5883L expected
QMC5883L compass ready
Servo slowdown factor is 10ms per degree of rotation
```
My config file-> [https://pastebin.com/NWVL3vHd](https://pastebin.com/NWVL3vHd)
My comps -> 2x180 servos, esp32 dev board, 124x64 oled screen, Matek m8q-5883 GPS+Compass

FC gps info -> 
[image]

strange message when i'm not on gps reach -> 
[image]

zs6buj, 2022-03-06:
Hello again Bohdan

May I suggest you activate 

#define Debug_All

in the config.h, compile, flash and run again, then let's see the log.

**Serial Telemetry In** simply means that is what you have selected. I does not mean that the Mavlink 2 telemetry is being read properly. From what I can see, no telemetry is being read at the serial port. The debug above should help. Make sure that the RX and TX wires cross over, and that the speed is correct.

bsh314, 2022-03-06:
Hello!
 I´ve changed some of the config file options, seems like setting mavlink 2 as telemetry does not allow to proceed with startup. Debuggin options was not working either, esp32 was "stuck" in that point after spitting the initial serial info. After changing telemetry to auto and forcing GPS baud rate to 115200, seems like it starts properly.

Next issue: ATT starts and reports GPS is good, moves both servos and tries to point the direction of the UAV, but as per documentation of M8Q from Matek, this specific GPS/compass unit reqires 270deg Flip rotation to get "true" data. So my first task will be to add a compass orientation config parameters to override compass output values in consonanse, i'll check betaflight/ivan repos to get approx idea how to do it. I'll continue with my previous changes after that.

Thank you @zs6buj for your help!

zs6buj, 2022-03-07:
Ok, thanks for the feedback Bodan. I'll keep an eye on this thread.

bsh314, 2022-03-08:
I've created a PR for the compass alignment and some misc stuff
[https://github.com/zs6buj/AntTracker/pull/45](https://github.com/zs6buj/AntTracker/pull/45)

zs6buj, 2022-03-09:
Thank you for this useful add-on Bohdan. Merged!

closed 2023-03-28 by zs6buj