zs6buj/AntTracker #46: [Feature Request] BLE Support for Telemetry

Razsmithii, 2022-04-09:
Ant Tracker seems to be a great and smart solution. What i miss: BLE Support for getting temelemtry Data. When an external TX is connected to taranis X-Lite, there is no easy serial Port available. Why Not usw BLE, like in "Telemetry Viewer" (Smartphone App)?

arifly, 2022-04-10:
Ive got a modified  version with BLE, Crossfire and Touch working, ..sort of. 
Ive someone is interested i put it on git

zs6buj, 2022-04-11:
Arifly:

Yes please. That would be a nice feature to add.

arifly, 2022-04-15:
noob mod: 
[https://github.com/arifly/AntTracker/tree/master/Source/AntTrack](url)
to use with a ESP32 (DevKit), the Frsky X10 Bluetooth Module, OpenTX in Telemetry Mirror Mode and Crossfire (hardcoded atm)

The serviceUUID need to be changed to your devices adress

I will try to make PRs(?)

zs6buj, 2022-04-16:
Ok, thank you. I'm away from home until early May, but can stay in touch.

Razsmithii, 2022-04-25:
BLE Connection works great - many thanks!

zs6buj, 2022-04-26:
That's great news. I'll check this out as soon as I can. I leave for home on Monday.

arifly, 2022-04-26:
yes, but sometimes the mainloop didn't get time to do his thing.
And the bluetooth-stack can freeze.
To overcome this i use a second ESP with watchdog enabled.
To get away with no PRs, i add a converter to LTM, so nearly no changes to the Original Code are needed :>

As we use HardwareSerial the default speed for LTM can changed to 115200.
Or change the speed in "BLE-crsf2ltm.ino". LTM messages are send as fast as possible. 

[https://github.com/arifly/BLE-crsf2ltm](https://github.com/arifly/BLE-crsf2ltm)

closed 2022-05-10 by Razsmithii