zs6buj/AntTracker #47: Calculating issue of heading

Razsmithii, 2022-04-25:
Hello,

Everything works fine in the Code: great config, compiling, getting Data via Protocol, getting GPS, compass heading..
BUT: für calculation of heading is Not correct:

 21:42:22.500 hom.lat = 47.9879341 hom.lon = 7.8289866 hom.alt = 300 cur.lat = 47.9878387 cur.lon = 7.8294182 cur.alt = 292 cur.alt_ag = 0
21:42:22.505   hc_vector.az= 236.05  hc_vector.el= 0.00  hc_vector.dist= 14388836  Elapsed= 117.559

I tried own GPS (Option 4) and GPS on plane: Same Problem!

At least, distance can not be true, it should be around 35m.
Do you have any advice?
Thanks in advance....

zs6buj, 2022-04-26:
Hmm. Yes, the calc result looks obviously wrong, but I have never seen this before.

I simply use the trig Pythagoras equations.  Maybe try declaring the variables volatile.

Maybe isolate the code in a separate small sketch and test it there.

Razsmithii, 2022-05-10:
Problem is found, but seems to be related to arifly´s BLE version (https://github.com/arifly/AntTracker):
Inside the "AzEl"-Sktech, the following is calculated:

#if (Telemetry_In == 5) 
      float lon2 = cur.lon;
      float lat2 = cur.lat;

but this one is correct (and tested) for me:

float lon2 = cur.lon / 180 * PI;
float lat2 = cur.lat / 180 * PI;

closed 2022-05-10 by Razsmithii