zs6buj/AntTracker #50: No Header Fioe (.h)

kusnendar, 2022-09-01:
Get error not header file (.h) frome new libraries file new version
[image]

zs6buj, 2022-09-01:
Only copy these folders from my GitHub STMF1xx libraries file. They may
well be out of date, so you might need to use Library Manager in the
Arduino GUI to update them.

[image: image.png]

kusnendar, 2022-09-01:
> 

I has copy these folder. But same result. & I checked at libraries folder not found file (.h)

zs6buj, 2022-09-01:
I just compiled it here. I clean compiles, but is too big for flash memory. I guess I added too many "features" into the ESP32 version. Lol.

[image]

I'll see if I can cut it down.

So, in YOUR Arduino library folder, delete (or save somewhere  else) all the libraries. Now try to compile, and it will complain about a missing library. Copy back only that library folder.  COPY ONLY THE LIBRARY NAME FOLDER, NOT THE COLLECTIVE FOLDER WITH THEM INSIDE.

You might want to tick "Show verbose output during compilation" in File/Preferences
[image]

Read the resulting compilation logs carefully when you try to compile.

zs6buj, 2022-09-01:
Ok, it clean compiles when I select 128k variant:

[image]

[image]

kusnendar, 2022-09-01:
> Ok, it clean compiles when I select 128k variant:
> 
> [image]
> 
> [image]

Do can i use my bluepill STM22F103C8(64k) for this project?

kusnendar, 2022-09-01:
> I just compiled it here. I clean compiles, but is too big for flash memory. I guess I added too many "features" into the ESP32 version. Lol.
>
> [image]
>
> I'll see if I can cut it down.
>
> So, in YOUR [...]

Please show your  libraries folder(take picture)

kusnendar, 2022-09-01:
> I just compiled it here. I clean compiles, but is too big for flash memory. I guess I added too many "features" into the ESP32 version. Lol.
>
> [image]
>
> I'll see if I can cut it down.
>
> So, in YOUR [...]

I hope your make 2 file config for STM32 (64k) & ESP32

zs6buj, 2022-09-01:
Ok, but my library folder has many libraries, also for other boards. Also, be aware that every board type has its own package folders, which contain libraries for those board variants:
[image]

[image]

zs6buj, 2022-09-01:
You can put library folders into Documents/arduino/libraries/

and the IDE usually looks around for the best ones

zs6buj, 2022-09-01:
Do NOT put folder hierarchies in there

zs6buj, 2022-09-01:
I also cleaned up the version numbers slightly on GitHub, but you should not need to download new version 2.19.09

zs6buj, 2023-03-28:
Stale

closed 2023-03-28 by zs6buj