zs6buj/AntTracker #51: Config for Auto Connect Bluetooth

kusnendar, 2022-09-02:
Hi! now i use board ESP32. 
Where is configurator in "config.h" to ESP32 auto connect with my bluetooth module HC-5?

My bluetooth module HC-5 is "kusnendar fpv" & password: 1234
My modulw HC-5 transmitt telemetry mavlink from openHD

zs6buj, 2022-09-02:
Hi [kusnendar](https://github.com/kusnendar)

Do you know how to setup ESP32 in Arduino IDE ( or PlarformIO)? If you do, I can make up a config.h that should work. But first you need to install the esp32 core onto the IDE.  https://randomnerdtutorials.com/installing-the-esp32-board-in-arduino-ide-windows-instructions/

Then you need to select the correct variant in the ide:

[image]

You need to tell me which esp32 board you have, with maybe a picture.

zs6buj, 2022-09-02:
I could also compile a binary for you, if you know how to flash a binary with the espressif flash tool

zs6buj, 2022-09-02:
I leave on a long flight on Monday/Tuesday

kusnendar, 2022-09-02:
> I could also compile a binary for you, if you know how to flash a binary with the espressif flash tool

This my esp32
[image]

zs6buj, 2022-09-02:
Looks like an ordinary Dev Module clone. Are you expecting Mavlink telemetry in?

zs6buj, 2022-09-02:
Ok, see that you mentioned mavlink above.

kusnendar, 2022-09-02:
> Ok, see that you mentioned mavlink above. Do you plan to use a compass and / or GPS module attached on the tracker?
> 
> Also, (1) you can determine HOME location by pressing a button on the ESP32, or (2) you can determine HOME when you arm your drone/plane.
> 
> 1 or 2?

- I'm not use compass/gps module on tracker (just bluetooth module hc-05)
-  (2) home location by arm from my rc plane

zs6buj, 2022-09-02:
[config.txt](https://github.com/zs6buj/AntTracker/files/9477082/config.txt)

You need to change the extension from .txt to .h

kusnendar, 2022-09-02:
> [config.txt](https://github.com/zs6buj/AntTracker/files/9477082/config.txt)
> 
> You need to change the extension from .txt to .h

Do not input password my bluetooth HC-05? Password is 1234

zs6buj, 2022-09-02:
[AntTrack.ino.esp32.zip](https://github.com/zs6buj/AntTracker/files/9477124/AntTrack.ino.esp32.zip)

binary is in the zip file. I don't recall setting up a pw for BT, but if it asks let me know.

kusnendar, 2022-09-02:
> [config.txt](https://github.com/zs6buj/AntTracker/files/9477082/config.txt)
> 
> You need to change the extension from .txt to .h

2 more question

1. Do this right for setup no compass & GPS module attached on the tracker? but use bluetooth module 

// Select one heading source. We need this to relate the external world of co-ordinates to the internal tracker co_ordinates.
//#define Heading_Source  1     // 1=Flight Computer GPS 
#define Heading_Source  2     // 2=Flight Computer Compass
//#define Heading_Source  3     // 3=Trackerbox_Compass 
//#define Heading_Source  4     // 4=Trackerbox_GPS_And_Compass
// Select GPS module serial link speed. Many GPS modules are capable of using multiple serial speed out from the box.
// This information should be provided by the manufacturer.
// If not defined, speed will be selected automatically.
// #define Box_GPS_Baud 9600

2. How to flash bia bin file? do you have tutorial?

kusnendar, 2022-09-02:
do this right for my parameter if build new config?

[image]

zs6buj, 2022-09-02:
Yes, it means that it uses the plane's compass to orientated the tracker when you start. So you point the plane in the same direction the tracker is facing.

zs6buj, 2022-09-02:
https://m.youtube.com/watch?v=oLzR5gjZKJY

kusnendar, 2022-09-02:
> [config.txt](https://github.com/zs6buj/AntTracker/files/9477082/config.txt)
> 
> You need to change the extension from .txt to .h

LAST QUESTION

I want try flash via Arduino IDE, do my parameter is right before flashed?
[image]

also i not use OLED display, do your [config.txt](https://github.com/zs6buj/AntTracker/files/9477082/config.txt) has change for no OLED Display?

zs6buj, 2022-09-02:
It looks ok but I'm on my phone

kusnendar, 2022-09-02:
> It looks ok but I'm on my phone

also i not use OLED display, do your [config.txt](https://github.com/zs6buj/AntTracker/files/9477082/config.txt) has change for no OLED Display?

zs6buj, 2022-09-02:
I disabled display

kusnendar, 2022-09-03:
> I disabled display

Thanks. I will try to flash

kusnendar, 2022-09-03:
> I disabled display

I Use ESP32, where is PIN for azimuth & elevator servo?

zs6buj, 2022-09-03:
#define azPWM_Pin        32   
  #define elPWM_Pin         33 

See below for Dev Module pins

#if (ESP32_Variant == 1)          // ESP32 Dev Module
  uint8_t in_rxPin =        27;  // uart1
  #define in_txPin          17 
  uint8_t gps_rxPin =       13;  // uart2 for tracker box GPS if applicable
  #define gps_txPin          4  
  bool rxInvert = true;          // ONLY FOR FrSky S.Port, NOT F.Port, NOT MAVLINK
  #define SetHomePin        34   // LOW == pushed    
  #define StatusLed         25   // Off=No good GPS yet, flashing=good GPS but home not set yet, solid = ready to track
  #define BuiltinLed        99
  #define azPWM_Pin         32   // azimuth servo (can't be 34,35,36,39 because input only !!)
  #define elPWM_Pin         33   // elevation servo(can't be 34,35,36,39 because input only !!)
  #define BuiltinLed        02   // PB1

kusnendar, 2022-09-03:
after esp32 flashed always rebooting
[image]

kusnendar, 2022-09-03:
bloetooth not pairing:

https://youtu.be/ZL7uy35cXKw

zs6buj, 2022-09-04:
Can you please show me the flashing process. 

I found this regarding BT password:  https://electronics.stackexchange.com/questions/594094/esp32-classic-bluetooth-set-password

so maybe I can look at it after I arrive at my destination next week.

I'm packing up equipment today, and travel tomorrow, but I'll have my tablet and phone

kusnendar, 2022-09-04:
> Can you please show me the flashing process.
>
> I found this regarding BT password: https://electronics.stackexchange.com/questions/594094/esp32-classic-bluetooth-set-password
>
> so maybe I can look at [...]

I don't use bluetooth because it's difficult,

then I use serial port.

And it works when I select Board: DOIT DEVESP32 DEV KIT1.

do my flashing result successful?
[image]

zs6buj, 2022-09-08:
Yes it looks like your flashing effort was successful.

closed 2023-03-28 by zs6buj