zs6buj/AntTracker #54: Bug found in Compass.ino

DiscoMan18, 2023-04-05:
#if defined Compass_Align
          fHeading = applySensorAlignment(fHeading, Compass_Align);
        #endif

Change Compass_Align to Compass_Rotation :) 

Sorry, I'm writing this from a phone, can't use proper formatting. Also it would be good too see one day compass calibration and tilt compensation. 

Great project anyways, all the fpv community should thank you for this:)

DiscoMan18, 2023-04-05:
A separate thanks goes for QLRS support:)

zs6buj, 2023-04-06:
@DiscoMan18 

Thank you for the feedback. Could you give me a little more reasoning for the change to Bohan's PR? I'm travelling and also working on my phone :)

DiscoMan18, 2023-04-06:
Well, the reason for this change in code is simple: Compass_Align is a structure, and Compass_Rotation is a constant that should have only of one of the values the struct contains. When you pass Compass_Align to the function it simply doesn't work and proper compass align is not applied at all. I tested it yesterday with TTGO T-Display board and HMC5883L compass module. After the change in code that I proposed it started finally working.

zs6buj, 2023-04-06:
I'm looking at it now on the big screen, and I can see it needs a bit of work.  Thanks again for the feedback.

zs6buj, 2023-04-06:
V2.19.10  . Could you please try it when you get the chance.

DiscoMan18, 2023-04-06:
> V2.19.10 . Could you please try it when you get the chance.

Sorry, I can't, as I used your code and have completely rewritten it to suit my purposes. But it should be working, as I already tested these changes on my own.

closed 2023-04-07 by zs6buj