zs6buj/AntTracker #55: Error installing on esp32

brunoproza, 2023-07-02:
Hello.

I followed the steps from the link https://www.rcgroups.com/forums/showthread.php?2942118-Build-a-Reliable-and-Cheap-Antenna-Tracker-this-Weekend-%C2%96-by-zs6buj /page44 in your answer #657

and everything is ok, however I am getting the error with the serial ports.

 #if defined STM32F1xx
    #if defined NoGenericSerial
      // NOTE NOTE NOTE NOTE! In IDE select Tools/U(S)ART support: "Enabled (no generic 'Serial')"
      // Now we map USARTS in a sensible way
      HardwareSerial Serial(USART1);  //rx1=PA10 tx1=PA9  - for flashing and monitor
      HardwareSerial Serial1(USART2); //rx2=PA3  tx2=PA2  - for telemetry in
      HardwareSerial Serial2(USART3); //rx3=PB11 tx3=PB10 - for GPS if present
    #endif
    #include <SoftwareSerial.h>  
    SoftwareSerial inSerial(in_rxPin, in_txPin, rxInvert); // RX=10, TX=11 
  #endif

  #define log                   Serial         // USB / Serial

i get the error below

`
config.h:850:33: error: 'Serial' was not declared in this scope; did you mean 'Serial3'?
  850 |   #define log                   Serial         // USB / Serial
      |                                 ^~~~~~
`

could you help me to solve it?

zs6buj, 2023-07-03:
Hey Carlo, sure I can help.

I just clean compiled on my laptop, so I guess there might be a problem with your setup or  config.h settings. I assume you are using the latest version, and that you select the  ESP32 dev board.

[image]

Also, did you copy the ESP32 libraries over from your cloned github folder to your arduino library folder?

brunoproza, 2023-07-03:
I'm sorry, I got confused with the name of the board

I'm actually using a blue pill stm32.

I'm using the latest version of the Tracker antenna, I copied all the libraries, and adjusted the arduino ide settings as your post quoted above

but I always get this serial error :/

zs6buj, 2023-07-03:
Check out this link:
[]https://github.com/stm32duino/Arduino_Core_STM32

Did you use this json?

Add this link in the "Additional Boards Managers URLs" field:

https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json

brunoproza, 2023-07-03:
I changed it to the JSON informed above, I will perform a new test

brunoproza, 2023-07-03:
Again the same error, here are some prints that maybe can help to show my problem more.

I put the informed JSON
[image]

installed all dependencies
[image]

I set the options to STM32

[image]

but when running the code verification I get the error below. (all apparently related to the serial)

In file included from C:\temp\AntTracker\Source\AntTrack\AntTrack.ino:126:
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino: In function 'void setup()':
C:\temp\AntTracker\Source\AntTrack\config.h:850:33: error: 'Serial' was not declared in this scope; did you mean 'Serial3'?
  850 |   #define log                   Serial         // USB / Serial
      |                                 ^~~~~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino:350:3: note: in expansion of macro 'log'
  350 |   log.begin(115200);
      |   ^~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino: In function 'void loop()':
C:\temp\AntTracker\Source\AntTrack\config.h:850:33: error: 'Serial' was not declared in this scope; did you mean 'Serial3'?
  850 |   #define log                   Serial         // USB / Serial
      |                                 ^~~~~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino:1005:15: note: in expansion of macro 'log'
 1005 |               log.println("Dynamic tracking (moving home) started");  // moving home tracking
      |               ^~~
C:\temp\AntTracker\Source\AntTrack\config.h:850:33: error: 'Serial' was not declared in this scope; did you mean 'Serial3'?
  850 |   #define log                   Serial         // USB / Serial
      |                                 ^~~~~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino:1015:15: note: in expansion of macro 'log'
 1015 |               log.println("Dynamic tracking (moving home) failed. No good tracker box compass!");
      |               ^~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino: In function 'void FinalStoreHome()':
C:\temp\AntTracker\Source\AntTrack\config.h:850:33: error: 'Serial' was not declared in this scope; did you mean 'Serial3'?
  850 |   #define log                   Serial         // USB / Serial
      |                                 ^~~~~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino:1118:11: note: in expansion of macro 'log'
 1118 |           log.print("Heading calculated = "); log.print(hom.hdg, 1);
      |           ^~~
C:\temp\AntTracker\Source\AntTrack\config.h:850:33: error: 'Serial' was not declared in this scope; did you mean 'Serial3'?
  850 |   #define log                   Serial         // USB / Serial
      |                                 ^~~~~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino:1151:9: note: in expansion of macro 'log'
 1151 |         log.println("Trackerbox GPS/Compass. Should never get here because home is dynamic!");
      |         ^~~
C:\temp\AntTracker\Source\AntTrack\config.h:850:33: error: 'Serial' was not declared in this scope; did you mean 'Serial3'?
  850 |   #define log                   Serial         // USB / Serial
      |                                 ^~~~~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino:1157:9: note: in expansion of macro 'log'
 1157 |         log.println("No headingSource!");
      |         ^~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino: In function 'void FirstStoreHome()':
C:\temp\AntTracker\Source\AntTrack\config.h:850:33: error: 'Serial' was not declared in this scope; did you mean 'Serial3'?
  850 |   #define log                   Serial         // USB / Serial
      |                                 ^~~~~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino:1176:5: note: in expansion of macro 'log'
 1176 |     log.print("Static home location AUTO set to ");
      |     ^~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino: In function 'void DisplayHome()':
C:\temp\AntTracker\Source\AntTrack\config.h:850:33: error: 'Serial' was not declared in this scope; did you mean 'Serial3'?
  850 |   #define log                   Serial         // USB / Serial
      |                                 ^~~~~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino:1187:5: note: in expansion of macro 'log'
 1187 |     log.print("Static home location set to Lat = "); log.print(hom.lat,7);
      |     ^~~
C:\temp\AntTracker\Source\AntTrack\AntTrack.ino: In function 'void DisplayHeadingSource()':
C:\temp\AntTracker\Source\AntTrack\config.h:850:33: error: 'Serial' was not declared in this scope; did you mean 'Serial3'?
[... 207 lines cut ...]
  339 |           log.println("Flight Computer GPS timeout");
      |           ^~~
C:\temp\AntTracker\Source\AntTrack\config.h:850:33: error: 'Serial' was not declared in this scope; did you mean 'Serial3'?
  850 |   #define log                   Serial         // USB / Serial
      |                                 ^~~~~~
C:\temp\AntTracker\Source\AntTrack\Utilities.ino:348:14: note: in expansion of macro 'log'
  348 |              log.println("Good box GPS lock");
      |              ^~~
C:\temp\AntTracker\Source\AntTrack\config.h:850:33: error: 'Serial' was not declared in this scope; did you mean 'Serial3'?
  850 |   #define log                   Serial         // USB / Serial
      |                                 ^~~~~~
C:\temp\AntTracker\Source\AntTrack\Utilities.ino:351:13: note: in expansion of macro 'log'
  351 |             log.println("No box GPS lock!");
      |             ^~~
C:\temp\AntTracker\Source\AntTrack\Utilities.ino: In function 'void LostPowerCheckAndRestore(uint32_t)':
[... cut ...] lared in this scope; did you mean 'Serial3'?
  850 |   #define log                   Serial         // USB / Serial
      |                                 ^~~~~~
C:\temp\AntTracker\Source\AntTrack\Utilities.ino:379:7: note: in expansion of macro 'log'
  379 |       log.printf("Home data restored from NVM, decay %d secs is within limit of %d secs\n", decay_secs, home_decay_secs);
      |       ^~~

exit status 1

Compilation error: 'Serial' was not declared in this scope; did you mean 'Serial3'?

zs6buj, 2023-07-03:
[image]

U(S)ART Support Enabled (generic serial)

brunoproza, 2023-07-03:
I'm going to test this but if it doesn't work I'm going to change the card to an esp32

brunoproza, 2023-07-03:
It worked ;) Now I'm going to do the other settings, such as adding display support.

Thanks!

[image]

closed 2023-07-03 by brunoproza