zs6buj/AntTracker #56: compiling error

sosko71, 2023-08-12:
[compile error.txt](https://github.com/zs6buj/AntTracker/files/12328082/compile.error.txt)
Hi I have problem with compiling code for my LILYGO T-DISPLAY

zs6buj, 2023-08-17:
Hello @sosko71

It looks like you don't gave the Mavlink library in your library folder in Arduino. You must go here: [https://github.com/mavlink/c_library_v2](url) and clone or download the entire library, and copy it to your Arduino library folder

sosko71, 2023-08-19:
All required libraries are installed. 
here is complete error mesage

In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:2:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_conversions.h:39:1: error: 'MAVLINK_HELPER' does not name a type
 MAVLINK_HELPER void mavlink_quaternion_to_dcm(const float quaternion[4], float dcm[3][3])
 ^~~~~~~~~~~~~~
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_conversions.h:39:1: note: the macro 'MAVLINK_HELPER' had not yet been defined
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:7:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/protocol.h:74: note: it was later defined here
     #define MAVLINK_HELPER static inline
 
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:2:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_conversions.h:69:1: error: 'MAVLINK_HELPER' does not name a type
 MAVLINK_HELPER void mavlink_dcm_to_euler(const float dcm[3][3], float* roll, float* pitch, float* yaw)
 ^~~~~~~~~~~~~~
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_conversions.h:69:1: note: the macro 'MAVLINK_HELPER' had not yet been defined
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:7:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/protocol.h:74: note: it was later defined here
     #define MAVLINK_HELPER static inline
 
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:2:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_conversions.h:103:1: error: 'MAVLINK_HELPER' does not name a type
 MAVLINK_HELPER void mavlink_quaternion_to_euler(const float quaternion[4], float* roll, float* pitch, float* yaw)
 ^~~~~~~~~~~~~~
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_conversions.h:103:1: note: the macro 'MAVLINK_HELPER' had not yet been defined
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:7:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/protocol.h:74: note: it was later defined here
     #define MAVLINK_HELPER static inline
 
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:2:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_conversions.h:119:1: error: 'MAVLINK_HELPER' does not name a type
 MAVLINK_HELPER void mavlink_euler_to_quaternion(float roll, float pitch, float yaw, float quaternion[4])
 ^~~~~~~~~~~~~~
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_conversions.h:119:1: note: the macro 'MAVLINK_HELPER' had not yet been defined
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:7:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/protocol.h:74: note: it was later defined here
     #define MAVLINK_HELPER static inline
 
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:2:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_conversions.h:147:1: error: 'MAVLINK_HELPER' does not name a type
 MAVLINK_HELPER void mavlink_dcm_to_quaternion(const float dcm[3][3], float quaternion[4])
 ^~~~~~~~~~~~~~
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_conversions.h:147:1: note: the macro 'MAVLINK_HELPER' had not yet been defined
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:7:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/protocol.h:74: note: it was later defined here
     #define MAVLINK_HELPER static inline
 
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:2:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_conversions.h:190:1: error: 'MAVLINK_HELPER' does not name a type
 MAVLINK_HELPER void mavlink_euler_to_dcm(float roll, float pitch, float yaw, float dcm[3][3])
 ^~~~~~~~~~~~~~
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_conversions.h:190:1: note: the macro 'MAVLINK_HELPER' had not yet been defined
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:7:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/protocol.h:74: note: it was later defined here
     #define MAVLINK_HELPER static inline
 
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:4:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_helpers.h: In function 'uint16_t mavlink_finalize_message_buffer(mavlink_message_t*, uint8_t, uint8_t, mavlink_status_t*, uint8_t, uint8_t, uint8_t)':
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_helpers.h:239:16: error: 'MAVLINK_STX' was not declared in this scope
   msg->magic = MAVLINK_STX;
                ^~~~~~~~~~~
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_helpers.h:239:16: note: the macro 'MAVLINK_STX' had not yet been defined
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:135:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/ardupilotmega/mavlink.h:12: note: it was later defined here
 #define MAVLINK_STX 253
 
In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:4:
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_helpers.h: In function 'const mavlink_msg_entry_t* mavlink_get_msg_entry(uint32_t)':
c:\Users\Silvo\Desktop\AntTrack\libraries\c_library_v2-master/mavlink_helpers.h:520:60: error: 'MAVLINK_MESSAGE_CRCS' was not declared in this scope
[... 34 lines cut ...]
     ino_t
[... cut ...]
  Not used: C:\Users\Silvo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\EEPROM
Multiple libraries were found for "Wire.h"
  Used: C:\Users\Silvo\Desktop\AntTrack\libraries\Wire
  Not used: C:\Users\Silvo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\Wire
Multiple libraries were found for "SPI.h"
  Used: C:\Users\Silvo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\SPI
  Not used: C:\Users\Silvo\Desktop\AntTrack\libraries\SPI
exit status 1

Compilation error: 'MAVLINK_HELPER' does not name a type

zs6buj, 2023-09-30:
@Silvo, this is rather confusing. I wish I could help.

"In file included from C:\Users\Silvo\Desktop\AntTrack\AntTrack.ino:2:"

There is no include in line 2, so I'm assuming line 125 or 128.

Have you changed the code, or removed the comments?  Are you library files in the arduino folder, then in the "libraries" folder?

[image]

zs6buj, 2023-09-30:
[image]

closed 2023-11-01 by zs6buj

sosko71, 2023-11-01:
Hi I dont know what is wrong, but I managed to compile on another PC and it is working. Thank you for your help