zs6buj/AntTracker #57: Problems with push button

Droneforest05, 2023-09-25:
Hello im having problems using the button
I tried every gpio on the board and in any ot them the push button works
now i set it to pin 34 and the other pin to gnd but nothing 
i dont know if you can do something but if you cant i see that im not the first to have that problem and it worked on version 2.18.7 so if you can at least add the old ones like you used to 
I will be grateful
All the other features works (compass telem in....._)
#if (ESP32_Variant == 1)          // ESP32 Dev Module
  uint8_t in_rxPin =        27;  // uart1
  #define in_txPin          17 
  uint8_t gps_rxPin =       13;  // uart2 for tracker box GPS if applicable
  #define gps_txPin          4  
  bool rxInvert = true;          // ONLY FOR FrSky S.Port, NOT F.Port, NOT MAVLINK
  #define SetHomePin        34   // LOW == pushed    
  #define StatusLed         25   // Off=No good GPS yet, flashing=good GPS but home not set yet, solid = ready to track
  #define BuiltinLed        99
  #define azPWM_Pin         32   // azimuth servo (can't be 34,35,36,39 because input only !!)
  #define elPWM_Pin         33   // elevation servo(can't be 34,35,36,39 because input only !!)
  #define BuiltinLed        02   // PB1   

    #define displaySupport       // uncomment me if you have a SSD1306 display
    #if (defined displaySupport)   // Display type defined with # define displaySupport   
      #define SSD1306_Display         // OLED display type    
      /* Below please choose either Touch pin-pair or Digital pin-pair for display scrolling
       *  Pin == 99 means the pin-pair is not used
       */ 
      #define Pup           99        // Board Button 1 to scroll the display up
      #define Pdn           99        // Board Button 2 to scroll the display down   
      #define Tup           33        // Touch pin to scroll the display up
      #define Tdn           32        // Touch pin to scroll the display down   
          
      #define SDA           21        // I2C OLED board and/or Compass
      #define SCL           22        // I2C OLED board and/or Compass
      #define display_i2c_addr      0x3C       // I2C OLED board

Droneforest05, 2023-11-11:
hello any one can help?

zs6buj, 2023-11-13:
[Droneforest05](https://github.com/Droneforest05)

I will see if I can test this sometime today.

zs6buj, 2023-11-14:
Hello [Droneforest05](https://github.com/Droneforest05)

I managed to look at this today, and there is indeed a problem with using pin 34. It appears that the read-only pins behave differently. So my proposed solution is this:

1 please change SetHomePin to use gpio 12 in config.h
2 find the bool homeButtonPushed() function in the Utilities tab, then comment out this line:
   hbp = !hbp; like so 

//  hbp = !hbp;

it should be around line 1584.

I'll be posting a new version in the next few days, including crsf/elrs support, and of course I'll fix this problem. Thanks for reporting it.

Droneforest05, 2023-11-14:
i changed that i will test this weekend because in the middle of the city
having gps look on air is quitte difficult
just two questions one a bit stupid
-The button in this case should short between ground and gpio 12 no?
-Is there any other requirement besides having GPS lock in the air so that
after pressing the button it starts tracking? I say this because I have had
it for a while so that it starts tracking when it was armed but it didn't
work, it appears on the screen that it is armed while the LED It flashed
fast but I didn't get the "start to trak" or something like that, it was
the message.

zs6buj, 2023-11-15:
> The button in this case should short between ground and gpio 12 no?

Yes

"LED It flashed
fast"

the led should be on solid

> so that it starts tracking when it was armed but I didn't get the "start to trak" or something like that

This is a good point. I will add a message like that for set home when motors armed.

zs6buj, 2023-11-15:
> because in the middle of the city
> having gps look on air is quite difficult

I had to smile at this. My office where my dev box is set up also does not have GPS view of the sky. So this is how I test:

[image]

[image]

zs6buj, 2023-11-15:
> I have had
> it for a while so that it starts tracking when it was armed

So can you confirm you activated this macro?
```

//                              S E T   H O M E   O N   A R M
#define SET_HOME_AT_ARM_TIME      // else set home location with push button

and can you confirm which heading source you use:

```
//#define HEADING_SOURCE  1     // 1=Flight Computer GPS, 
#define HEADING_SOURCE  2     // 2=Flight Computer Compass
//#define HEADING_SOURCE  3     // 3=Trackerbox_Compass 
//#define HEADING_SOURCE  4     // 4=Trackerbox_GPS_And_Compass
```

Droneforest05, 2023-11-15:
Yeah I set to start tracking with arming and I select option 3 ( tracker
compass) I have connected a GPS but I can't get it worked so I give up on
this idea

Droneforest05, 2023-11-15:
Later I can send you the complete Arduino code if you want and photos of
the my schematic

Droneforest05, 2023-11-20:
I change it to gpio 12 and now it works well

zs6buj, 2023-11-20:
Great. The new version is also up on git, under Beta

closed 2024-03-11 by zs6buj