zs6buj/AntTracker #58: GPIO output gpio_num error

sosko71, 2023-11-18:
Anyone any idea what caused this error I'm getting? Code compiles and uploads fine,
Two different versions loaded on and the problem is the same in both.

Serial monitor output

version to work with BLE

Starting AntTrack version:2.19.07
Setting up wire I2C SDA:21 SCL:22
Display support activated: Landscape
135x240 text_size=2 char_w_px=12 char_h_px=16 scr_h_ch=8 scr_w_ch=20
Target Board is Target Board = 3 ESP32 / Variant is LILYGO® TTGO T-Display ESP32 1.14 inch ST7789 Colour LCD
FrSky BLE In
Selected protocol is F.Port 1
**E (2610) gpio: GPIO can only be used as input mode
E (2611) gpio: gpio_set_level(226): GPIO output gpio_num error**
Dynamic heading source = 4 Tracker Box Compass
Compass type HMC5883L expected
Servo slowdown factor is 10ms per degree of rotation
autoBaud - sensing pin 13
Tracker box GPS baud rate detected is 9600b/s
BLE Start - Scanning....
Display width of 20 exceeded for |BLE Start - Scanning....|
Forming a connection to d0:2e:ab:b1:d1:60
- Connected to server
- Found our service
- Found our characteristic
We are now connected to the BLE Server.
**E (7576) gpio: gpio_set_level(226): GPIO output gpio_num error**

Version to work with serial port

Starting AntTrack version:2.19.10
Setting up wire I2C SDA:21 SCL:22
Display support activated: Landscape
135x240 text_size=2 char_w_px=12 char_h_px=16 scr_h_ch=8 scr_w_ch=20
Target Board = 3 ESP32 / Variant is LILYGO® TTGO T-Display ESP32 1.14 inch ST7789 Colour LCD
Serial Telemetry In
Selected protocol is Mavlink 1
Dynamic heading source = 4 Tracker Box Compass
Compass type HMC5883L expected
Testing Servos
autoBaud - sensing pin 13
Tracker box GPS baud rate detected is 9600b/s
**E (13470) gpio: gpio_set_level(226): GPIO output gpio_num error**

casfra96, 2024-02-07:
GPIO pin 34 35 36 and 39  sholud be input only, check if you used one of those pins for output.

zs6buj, 2024-02-08:
Starting AntTrack version:2.19.07
....
E (13470) gpio: gpio_set_level(226): GPIO output gpio_num error

I think this was fixed in v2.19.10, but I recommend you try the latest (beta)

v2.19.10 2023-04-06 Refined BOX_COMPASS_ALIGN routine     
                    Fixed "E (2613) gpio"

closed 2024-03-11 by zs6buj