zs6buj/AntTracker #60: Problems using LTM protocol

casfra96, 2024-02-06:
Hi,
I  found two possible bugs using those configurations:

`#define Telemetry_In  0    // Serial Port (default) - all protocols`
and
`#define PROTOCOL 6     // LTM`

### First problem
 In the file AntTrack.ino, row 978, the function `CheckStatusAndTimeouts(); // and service status LED`  is called.
That function is definited in the file Utilities.ino, and contains the following code:

`if ((millis() - hbGood_millis) > ((2*timeout_secs) * 1000)) {
       mavGood = false; 
       hbGood = false;         
     }`

This code is ran regardless of the protocol selected in config.h file.

The problem is that the function LTM.ino does not upgrade the variable **hbGood_millis**, so the variable **hbGood**  will always be **false**.

I thought of 3 possible solutions:
1)  in the file AntTrack.ino, row 978, the function `CheckStatusAndTimeouts(); // and service status LED` , don't call the function if we are using LTM mode (check variable PROTOCOL);

2) same thing in the file Utilities.ino, we don't run the if statement mentioned before if we are using LTM mode  (check variable PROTOCOL);

3) in the file LTM, after the while cycle: ` while (iLth==0) {
 //   CheckStatusAndTimeouts();
    iLth=Serial1.available();
  } `
add `hbGood_millis = millis();`

I chose the 3th option and it is working correctly. Is it the best solution?

### Second problem
I don't know why, but if I lost the signal (e.g. disconnecting the battery), lat and lon received values are still not zero. This will cause the variable `gpsGood` to remain `True` even though the aircraft's FC (and GPS also) was turned off. The other variables, however, immediately become zero (e.g. **iSat** ).

Possible solution:
In file LTM.ino (row 133-136) change
`  if ((!(Lat == 0)) && (!(Lon == 0))) {
    gpsGood=1;
    new_GPS_data = true;
  }`
to
`  if ((!(Lat == 0)) && (!(Lon == 0))  && (iSat > 0) ) {
    gpsGood=1;
    new_GPS_data = true;
  }`

zs6buj, 2024-02-07:
[casfra96](https://github.com/casfra96)

Thank you for the valuable feedback. Your diagnosis of the LTM issue was spot on. I added 

`hbGood_millis = millis();`

into the the condition below

`if ((!(Lat == 0)) && (!(Lon == 0))) {}`

and also set new_GPS_data = false when the gps times out.

New version 2.20.4 uploaded on GitHub. I don't have the means to test right now, so your further feedback would be appreciated.

Eric zs6buj

casfra96, 2024-02-07:
When i power off the aircraft, my TX16S continues to output the last detected latitude and longitude.

This will cause to pass the if statement: `if ((!(Lat == 0)) && (!(Lon == 0))) {`.

I don't know if this problem only affects me, but I solved changing this line in LTM.ino with:

`if ((!(Lat == 0)) && (!(Lon == 0))  && (iSat > 0) ) {`

zs6buj, 2024-02-07:
Glad you solved it. 😏

closed 2024-02-07 by zs6buj

casfra96, 2024-02-07:
I'm glad I could help :smile: