zs6buj/AntTracker #63: Help getting serial telemetry on bluepill board

brunoproza, 2024-02-11:
Hi @zs6buj 

Firstly I want to thank you for this project.

But I'm facing 2 problems (in this topic I'll only address 1, the other is with another card so I'll open another topic)

My senate.
I designed the antenna to use with my Radiomaster TX16s

In the TX16s I placed a bluetooth module (hc05 master mode) through the UART output

In the antenna I use the bluepill board and I have another HC05 (slave mode).

I already did the test using the 2 hc05 and both are "talking"

also test connecting the cell phone via Bluetooth to the TX16S and I can read the telemetry using the telemetry viewer app

but for some reason the tracker doesn't understand this telemetry.

My link system is crossfire using mavlink, and my airplane run Inav.

see below that the board turns on, starts, identifies what has something in serial using baud rate 115200 (which is the connection speed between the 2 HC05) but after that the tracker screen shows "no telemetry" several times

18:55:11.473 -> Starting AntTrack version:2.20.04
18:55:11.473 -> Default I2C pins are defined in Wire.h
18:55:11.473 -> Display support activated: Landscape 
18:55:11.473 -> 64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
18:55:11.515 -> Target Board = 1  STM32F1xx
18:55:11.515 -> Serial Telemetry In
18:55:11.547 -> Selected protocol is AUTO
18:55:11.547 -> headingsource = 2 FC Compass
18:55:11.579 -> Testing Servos
18:55:22.618 -> Serial port rx pin -61 is IDLE_HIGH, regular rx polarity retained
18:55:22.618 -> autoBaud - sensing pin 195
18:55:22.890 -> Serial input baud rate detected is 115200 b/s

Any idea what might solve it?

in advance, thank you very much!

zs6buj, 2024-02-12:
Hi Bruno, I suggest you select the crsf or mavlink protocol, and don't use Auto. Also try selecting a fixed baud rate. You could also activate any of the debug macros that look potentially helpful. I can't remember the exact ones right now.

I'm travelling for the next few days, and unfortunately won't always have internet.

brunoproza, 2024-02-12:
Thank you, I will carry out tests with CRSF and Mavlink.

I'll also look at the debug options, and get back to you.

but I'm thinking about changing the project to ESP32 (but I'm having errors, I'll cover that in another topic soon)

brunoproza, 2024-02-12:
I tested switching to protocol 9

#define PROTOCOL 9 //CRFS/ELRS

but I get the error below

\config.h:873:22: error: conflicting declaration 'HardwareSerial inSerial'

\config.h:864:20: note: previous declaration as 'SoftwareSerial inSerial'

zs6buj, 2024-02-19:
If I remember correctly, you can select a board variant with larger memory in the IDE, before you try to compile.

zs6buj, 2024-03-11:
Stale?

closed 2024-03-11 by zs6buj