zs6buj/AntTracker #64: ESP32 Dev Module - Help

brunoproza, 2024-02-12:
I'm also trying to put the code in an ESP32 Dev Module (NodeMCU-ESP32 DevKITV1)

I placed all the libraries (from the ALL folder and also from the ESP32 folder), in my IDE I selected the ESP32 Dev Module board

To install the board I used the link https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json and installed ESP32 Boards (by Espressif System) version 2.0.14

And just for a check (of the libraries) I ask him to compile the code, but even without any changes I receive several errors, I don't know where I'm going wrong in my IDE, is there any other option besides selecting the COM port and the board version ?

I also tested by editing what I wanted, such as wifi telemetry, and I also receive several missing file errors, below I leave the IDE errors for each of the scenarios.

1 - Errors without any changes just the board selection in the IDE

> In file included from C:\Users\bruno.roza\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/esp32-hal.h:87,
                 from C:\Users\bruno.roza\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/Arduino.h:36,
                 from C:\Users\bruno.roza\AppData\Local\Temp\arduino\sketches\2FADA379EA7959E8A65BE99444603820\sketch\AntTrack.ino.cpp:1:
C:\Users\bruno.roza\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/esp32-hal-spi.h:49:19: error: expected unqualified-id before numeric constant
 #define SPI_MODE0 0
                   ^
c:\Users\bruno.roza\OneDrive - Programmers Beyond IT\Documentos\Arduino\libraries\SPI/SPI.h:39:15: note: in expansion of macro 'SPI_MODE0'
 const uint8_t SPI_MODE0 = 0x00; ///<  CPOL: 0  CPHA: 0
               ^~~~~~~~~
C:\Users\bruno.roza\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/esp32-hal-spi.h:50:19: error: expected unqualified-id before numeric constant
 #define SPI_MODE1 1
                   ^
c:\Users\bruno.roza\OneDrive - Programmers Beyond IT\Documentos\Arduino\libraries\SPI/SPI.h:40:15: note: in expansion of macro 'SPI_MODE1'
 const uint8_t SPI_MODE1 = 0x01; ///<  CPOL: 0  CPHA: 1
               ^~~~~~~~~
C:\Users\bruno.roza\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/esp32-hal-spi.h:51:19: error: expected unqualified-id before numeric constant
 #define SPI_MODE2 2
                   ^
c:\Users\bruno.roza\OneDrive - Programmers Beyond IT\Documentos\Arduino\libraries\SPI/SPI.h:41:15: note: in expansion of macro 'SPI_MODE2'
 const uint8_t SPI_MODE2 = 0x10; ///<  CPOL: 1  CPHA: 0
               ^~~~~~~~~
C:\Users\bruno.roza\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/esp32-hal-spi.h:52:19: error: expected unqualified-id before numeric constant
 #define SPI_MODE3 3
                   ^
c:\Users\bruno.roza\OneDrive - Programmers Beyond IT\Documentos\Arduino\libraries\SPI/SPI.h:42:15: note: in expansion of macro 'SPI_MODE3'
 const uint8_t SPI_MODE3 = 0x11; ///<  CPOL: 1  CPHA: 1
               ^~~~~~~~~
In file included from C:\Users\bruno.roza\Downloads\AntTracker-master\AntTracker-master\source\stable\v2.20.04\AntTrack\config.h:592,
                 from C:\Users\bruno.roza\Downloads\AntTracker-master\AntTracker-master\source\stable\v2.20.04\AntTrack\AntTrack.ino:126:
c:\Users\bruno.roza\OneDrive - Programmers Beyond IT\Documentos\Arduino\libraries\Wire\src/Wire.h:44:5: error: 'i2c_t' does not name a type; did you mean 'ino_t'?
     i2c_t * i2c;
     ^~~~~
     ino_t
c:\Users\bruno.roza\OneDrive - Programmers Beyond IT\Documentos\Arduino\libraries\Wire\src/Wire.h:64:5: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?
     i2c_err_t last_error; // @stickBreaker from esp32-hal-i2c.h
     ^~~~~~~~~
     esp_err_t
c:\Users\bruno.roza\OneDrive - Programmers Beyond IT\Documentos\Arduino\libraries\Wire\src/Wire.h:87:5: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?
     i2c_err_t writeTransmission(uint16_t address, uint8_t* buff, uint16_t size, bool sendStop=true);
     ^~~~~~~~~
     esp_err_t
c:\Users\bruno.roza\OneDrive - Programmers Beyond IT\Documentos\Arduino\libraries\Wire\src/Wire.h:88:5: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?
     i2c_err_t readTransmission(uint16_t address, uint8_t* buff, uint16_t size, bool sendStop=true, uint32_t *readCount=NULL);
     ^~~~~~~~~
     esp_err_t
 exit status 1
 Compilation error: exit status 1

2 - Errors after selecting wifi telemetry

> c:\Users\bruno.roza\OneDrive - Programmers Beyond IT\Documentos\Arduino\libraries\WiFi\src\ETH.cpp:22:10: fatal error: eth_phy/phy.h: No such file or directory
 #include "eth_phy/phy.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Compilation error: exit status 1

In this second case, I already went through the library and added several files, but each time there are more and more errors :(

zs6buj, 2024-02-12:
Try this for the espressif boards

https://espressif.github.io/arduino-esp32/package_esp32_index.json

brunoproza, 2024-02-12:
ADD 
[image]

ESP Board install

[image]

Board Select

[image]

Still with errors :(

In file included from C:\Users\bruno.roza\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/esp32-hal.h:87,
                 from C:\Users\bruno.roza\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/Arduino.h:36,
                 from C:\Users\bruno.roza\AppData\Local\Temp\arduino\sketches\52CFA84D638C125815D69FD750DB93A6\sketch\AntTrack.ino.cpp:1:
C:\Users\bruno.roza\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/esp32-hal-spi.h:49:19: error: expected unqualified-id before numeric constant
 #define SPI_MODE0 0
                   ^
c:\Users\bruno.roza\OneDrive\Documentos\Arduino\libraries\SPI/SPI.h:39:15: note: in expansion of macro 'SPI_MODE0'
 const uint8_t SPI_MODE0 = 0x00; ///<  CPOL: 0  CPHA: 0
               ^~~~~~~~~
C:\Users\bruno.roza\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/esp32-hal-spi.h:50:19: error: expected unqualified-id before numeric constant
 #define SPI_MODE1 1
                   ^
c:\Users\bruno.roza\OneDrive\Documentos\Arduino\libraries\SPI/SPI.h:40:15: note: in expansion of macro 'SPI_MODE1'
 const uint8_t SPI_MODE1 = 0x01; ///<  CPOL: 0  CPHA: 1
               ^~~~~~~~~
C:\Users\bruno.roza\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/esp32-hal-spi.h:51:19: error: expected unqualified-id before numeric constant
 #define SPI_MODE2 2
                   ^
c:\Users\bruno.roza\OneDrive\Documentos\Arduino\libraries\SPI/SPI.h:41:15: note: in expansion of macro 'SPI_MODE2'
 const uint8_t SPI_MODE2 = 0x10; ///<  CPOL: 1  CPHA: 0
               ^~~~~~~~~
C:\Users\bruno.roza\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32/esp32-hal-spi.h:52:19: error: expected unqualified-id before numeric constant
 #define SPI_MODE3 3
                   ^
c:\Users\bruno.roza\OneDrive\Documentos\Arduino\libraries\SPI/SPI.h:42:15: note: in expansion of macro 'SPI_MODE3'
 const uint8_t SPI_MODE3 = 0x11; ///<  CPOL: 1  CPHA: 1
               ^~~~~~~~~
In file included from c:\Users\bruno.roza\OneDrive\Documentos\Arduino\libraries\Adafruit_SSD1306/Adafruit_SSD1306.h:41,
                 from C:\Users\bruno.roza\Downloads\AntTrack\config.h:589,
                 from C:\Users\bruno.roza\Downloads\AntTrack\AntTrack.ino:126:
c:\Users\bruno.roza\OneDrive\Documentos\Arduino\libraries\Wire\src/Wire.h:44:5: error: 'i2c_t' does not name a type; did you mean 'ino_t'?
     i2c_t * i2c;
     ^~~~~
     ino_t
c:\Users\bruno.roza\OneDrive\Documentos\Arduino\libraries\Wire\src/Wire.h:64:5: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?
     i2c_err_t last_error; // @stickBreaker from esp32-hal-i2c.h
     ^~~~~~~~~
     esp_err_t
c:\Users\bruno.roza\OneDrive\Documentos\Arduino\libraries\Wire\src/Wire.h:87:5: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?
     i2c_err_t writeTransmission(uint16_t address, uint8_t* buff, uint16_t size, bool sendStop=true);
     ^~~~~~~~~
     esp_err_t
c:\Users\bruno.roza\OneDrive\Documentos\Arduino\libraries\Wire\src/Wire.h:88:5: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?
     i2c_err_t readTransmission(uint16_t address, uint8_t* buff, uint16_t size, bool sendStop=true, uint32_t *readCount=NULL);
     ^~~~~~~~~
     esp_err_t

zs6buj, 2024-02-12:
I'm about to get on a 12hr flight. Tinker with it and see if you can sort it out. Sorry.

brunoproza, 2024-02-12:
There's no need to worry, just having done the project makes me grateful.

I'll test it and if I don't succeed when you get back or have time, I'll be grateful.

zs6buj, 2024-07-01:
I have posted V 2.21.07 |(under release_candidate) with fixes and some new features. Please could you check it when you get the chance.

zs6buj, 2024-07-01:
Also check out [https://github.com/zs6buj/CRSF_Utlilties](url) where you will find three utilities for CRSF to send telemetry from uart to UDP or BT, and read UDP. The utilities work with AntTrack.

brunoproza, 2024-07-01:
Thank you very much, I will test it today or tomorrow and get back to you

zs6buj, 2024-07-29:
Please reopen if you still need help :)

closed 2024-07-29 by zs6buj