zs6buj/AntTracker #66: Esp32 with Bluetooth telemetry

HKR1987, 2024-03-10:
I have opentx (open i6x) with elrs module. successful send telemetry over bluetooth to android telemetry app (telemetry viewer) but when i send it to anttracker its not working i got this output:

14:12:21.388 -> Starting AntTrack version:2.20.04
14:12:21.388 -> Setting up Wire I2C: SDA:21, SCL:22
14:12:21.495 -> Display support activated: Landscape 
14:12:21.495 -> 64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
14:12:21.495 -> Target Board = 3  ESP32 / Variant is Dev Module
14:12:21.495 -> FrSky Bluetooth In
14:12:21.495 -> Selected protocol is F.Port 2
14:12:21.495 -> headingsource = 2 FC Compass
14:12:21.495 -> Frs bluetooth master mode looking for slave name Frs2BT
14:12:22.630 -> Frs bluetooth connected!
14:12:22.663 -> EA<11<02<1F<05<67<FB<0D<6D<47<7C<00<02<3B<B0<03<E8<07<9E<EA<0C<14<C7<00<61<0A<00<08<02<DA<64<0B<E6<EA<0D<3A<EA<EE<10<00<00<75<4E<00<00<00<64<21<EA<0A<08<00<00<00<00<00<00<04<00<41<EA<0D<3A<EA<EE<10<00<00<75<4E<00<00<00<54<D7<EA<0C<14<C8<00<64<0B<00<08<02<DC<64<0A<62<EA<08<1E<00<7A<01<05<C4<36<3E<EA<0D<3A<EA<EE<10<00<00<75<4E<00<00<00<4B

There is data visble but something wrong...

zs6buj, 2024-03-10:
"Selected protocol is F.Port 2"

Could you please select CRSF protocol in the config.h. If I remember correctly, it is 9

HKR1987, 2024-03-11:
if  change to CRSF there it is, only warnings:
5:58:22.384 -> Starting AntTrack version:2.20.04
05:58:22.384 -> Setting up Wire I2C: SDA:21, SCL:22
05:58:22.496 -> Display support activated: Landscape 
05:58:22.496 -> 64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
05:58:22.496 -> Target Board = 3  ESP32 / Variant is Dev Module
05:58:22.496 -> FrSky Bluetooth In
05:58:22.496 -> Selected protocol is CRSF / ELRS
05:58:22.496 -> headingsource = 2 FC Compass
05:58:22.496 -> Testing Servos
05:58:33.011 -> Frs bluetooth master mode looking for slave name FSI6X
05:58:36.829 -> Frs bluetooth connected!
05:58:37.072 -> ASSERT_WARN(1 8), in lc_task.c at line 1408ASSERT_WARN(1 8), in lc_task.c at line 1408ASSERT_WARN(1 8), in lc_task.c at line 1408ASSERT_WARN(1 8

zs6buj, 2024-03-11:
Unfortunately I'm travelling, and working from memory, but I suspect the BT option was only ever added for Frsky and Mavlink, not elrs/crsf.  I need to look. Give me a few days and I'll take a look.

HKR1987, 2024-03-11:
ok, thanks

zs6buj, 2024-03-12:
I've added a generic BT version for CRSF, v 20.20.05 in the BETA folder, but I'm going to need some help testing it. Here where I am I have absolutely no boards or kit to test with!  You must add your information here, in config.h lies 125

```
#define BT_Mode  1           // Master Mode - active, initiate connection with slave (name)
//#define BT_Mode  2           // Slave Mode - passive, advertise our hostname & wait for master to connect to us
const char* BT_Slave_Name   =   "btslavename";
``` 

Also, when you must select Tools/Partition Scheme: "Minimal SPIFFS (1.9MB APP ...) in the Arduino IDE to make more space for the bluetooth code

zs6buj, 2024-03-14:
The assert warning is in one of the underlying c libraries. I would ignore it for now. The question is; why is the tracker Bluetooth failing to connect? It's not usually a source of problems.

zs6buj, 2024-03-14:
I found this link to the earning message. Very interesting.

https://github.com/espressif/arduino-esp32/issues/6193

zs6buj, 2024-03-14:
[image]

HKR1987, 2024-03-14:
Ok i will test it, but need time. Today my son born :)
I check it in free time and report here

zs6buj, 2024-03-14:
Congratulations my friend!

HKR1987, 2024-03-29:
finally find little free time to test it. After move closer the warning not show. Also i found some bugs:
line 891:  while(bt_connected). 
Wrong condition (without !)
Also i have problem when i try connect over name. But i change a little to make connection with MAC and it worked.
My changes:
```
  #if (Telemetry_In == 5)  // Generic BT for future
      #if (BT_Mode == 1)     // 1 master mode, connect to slave name
        log.printf("Bluetooth master mode, looking for slave name %s\n", BT_Slave_Name);          
        LogScreenPrintln("BT master cnnct");       
        inSerial.begin("Tracker", true);   
        inSerial.setPin("1234");          
      #else                  // 2 slave mode, advertise slave name
          log.printf("Bluetooth slave mode advertising slave name %s\n", mavBT_Slave_Name);            
          LogScreenPrintln("BT slave ready");   
          inSerial.begin(BT_Slave_Name);   
      #endif 
      
      bool bt_connected;
      uint8_t address[6] = {0x98, 0xDA, 0x50, 0x01, 0xC9, 0xF7};
      bt_connected = inSerial.connect(address);

      while(!bt_connected) {
        log.print(".");
        LogScreenPrintChar('.');  
        delay(1000);
        bt_connected = inSerial.connect(address);       
      }
      
      if(bt_connected) {
        btSuGood = true;
        log.println("Bluetooth connected!");
        LogScreenPrintln("BT connected!");
      } else {
        log.println("Bluetooth NOT connected!");
        LogScreenPrintln("BT NOT cnncted");    
      }
     #endif // end mavBT 
```

I get connection but nothing happen after this
```
11:25:02.640 -> Starting AntTrack version:2.20.05
11:25:02.640 -> Setting up Wire I2C: SDA:21, SCL:22
11:25:02.716 -> Display support activated: Landscape 
11:25:02.716 -> 64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
11:25:02.716 -> Target Board = 3  ESP32 / Variant is Dev Module
11:25:02.749 -> Expecting Bluetooth In
11:25:02.749 -> Selected protocol is CRSF / ELRS
11:25:02.749 -> headingsource = 2 FC Compass
11:25:02.749 -> Bluetooth master mode, looking for slave name FSI6X
11:25:08.968 -> .Bluetooth connected!
```

zs6buj, 2024-04-10:
Hello, I'm still travelling, but get back in about 10 days, and will look at this then. Thanks for the feedback.

HKR1987, 2024-04-12:
Hi,
i already make PR with some chenges which helps me make connection. Now i have output something like that:
14:01:20.108 -> .....
14:03:53.231 -> Starting AntTrack version:2.20.05
14:03:53.231 -> Setting up Wire I2C: SDA:21, SCL:22
14:03:53.352 -> Display support activated: Landscape 
14:03:53.352 -> 64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
14:03:53.352 -> Target Board = 3  ESP32 / Variant is Dev Module
14:03:53.352 -> Expecting Bluetooth In
14:03:53.352 -> Selected protocol is CRSF / ELRS
14:03:53.352 -> headingsource = 2 FC Compass
14:03:53.352 -> Bluetooth master mode, looking for slave name btslavename
14:04:25.855 -> .Bluetooth connected!
14:05:00.628 -> terseCRSF by zs6buj version:0.00.03
14:05:00.628 -> TELEMETRY_BUILD

Now need make some test with GPS lock :)
If i make something wrong with code fell free to make corrections

zs6buj, 2024-04-30:
Hey, I'm back in the home office, and I can look at this today. Thanks for your work so far. I need to set up a test source of BT elrs/crsf telem.

zs6buj, 2024-04-30:
My Kakute-h7 flight controller has BT, but it's flashed with APM and doesn't know what to do with it. I have an HC-05 somewhere in storage, about 5km away. I'll have to search :)

zs6buj, 2024-05-08:
I finally got this. 
V2.21.00  supports BT, WiFi or UART I/O for all protocols. I've tested CRSF/ELRS with BT, but from another uController, not from HC05. You might need to make your changes to support the MAC rather than slave name.

Please could you let me know how you go.

I have left these debug macros enabled in crsf.ino. You might want to comment them out later.

```
//=========  D E B U G   M A C R O S  ========
#define DEBUG_CRSF_GPS 
#define DEBUG_CRSF_BAT 
#define DEBUG_CRSF_ATTI
```

HKR1987, 2024-05-08:
Cool! Thank you. I test it and let you know :)

HKR1987, 2024-05-10:
I dont know what its wrong but in my case on terminal it stuck at TELEMETRY_BUILD and nothing happen.
On lcd the is showing info but only null
[image]

HKR1987, 2024-05-12:
How i can test if got good crsf frame?

zs6buj, 2024-05-13:
Yeah, it looks like you are not getting recognisable crsf frames, and yes, you can view incoming bytes. 

First, let me say that I test with a Kakute H7 V2 running Betaflight. 

[image]

Receiver is a CYCLONE Nano 2400RX ExpressLRS, and Transmitter/Controller is a Radiomaster TX16S with a HappyModel ES24TX Pro 2.4GHz 1000mW TX Module. This is successfully paired/(bound) and all works fine.

I take the test crsf telemetry signal off the flight controller, not the Transmitter, because the JR Bay is occupied on there, and it is more convenient to use the receiver side. Perhaps this is an issue?

[image]

Here the ESP32 converts the crsf telemetry to BT. I run the test Tracker firmware on a LILYGO® TTGO T-Display ESP32 1.14". Note that here we expect the CRSF signal to be inverted.

config.h line 357

`  bool in_invert = true;  `

If you are taking the crsf signal through to the transmitter, then into an HC05, perhaps the signal is idle high, or NOT inverted. So maybe try that.

If you want to print your incoming byte stream in hex, it's a little trickier. Find terseCRSF.cpp in the libraries, and on line 308 uncomment the line:

`      //printByte(b, ' ');    `

zs6buj, 2024-05-13:
Update:

I added this debug macro into terseCRSF.h v0.0.4

`#define SHOW_BYTE_STREAM`

HKR1987, 2024-05-13:
Okey i try that :) thanks
I have telemetry from inav and fc omnibus f4 pro via happymodel elrs connected to flysky i6x transmitter. And to transmitter port have connected BT module. I think this inverted option may help me... we will see

HKR1987, 2024-05-14:
ok i see the messages but something wrong. When i change to invert = false is the same. on ELRS i have selected protocol CRSF (not inverted). When i look the data its looking good.
I test it when not have GPS lock - that's why it may not work? or it should show the received data (link status, voltage, altitude etc)?
Here is dump from messages. i think it looking good. Could you take a look your way, what can be wrong?
```
08:43:32.273 -> EA 0C 14 D0 00 64 0A 00 08 02 DE 64 0B BD EA 07 21 57 41 49 54 00 AD EA 0D 3A EA EE 10 00 00 75 4E 00 00 00 8E A1 
EA 0C 14 D0 00 64 0A 00 08 02 DE 64 0B BD EA 04 07 00 00 23 EA 0D 3A EA EE 10 00 00 75 4E FF FF FF FA EC EA 11 02 00 00 00 00 00 00 00 00 00 00 00 00 03 E7 00 6E 
EA 0C 14 D0 00 64 0B 00 08 02 DF 64 09 D5 EA 0D 3A EA EE 10 00 00 75 4E 00 00 00 23 E8 EA 0D 3A EA EE 10 00 00 75 4E FF FF F7 36 11 EA 0A 08 00 6F 00 01 00 00 00 2E ED 
EA 0C 14 D0 00 64 0B 00 08 02 D9 64 0A DF EA 0D 3A EA EE 10 00 00 75 4E 00 00 08 21 CF EA 08 1E 01 7F 01 4B DC 01 DF 
EA 0C 14 D0 00 64 0B 00 08 02 E0 64 0B 44 EA 0D 3A EA EE 10 00 00 75 4E 00 00 01 10 BF EA 07 21 57 41 49 54 00 AD EA 0D 3A EA EE 10 00 00 75 4E 00 00 00 8A 5F 
EA 0C 14 D0 00 64 0B 00 08 02 DC 64 0A 2F EA 04 07 00 00 23 EA 0D 3A EA EE 10 00 00 75 4E FF FF FF A2 0A EA 11 02 00 00 00 00 00 00 00 00 00 00 00 00 03 E7 00 6E 
EA 0C 14 CF 00 63 0B 00 08 02 DE 64 09 C3 EA 0D 3A EA EE 10 00 00 75 4E FF FF F6 5A F4 EA 0A 08 00 6F 00 01 00 00 00 2E ED 
EA 0C 14 CF 00 63 0B 00 08 02 E3 64 0A 54 EA 0D 3A EA EE 10 00 00 75 4E 00 00 09 69 70 EA 08 1E 01 91 01 4B DB DE 4F 
EA 0C 14 D1 00 64 0B 00 08 02 E0 64 0B E6 EA 0D 3A EA EE 10 00 00 75 4E 00 00 01 BD F6 EA 07 21 57 41 49 54 00 AD 
EA 0C 14 D1 00 64 0B 00 08 02 DD 64 0C 8F EA 0D 3A EA EE 10 00 00 75 4E FF FF FF E2 97 EA 04 07 FF FF A2 EA 11 02 00 00 00 00 00 00 00 00 00 00 00 00 03 E7 00 6E EA 0D 3A EA EE 10 00 00 75 4E 00 00 00 0B 65 
EA 0C 14 CF 00 64 0B 00 08 02 DF 64 09 5C EA 0A 08 00 6F 00 01 00 00 00 2E ED EA 0D 3A EA EE 10 00 00 75 4E 00 00 00 50 29 
EA 0C 14 CF 00 64 0B 00 08 02 DF 64 0A F6 EA 08 1E 01 91 01 4B DB BB 5D EA 0D 3A EA EE 10 00 00 75 4E FF FF FF EC 3F 
EA 0C 14 D2 00 64 0B 00 08 02 DE 64 0B B8 EA 07 21 57 41 49 54 00 AD EA 0D 3A EA EE 10 00 00 75 4E 00 00 00 BB 7C 
EA 0C 14 D2 00 64 0B 00 08 02 E2 64 0B 06 EA 04 07 FF FF A2 EA 0D 3A EA EE 10 00 00 75 4E 00 00 00 0B 65 EA 11 02 00 00 00 00 00 00 00 00 00 00 00 00 03 E7 00 6E 
EA 0C 14 D3 00 64 0B 00 08 02 D4 64 09 50 EA 0D 3A EA EE 10 00 00 75 4E 00 00 00 64 21 EA 0D 3A EA EE 10 00 00 75 4E FF FF F7 A4 D3 EA 0A 08 00 6F 00 01 00 00 00 2E ED 
EA 0C 14 D3 00 64 0B 00 08 02 D8 64 0A 6F EA 0D 3A EA EE 10 00 00 75 4E 00 00 07 C9 59 EA 08 1E 01 91 01 3A DB 98 9F 
EA 0C 14 D0 00 64 0A 00 08 02 DF 64 0A EB EA 0D 3A EA EE 10 00 00 75 4E 00 00 01 4C A7 EA 07 21 57 41 49 54 00 AD
EA 0C 14 D0 00 64 0A 00 08 02 DD 64 0C B9 EA 04 07 FF FF A2 EA 0D 3A EA EE 10 00 00 75 4E FF FF FF EC 3F EA 11 02 00 00 00 00 00 00 00 00 00 00 00 00 03 E7 00 6E EA 0D 3A EA EE 10 00 00 75 4E FF FF FF F2 C5 
EA 0C 14 D1 00 64 0B 00 08 02 DD 64 0A 0E EA 0A 08 00 6F 00 01 00 00 00 2E ED EA 0D 3A EA EE 10 00 00 75 4E 00 00 00 55 02 
EA 0C 14 D1 00 64 0B 00 08 02 E0 64 0A 33 EA 08 1E 01 91 01 3A DB 87 B0 EA 0D 3A EA EE 10 00 00 75 4E 00 00 00 58 00 EA 0D 3A EA EE 10 00 00 75 4E FF FF F7 E0 B0 
EA 0C 14 D0 00 64 0B 00 08 02 E0 64 0B 44 EA 07 21 57 41 49 54 00 AD EA 0D 3A EA EE 10 00 00 75 4E 00 00 07 63 44 
EA 0C 14 D0 00 64 0B 00 08 02 DC 64 0B FA EA 04 07 FF FF A2 EA 0D 3A EA EE 10 00 00 75 4E 00 00 01 14 41 EA 11 02 00 00 00 00 00 00 00 00 00 00 00 00 03 E7 00 6E 
EA 0C 14 D0 00 64 0B 00 08 02 DC 64 0A 2F EA 0D 3A EA EE 10 00 00 75 4E 00 00 00 78 A4 EA 0A 08 00 6F 00 01 00 00 01 2E E6 
```

HKR1987, 2024-05-14:
ok, i will try it tommorow when be at home. I think that this is it :)

HKR1987, 2024-05-15:
ok, so... its working! :)
[image]

only i see the problem with flight mode but i dont know if this ruin something?
it show AIT? - should be WAIT...
```
08:56:00.875 -> Starting AntTrack version:2.21.00
08:56:00.875 -> Setting up Wire I2C: SDA:21, SCL:22
08:56:00.939 -> Display support activated: Landscape 
08:56:00.939 -> 64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
08:56:01.040 -> Target Board = 3  ESP32 / Variant is Dev Module
08:56:01.078 -> Expecting Bluetooth In
08:56:01.113 -> Selected protocol is CRSF / ELRS
08:56:01.113 -> headingsource = 2 FC Compass
08:56:01.146 -> Bluetooth master mode, looking for slave name "btslavename"
08:56:01.985 -> Bluetooth connected!
08:56:02.023 -> Initialising CRSF port
08:56:02.023 -> terseCRSF by zs6buj version:0.00.03
08:56:02.023 -> TELEMETRY_BUILD
08:56:02.055 -> ATTITUDE id:1E pitch:2.0deg  roll:2.3deg  yaw:307.0deg
08:56:02.254 -> FLIGHT_MODE id:21 lth:3 AIT?
08:56:02.663 -> CRSF_GPS:02 lat:0.0000000  lon:0.0000000  ground_spd:0.0km/hr  hdg:0.00deg  alt:65535m  sats:0
08:56:02.663 -> GPS lock good! Push set-home button (pin:12) anytime to start tracking 
08:56:02.908 -> BATTERY id:08 volts:0.0  amps:0.0  Ah_drawn:256000.0  remaining:  0%
08:56:03.023 -> ATTITUDE id:1E pitch:2.0deg  roll:2.3deg  yaw:307.0deg
08:56:03.255 -> FLIGHT_MODE id:21 lth:3 AIT?
```
now im need test it with servo maybe in weekend i get some free time for it.
So issue can be close now
one more question: how i can buy you virtual coffee? :)

zs6buj, 2024-05-15:
Haha. No need for coffee, but thank you. I'm pleased that it is coming together. You have helped me understand the iNav + Flysky i6x, and also we tested the BT together 

The "GPS lock good" looks suspect if latitude and longitude == 0.

Let's keep the issue open for now.

toofen, 2024-06-07:
I had a similar problem, the telemetry did not seem to be recognized by the tracker. 

I was using a wire for the telemetry, latest Inav, latest ELRS and latest Edge-Tx on a tx16s. Using a terminal i could see that the transmitter was outputting telemetry but as i said i could not get it to be recognized by the tracker. I then edited the  terseCRSF.h and changed CRSF_TEL_SYNC_BYTE to  0xEA like the example above and got it working! Some info was a bit off, like battery voltage (wrong decimal place) and amp-draw off by a factor of ten. Flightmode was also showing strange info just like above. 

I have not had time to really test that everyting actually works yet but i will get to that as soon as possible, right now im just happy to get some telemetry to the tracker working after spending too many hours chasing gremlings :)

zs6buj, 2024-06-07:
Thank you for the feedback. I think I might have fixed the flight mode. Could you ensure you use my latest terseCRSF,

HKR1987, 2024-06-07:
I also notice same. Not get working tracker but i got error FC GPS Timeout. Maybe there is something cut the data and thats why is problem with flight mode, voltage and this timeouts?

toofen, 2024-06-08:
A little more experimenting...

[image]
Here the alt displays wrong (more on that later) also the hdg shows wrong on the display.
[image]
Here you can seethat the volage and amp draw shows of by a factor of 10
[image]
[image]
Moved the airplane to see if the alt reading was wrong due to it showing minus -0.1 and its definetely better now. 
Interetingly the values that i get in Serial Monitor is spot on so i guess its just showing the wrong values on the display, except for CRSF cario that shows just HEX values

zs6buj, 2024-06-08:
Yes, this is great info, thank you!

"Moved the airplane to see if the alt reading was wrong due to it showing minus -0.1"

Yeah, what happens here is this: Once we have "home" altitude and the GPS alt drifts a little due to accuracy tolerance, our calc for alt above field can go negative. I should show negative numbers properly there.

I will try to look at this today, but wife has things arranged, lol.

zs6buj, 2024-06-08:
> I also notice same. Not get working tracker but i got error FC GPS Timeout. Maybe there is something cut the data and thats why is problem with flight mode, voltage and this timeouts?

Thanks. Could you please give me details. Versions, log, anything else.

toofen, 2024-06-08:
I was going to edit my original post but you were quick to respond! 
I also have some GPS Timeouts but the info still seems to get thru so i dont know what it could be, i will try to experiment a little more later and see if i can dig up something more. All right now tested with latest Beta anttracker and the latest TerseCRSF (i missed that bit in my first post), its hard to keep up with everyting :)

zs6buj, 2024-07-01:
I have posted V 2.21.07 |(under release_candidate) with fixes and some new features. Please could you check it when you get the chance.

zs6buj, 2024-07-01:
Also check out [https://github.com/zs6buj/CRSF_Utlilties](url) where you will find three utilities for CRSF to send telemetry from uart to UDP or BT, and read UDP. The utilities work with AntTrack.

HKR1987, 2024-07-01:
Cool! i will test when had some free time and let you know

toofen, 2024-07-05:
It compiles without problems but sadly i seem to  have fried my t-display unit somewhere along the way so i cant test it on the hardware right now, i ordered some new units and i will get back to testing as soon as they come in.

HKR1987, 2024-07-07:
I test version 2.21.07 and it wont work
on screen and uart output i have good position but tracker dont want turn around and tracking.
on screen i also noticed some errors with gps timeout
[image]

here is logs from uart (gps data is anonimized but it looks good for me - show correct location)
Altitude wont work below 0 and show 65535m

```
18:14:32.636 -> Starting antTrack version:2.21.07
18:14:32.636 -> Setting up Wire I2C: SDA:21, SCL:22
18:14:32.636 -> Display support activated: Landscape 
18:14:32.636 -> 64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
18:14:32.701 -> Target Board = 3  ESP32 / Variant is Dev Module
18:14:32.733 -> Expecting Bluetooth In
18:14:32.774 -> Selected protocol is CRSF
18:14:32.774 -> headingsource = 2 FC Compass
18:14:32.809 -> Testing Servos
18:14:43.353 -> Bluetooth master mode, looking for slave name "btslavename"
18:14:44.468 -> Bluetooth connected!
18:14:44.500 -> terseCRSF by zs6buj version:0.00.06
18:14:44.500 -> Telemetry Build. Expected source is EdgeTX/OpenTX
18:14:44.536 -> Good BT data stream
18:14:44.577 -> CRSF_BUF:len:11:EA 0A 08 00 6E 00 01 00 00 0D 2B 
18:14:44.577 -> BATTERY id:08 volts:110.0  amps:1.0  Ah_drawn:3371000.0  remaining: 43%
18:14:44.614 -> CRSF_BUF:len:13:EA 0C 14 CB 00 64 0B 00 08 02 D1 64 0A 
18:14:44.797 -> CRSF_BUF:len: 9:EA 08 1E 02 FF 00 F4 0D 17 
18:14:44.797 -> ATTITUDE id:1E pitch:4.4deg  roll:1.4deg  yaw:19.0deg
18:14:44.797 -> CRSF_BUF:len:13:EA 0C 14 CB 00 64 0B 00 08 02 D1 64 0A 
18:14:45.012 -> CRSF_BUF:len: 6:EA 05 21 4F 4B 00 
18:14:45.012 -> FLIGHT_MODE id:21 lth:2 OK�? motArmed:0
18:14:45.092 -> CRSF_BUF:len:13:EA 0C 14 CA 00 64 0B 00 08 02 D5 64 0C 
18:14:45.175 -> CRSF_BUF:len: 5:EA 04 07 FF F8 
18:14:45.207 -> CF_VARIO:len: 5:EA 04 07 FF F8 
18:14:45.278 -> CRSF_BUF:len:18:EA 11 02 22 22 22 22 22 22 22 22 00 00 17 5C 03 E7 0E 
18:14:45.278 -> CRSF_GPS:02 lat:52.2222222  lon:22.2222222  ground_spd:0.0km/hr  hdg:59.80deg  alt:65535m  sats:14  gpsfixGood:1
18:14:45.278 -> GPS lock good! Push set-home button (pin:12) anytime to start tracking 
18:14:45.421 -> CRSF_BUF:len:13:EA 0C 14 CA 00 64 0B 00 08 02 D4 64 0A 
18:14:45.522 -> CRSF_BUF:len:11:EA 0A 08 00 6E 00 01 00 00 0E 2B 
18:14:45.522 -> BATTERY id:08 volts:110.0  amps:1.0  Ah_drawn:3627000.0  remaining: 43%
18:14:45.629 -> CRSF_BUF:len:13:EA 0C 14 C8 00 64 0B 00 08 02 D3 64 0B 
18:14:45.708 -> ASSERT_WARN(1 8), in lc_task.c at line 1408CRSF_BUF:len: 9:EA 08 1E 02 FF 00 F4 0D 17 
18:14:45.779 -> ATTITUDE id:1E pitch:4.4deg  roll:1.4deg  yaw:19.0deg
18:14:45.822 -> CRSF_BUF:len:13:EA 0C 14 C8 00 64 0B 00 08 02 D9 64 0A 
18:14:46.012 -> CRSF_BUF:len: 6:EA 05 21 4F 4B 00 
18:14:46.012 -> FLIGHT_MODE id:21 lth:2 OK�? motArmed:0
18:14:46.012 -> gpsGood:1  gpsfixGood:1  lonGood:1  latGood:1  altGood:1  hdgGood:1  boxhdgGood:0 
18:14:46.012 -> Good flight computer GPS lock
18:14:46.079 -> CRSF_BUF:len:13:EA 0C 14 C9 00 64 0B 00 08 02 D9 64 0B 
18:14:46.079 -> CRSF_BUF:len: 5:EA 04 07 FF F8 
18:14:46.079 -> CF_VARIO:len: 5:EA 04 07 FF F8 
18:14:46.221 -> CRSF_BUF:len:18:EA 11 02 22 22 22 22 22 22 22 22 00 01 17 5C 03 E7 0E 
18:14:46.221 -> CRSF_GPS:02 lat:52.2222222  lon:22.2222222  ground_spd:0.1km/hr  hdg:59.80deg  alt:65535m  sats:14  gpsfixGood:1
18:14:46.360 -> CRSF_BUF:len:13:EA 0C 14 C9 00 64 0B 00 08 02 D6 64 0B 
18:14:46.478 -> CRSF_BUF:len:11:EA 0A 08 00 6E 00 01 00 00 0E 2B 
18:14:46.478 -> BATTERY id:08 volts:110.0  amps:1.0  Ah_drawn:3627000.0  remaining: 43%
18:14:46.605 -> CRSF_BUF:len:13:EA 0C 14 C8 00 64 0B 00 08 02 D7 64 0A 
18:14:46.712 -> CRSF_BUF:len: 9:EA 08 1E 02 FF 01 05 0D 05 
18:14:46.712 -> ATTITUDE id:1E pitch:4.4deg  roll:1.5deg  yaw:19.0deg
18:14:46.776 -> CRSF_BUF:len:13:EA 0C 14 C8 00 64 0B 00 08 02 D5 64 0A 
18:14:46.812 -> CRSF_BUF:len: 6:EA 05 21 4F 4B 00 
18:14:46.812 -> FLIGHT_MODE id:21 lth:2 OK�? motArmed:0
18:14:46.976 -> CRSF_BUF:len:13:EA 0C 14 C8 00 64 0B 00 08 02 D9 64 0B 
18:14:46.976 -> CRSF_BUF:len: 5:EA 04 07 FF F8 
18:14:46.976 -> CF_VARIO:len: 5:EA 04 07 FF F8 
18:14:47.252 -> CRSF_BUF:len:18:EA 11 02 22 22 22 22 22 22 22 22 00 03 17 5C 03 E7 0E 
18:14:47.252 -> CRSF_GPS:02 lat:52.2222222  lon:22.2222222  ground_spd:0.3km/hr  hdg:59.80deg  alt:65535m  sats:14  gpsfixGood:1
18:14:47.362 -> CRSF_BUF:len:13:EA 0C 14 C8 00 64 0B 00 08 02 D5 64 0A 
18:14:47.402 -> CRSF_BUF:len:11:EA 0A 08 00 6E 00 01 00 00 0E 2B 
18:14:47.402 -> BATTERY id:08 volts:110.0  amps:1.0  Ah_drawn:3627000.0  remaining: 43%
[... 62 lines cut ...]
18:15:17.155 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.155 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.155 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.155 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.155 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.155 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.155 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.188 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.188 -> CRSF_BUF:len:11:EA 0A 08 00 6E 00 01 00 00 0F 2B 
18:15:17.188 -> BATTERY id:08 volts:110.0  amps:1.0  Ah_drawn:3883000.0  remaining: 43%
18:15:17.188 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.188 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.188 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.188 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
[... cut ...]
18:15:17.220 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.220 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.220 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.220 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.220 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.220 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.251 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
18:15:17.251 ->  Bad alt! cur.alt=0 hom.alt=65535  Packet ignored
```

zs6buj, 2024-07-08:
Ok, thanks. Did you do this?

`18:14:45.278 -> GPS lock good! Push set-home button (pin:12) anytime to start tracking` 

This is a requirement if you are using :

`18:14:32.774 -> headingsource = 2 FC Compass`

Point the 'plane/drone in the default tracking direction, and tracker stores the FC compass reading as "home" direction

Not sure why no display tho

HKR1987, 2024-07-08:
i dont remember if i push the button with this test but if i have this:
#define home_decay_secs 60
it should set after 60 seconds...?
the display show correct coordinates but sometimes start showing errors like above

zs6buj, 2024-07-08:
` #define home_decay_secs 60`

This is something else, to do with when I write info to eeprom.

"the display show correct coordinates but sometimes start showing errors like above"

This is being caused by the long delay between GPS messages from Edge/TC box. There are many link statistics messages, and the GPS messages time out. You can change 

`    const uint8_t timeout_secs = 8;   `

HKR1987, 2024-07-08:
```
18:14:51.240 -> BATTERY id:08 volts:110.0  amps:1.0  Ah_drawn:3627000.0  remaining: 43%
```
volatge, amps and ah_drawn also looking wrong

zs6buj, 2024-07-08:
Yeah, I see that. Divided V and A by 10, and MhA by 1E3. Please could you try terseCRSF lib v0.0.7 when you get the chance.

HKR1987, 2024-07-08:
Ofcourse i test it and back with results

toofen, 2024-07-09:
So...I have it mostly sort of working now, on a newer Lilygo T-display-S3 of all things. Like i said i fried my original T-display and had a s3 laying unused so i gave it a go and with a bit of tinkering i got i working, its right now a bit of a botch job and changing the existing parameters for the T2 until i got it working. 

 I could not use the bluetooth connection due to it being only Bluetooth LE but i got the display working and it receiving telemetry thru UART. The only issue i have now is that i cant get my servos working. I had exactly the same problem with my original T-display before i fried it, and any pointers would be very helpful.

zs6buj, 2024-07-10:
Thanks for the test. Make sure you power the servos separately from the ESP32. I use a separate 5V regulator. Don't use the 5v pin on the esp32, it burns the voltage reg on the board, or burns off the trace. I'm not sure what the LEDC thing is. I'll try to look later.

toofen, 2024-07-10:
> Thanks for the test. Make sure you power the servos separately from the ESP32. I use a separate 5V regulator. Don't use the 5v pin on the esp32, it burns the voltage reg on the board, or burns off the trace. I'm not sure what the LEDC thing is. I'll try to look later.

Yes i have a separate bec for the servos.I also measured the output using i Lipo charger i have that can measure PWM signals and it shows nothing. Im pretty sure that it doesnt output any pwm signal. Based on some limited googling and guesses i think the LEDC problem and me not getting any pwm out is related, something with timers used to generate the pwm signal, but all the info i found went way over my knowledge level :)

HKR1987, 2024-07-10:
@toofen what esp variant you have in config? 5 or 6?

toofen, 2024-07-11:
> @toofen what esp variant you have in config? 5 or 6?

Right now im using a modified version of 6. I had to modify option 6 to get it working with my Lilygo T-Display-S3

zs6buj, 2024-07-12:
The datsheet says you can map any pins for motor control PWM.  [https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf](url)

[image]

> 3.5.13 Motor Control PWM (MCPWM)
> ESP32-S3 integrates two MCPWM that can be used to drive digital motors and smart light. Each MCPWM
> peripheral has one clock divider (prescaler), three PWM timers, three PWM operators, and a capture module.
> PWM timers are used for generating timing references. The PWM operators generate desired waveform based
> on the timing references. Any PWM operator can be configured to use the timing references of any PWM timers.
> Different PWM operators can use the same PWM timer’s timing references to produce related PWM signals.
> PWM operators can also use different PWM timers’values to produce the PWM signals that work alone.
> Different PWM timers can also be synchronized together

I used these pin for another project, but no specific PWM signals.

```
 #if (ESP32_Variant == 8)          
/* 
  Lilygo T-Display-S3 ESP32-S3 1.9 in ST7789V LCD no touch screen    
  FC from wifi, mav and s.port to GCS
  see C:\Users\erics\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\variants\lilygo_t_display_s3\pins_arduino.h                                
  uart0(flash & monitor), uart1 and uart2(map pins)   
  static const uint8_t TX = 43;       // u0txd default
  static const uint8_t RX = 44;       // u0rxd
*/
    #define MavStatusLed     1        // No onboard LED, choose a GPIO
    #define InvertMavLed   false     
    #define BufStatusLed    -1        // none  
    #define fc_rxPin        16        // u2rxd - map a spare pin
    #define fc_txPin        21        // u2txd - map a spare pin
    #define fr_rxPin        18        // FPort u1rxd - map a spare pin
    #define fr_txPin        17        // FPort u1txd - map a spare pin
    #define sbus_rxPin      -1        // not used - don't care
    #define sbus_txPin      -1        // not used
    #define startWiFiPin    -1        // not used
    #define resetEepromPin  -1        // not used  non digital touch pin  
    #define gs_rxPin        18        // GCS u1rxd default pins, still mapped
    #define gs_txPin        17        // GCS u1txd default pins, still mapped
    #if !defined displaySupport       // I2C OLED board is built into Heltec WiFi Kit 32
      #define displaySupport
    #endif  
    #define ST7789V_Display         // 170 x 320 dot 262K Color TFT LCD
    #define SCR_ORIENT   1          // 1 Landscape or 0 Portrait 
    /* Below please choose either Touch pin-pair or Digital pin-pair for display scrolling
     *  Pin == -1 means the pin-pair is not used
     */ 
    #define Pup            0        // "BUTTON_1" to scroll the display up     LOW=Press
    #define Pdn           14        // "BUTTON_2" to scroll the display down   LOW=Press   
    #define Tup           -1        // 33 Touch pin to scroll the display up
    #define Tdn           -1        // 32 Touch pin to scroll the display down   
    //#define TOUCH_CS       3      // LCD is not touch screen
    
    /*  
     Apparently unused pins right now 
                        01
                        02
                        03
                        10
                        11X
                        12X
                        13
                        16X
                        21X
                        43 u0txd default
                        44 u0rxd default
       
     ST7789V Display reserved pins
      LCD_Power_on      15
      LCD_BL            38 Backlight
      LCD_D0            39
      LCD_D1            40 
      LCD_D2            41
      LCD_D3            42
      LCD_D4            45 
      LCD_D5            46
      LCD_D6            47
      LCD_D7            48
      LCD_WR            08
      LCD_RD            09
      LCD_DC            07
      LCD_CS            06
      LCD_RES           05                 
    */
  #endif  
```

zs6buj, 2024-07-12:
You could also search to pdf for "LED PWM"

toofen, 2024-07-12:
after loads of reading and scratching my head i managed to get the LEDC error to go away. It was due to me using ESP version 2.0.14 due to TFT_eSPI not working on any higher version, but i found a work-around for that in https://github.com/Bodmer/TFT_eSPI/issues/3355#issuecomment-2169396713

and using that in combination with updating ESP to version 3.0.2 made the LEDC errors go away. I still cant get my servos to work but now i have something to work with at least.

zs6buj, 2024-07-13:
Good work, and thanks for the feedback!

HKR1987, 2024-07-15:
ok i make test again with push home button. Version 2.21.07 (terseCRSF v0.0.7)
The values looks good, but tracker wont track. I have set flag TEST_SERVOS and on starup my servos working.
After push home button nothing more is happen. What else can i check to find problem?
There is also sometimes show FC GPS timeout.

[image]
[image]
```
18:23:42.504 -> Starting antTrack version:2.21.07
18:23:42.504 -> Setting up Wire I2C: SDA:21, SCL:22
18:23:42.504 -> Display support activated: Landscape 
18:23:42.504 -> 64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
18:23:42.578 -> Target Board = 3  ESP32 / Variant is Dev Module
18:23:42.643 -> Expecting Bluetooth In
18:23:42.643 -> Selected protocol is CRSF
18:23:42.679 -> headingsource = 2 FC Compass
18:23:42.679 -> Testing Servos
18:23:53.201 -> Bluetooth master mode, looking for slave name "btslavename"
18:23:56.738 -> Bluetooth connected!
18:23:56.823 -> terseCRSF by zs6buj version:0.00.07
18:23:56.823 -> Telemetry Build. Expected source is EdgeTX/OpenTX
18:23:56.823 -> Good BT data stream
18:23:56.857 -> CRSF_BUF:EA 0C 14 D1 00 64 0B 00 08 02 E2 64 0B 
18:23:57.006 -> CRSF_BUF:EA 04 07 FF FF 
18:23:57.006 -> CF_VARIO:EA 04 07 FF FF 
18:23:57.123 -> CRSF_BUF:EA 0C 14 D1 00 64 0B 00 08 02 E1 64 0B 
18:23:57.153 -> CRSF_BUF:EA 11 02 12 34 56 78 12 34 56 78 00 02 32 64 03 E7 11 
18:23:57.153 -> CRSF_GPS:02 lat:52.1234567  lon:22.1234567  ground_spd:0.2km/hr  hdg:129.00deg  alt:-1m  sats:17  gpsfixGood:1
18:23:57.189 -> GPS lock good! Push set-home button (pin:12) anytime to start tracking 
18:23:57.296 -> CRSF_BUF:EA 0A 08 00 6D 00 01 00 00 0D 28 
18:23:57.296 -> BATTERY id:08 volts:10.9  amps:0.1  Ah_drawn:3368.0  remaining: 40%
18:23:57.296 -> CRSF_BUF:EA 0C 14 D3 00 64 0B 00 08 02 E1 64 0A 
18:23:57.507 -> CRSF_BUF:EA 08 1E 00 45 FF A9 3F E1 
18:23:57.545 -> ATTITUDE id:1E pitch:0.4deg  roll:-0.5deg  yaw:93.0deg
18:23:57.545 -> CRSF_BUF:EA 0C 14 D3 00 64 0B 00 08 02 E0 64 0A 
18:23:57.729 -> CRSF_BUF:EA 05 21 4F 4B 00 
18:23:57.729 -> FLIGHT_MODE id:21 lth:2 OK�? motArmed:0
18:23:57.729 -> gpsGood:1  gpsfixGood:1  lonGood:1  latGood:1  altGood:1  hdgGood:1  boxhdgGood:0 
18:23:57.729 -> Good flight computer GPS lock
18:23:57.808 -> CRSF_BUF:EA 0C 14 CF 00 63 0B 00 08 02 E2 64 0C 
18:23:57.889 -> ASSERT_WARN(1 8), in lc_task.c at line 1409CRSF_BUF:EA 04 07 FF FF 
18:23:57.922 -> CF_VARIO:EA 04 07 FF FF 
18:23:57.991 -> CRSF_BUF:EA 0C 14 CF 00 63 0B 00 08 02 CA 64 0B 
18:23:58.101 -> CRSF_BUF:EA 11 02 12 34 56 78 12 34 56 78 00 01 32 64 03 E7 11 
18:23:58.139 -> CRSF_GPS:02 lat:52.1234567  lon:22.1234567  ground_spd:0.1km/hr  hdg:129.00deg  alt:-1m  sats:17  gpsfixGood:1
18:23:58.212 -> CRSF_BUF:EA 0A 08 00 6D 00 01 00 00 0D 28 
18:23:58.212 -> BATTERY id:08 volts:10.9  amps:0.1  Ah_drawn:3368.0  remaining: 40%
18:23:58.347 -> CRSF_BUF:EA 0C 14 CF 00 64 0B 00 08 02 DB 64 09 
18:23:58.451 -> CRSF_BUF:EA 08 1E 00 45 FF BB 3F E1 
18:23:58.451 -> ATTITUDE id:1E pitch:0.4deg  roll:-0.4deg  yaw:93.0deg
18:23:58.521 -> CRSF_BUF:EA 0C 14 CF 00 64 0B 00 08 02 E0 64 0A
18:23:58.738 -> CRSF_BUF:EA 05 21 4F 4B 00 
18:23:58.738 -> FLIGHT_MODE id:21 lth:2 OK�? motArmed:0
18:23:58.738 -> CRSF_BUF:EA 0C 14 CD 00 64 0B 00 08 02 DF 64 0B 
18:23:58.923 -> CRSF_BUF:EA 04 07 00 00 
18:23:58.923 -> CF_VARIO:EA 04 07 00 00 
18:23:58.962 -> CRSF_BUF:EA 0C 14 CD 00 64 0B 00 08 02 D9 64 0B 
18:23:59.115 -> CRSF_BUF:EA 11 02 12 34 56 78 12 34 56 78 00 01 32 64 03 E7 11 
18:23:59.115 -> CRSF_GPS:02 lat:52.1234567  lon:22.1234567  ground_spd:0.1km/hr  hdg:129.00deg  alt:-1m  sats:17  gpsfixGood:1
18:23:59.189 -> CRSF_BUF:EA 0A 08 00 6D 00 01 00 00 0D 28 
18:23:59.189 -> BATTERY id:08 volts:10.9  amps:0.1  Ah_drawn:3368.0  remaining: 40%
18:23:59.382 -> CRSF_BUF:EA 0C 14 CD 00 64 0B 00 08 02 D0 64 0A 
18:23:59.414 -> CRSF_BUF:EA 08 1E 00 45 FF A9 3F E1 
18:23:59.414 -> ATTITUDE id:1E pitch:0.4deg  roll:-0.5deg  yaw:93.0deg
18:23:59.539 -> CRSF_BUF:EA 0C 14 CD 00 64 0B 00 08 02 D5 64 0A 
18:23:59.672 -> CRSF_BUF:EA 05 21 4F 4B 00 
18:23:59.672 -> FLIGHT_MODE id:21 lth:2 OK�? motArmed:0
18:23:59.738 -> CRSF_BUF:EA 0C 14 CC 00 64 0B 00 08 02 E0 64 0B 
18:23:59.915 -> CRSF_BUF:EA 04 07 00 00 
18:23:59.915 -> CF_VARIO:EA 04 07 00 00 
[... 122 lines cut ...]
18:24:07.979 -> CRSF_BUF:EA 0C 14 D0 00 64 0B 00 08 02 E0 64 0A 
18:24:08.051 -> CRSF_BUF:EA 08 1E 00 57 FF BB 3F E1 
18:24:08.093 -> ATTITUDE id:1E pitch:0.5deg  roll:-0.4deg  yaw:93.0deg
18:24:08.196 -> CRSF_BUF:EA 0C 14 D0 00 64 0B 00 08 02 E0 64 0A 
18:24:08.336 -> CRSF_BUF:EA 05 21 4F 4B 00 
18:24:08.336 -> FLIGHT_MODE id:21 lth:2 OK�? motArmed:0
18:24:08.371 -> CRSF_BUF:EA 0C 14 D1 00 63 0B 00 08 02 E0 64 0A 
18:24:08.559 -> CRSF_BUF:EA 04 07 FF FC 
18:24:08.559 -> CF_VARIO:EA 04 07 FF FC 
18:24:08.559 -> CRSF_BUF:EA 0C 14 D1 00 63 0B 00 08 02 DF 64 0B 
18:24:08.783 -> CRSF_BUF:EA 11 02 12 34 56 78 12 34 56 78 00 01 32 64 03 E7 11 
18:24:08.783 -> CRSF_GPS:02 lat:52.1234567  lon:22.1234567  ground_spd:0.1km/hr  hdg:129.00deg  alt:-1m  sats:17  gpsfixGood:1
18:24:08.783 -> CRSF_BUF:EA 0A 08 00 6D 00 01 00 00 0D 28 
18:24:08.783 -> BATTERY id:08 volts:10.9  amps:0.1  Ah_drawn:3368.0  remaining: 40%
18:24:08.783 -> CRSF_BUF:EA 0C 14 D1 00 64 0B 00 08 02 DF 64 0A 
18:24:08.980 -> CRSF_BUF:EA 08 1E 00 57 FF BB 3F E1 
18:24:08.980 -> ATTITUDE id:1E pitch:0.5deg  roll:-0.4deg  yaw:93.0deg
18:24:09.164 -> CRSF_BUF:EA 0C 14 D1 00 64 0B 00 08 02 DE 64 0A 
18:24:09.164 -> CRSF_BUF:EA 05 21 4F 4B 00 
18:24:09.164 -> FLIGHT_MODE id:21 lth:2 OK�? motArmed:0
18:24:09.345 -> CRSF_BUF:EA 0C 14 CE 00 64 0B 00 08 02 E0 64 0A 
18:24:09.345 -> CRSF_BUF:EA 04 07 FF FB 
18:24:09.345 -> CF_VARIO:EA 04 07 FF FB 
18:24:09.562 -> CRSF_BUF:EA 0C 14 CE 00 64 0B 00 08 02 CB 64 0C 
```

zs6buj, 2024-07-16:
Since we see this:

`Static home location set to Lat = 52.1234567 Lon = 22.1234567 Alt = 0 Hdg = 93`

we should start tracking provided that the 'plane/drone has moved more than minDist, which is presently 4m. I can't see from the logs if latitude or longitude have changed.

For debugging I recommend that you activate these macros:

#define DEBUG_AzEl
#define DEBUG_Servos

HKR1987, 2024-07-24:
ok, i made some test and something is still wrong. Test servo working, after set home location and go further with drone the tracker set in one position only, not moving  after drone (16:34:18.417  in log)
there is also some GPS timeouts - maybe this causes this problem?
v2.21.07
with terseCRSF v0.00.07
log with anonimized GPS coords but it showing good location
```
16:33:46.028 -> 
16:33:46.028 -> Starting antTrack version:2.21.07
16:33:46.028 -> Setting up Wire I2C: SDA:21, SCL:22
16:33:46.028 -> Display support activated: Landscape 
16:33:46.028 -> 64x128  text_size=1  char_w_px=6  char_h_px=8  scr_h_ch=8  scr_w_ch=21
16:33:46.094 -> Target Board = 3  ESP32 / Variant is Dev Module
16:33:46.162 -> Expecting Bluetooth In
16:33:46.193 -> Selected protocol is CRSF
16:33:46.193 -> headingsource = 2 FC Compass
16:33:46.193 -> moveServos() az=90  el=0
16:33:46.193 -> az reversed Servo write az=90  el=0
16:33:46.238 -> Testing Servos
16:33:47.717 -> moveServos() az=0  el=0
16:33:47.763 -> az reversed Servo write az=180  el=0
16:33:49.225 -> moveServos() az=90  el=0
16:33:49.225 -> az reversed Servo write az=90  el=0
16:33:50.748 -> moveServos() az=90  el=90
16:33:50.748 -> az reversed Servo write az=90  el=90
16:33:52.255 -> moveServos() az=90  el=180
16:33:52.255 -> az reversed Servo write az=90  el=180
16:33:53.742 -> moveServos() az=90  el=0
16:33:53.742 -> az reversed Servo write az=90  el=0
16:33:55.254 -> moveServos() az=180  el=0
16:33:55.254 -> az reversed Servo write az=0  el=0
16:33:56.717 -> moveServos() az=90  el=0
16:33:56.750 -> az reversed Servo write az=90  el=0
16:33:56.750 -> moveServos() az=90  el=0
16:33:56.750 -> az reversed Servo write az=90  el=0
16:33:56.750 -> Bluetooth master mode, looking for slave name "btslavename"
16:34:00.263 -> Bluetooth connected!
16:34:00.336 -> terseCRSF by zs6buj version:0.00.07
16:34:00.339 -> Telemetry Build. Expected source is EdgeTX/OpenTX
16:34:00.339 -> Good BT data stream
16:34:00.383 -> CRSF_BUF:EA 0A 08 00 6D 00 01 00 00 12 25 
16:34:00.496 -> CRSF_BUF:EA 0C 14 CE 00 64 0A 00 08 02 DC 64 0A 
16:34:00.608 -> CRSF_BUF:EA 08 1E 00 11 03 D1 BC A5 
16:34:00.681 -> CRSF_BUF:EA 0C 14 CE 00 64 0B 00 08 02 D7 64 0A 
16:34:00.859 -> CRSF_BUF:EA 05 21 4F 4B 00 
16:34:00.859 -> CRSF_BUF:EA 0C 14 CE 00 64 0B 00 08 02 DA 64 0A 
16:34:00.897 -> CRSF_BUF:EA 04 07 00 09 
16:34:00.897 -> CF_VARIO:EA 04 07 00 09 
16:34:01.115 -> GPS lock good! Push set-home button (pin:12) anytime to start tracking 
16:34:01.228 -> CRSF_BUF:EA 0C 14 CD 00 64 0A 00 08 02 D6 64 0B 
16:34:01.268 -> CRSF_BUF:EA 0A 08 00 6D 00 01 00 00 12 25 
16:34:01.442 -> CRSF_BUF:EA 0C 14 CD 00 64 0A 00 08 02 DB 64 0B 
16:34:01.481 -> CRSF_BUF:EA 08 1E 00 00 03 D1 BC B6 
16:34:01.629 -> CRSF_BUF:EA 0C 14 CD 00 64 0B 00 08 02 D7 64 0A 
16:34:01.661 -> CRSF_BUF:EA 05 21 4F 4B 00 
16:34:01.846 -> CRSF_BUF:EA 0C 14 CD 00 64 0B 00 08 02 DA 64 0A 
16:34:01.922 -> CRSF_BUF:EA 04 07 00 09 
16:34:01.922 -> CF_VARIO:EA 04 07 00 09 
16:34:02.098 -> CRSF_BUF:EA 0C 14 CE 00 63 0A 00 08 02 D9 64 0B 
16:34:02.280 -> CRSF_BUF:EA 0A 08 00 6D 00 01 00 00 12 25 
16:34:02.314 -> CRSF_BUF:EA 0C 14 CE 00 63 0A 00 08 02 DB 64 0B 
16:34:02.496 -> CRSF_BUF:EA 08 1E 00 00 03 BF BC B6 
16:34:02.614 -> CRSF_BUF:EA 0C 14 CE 00 64 0B 00 08 02 D7 64 0A 
16:34:02.726 -> CRSF_BUF:EA 05 21 4F 4B 00 
16:34:02.797 -> CRSF_BUF:EA 0C 14 CE 00 64 0B 00 08 02 DB 64 0B 
16:34:02.925 -> CRSF_BUF:EA 04 07 00 07 
16:34:02.925 -> CF_VARIO:EA 04 07 00 07 
16:34:03.097 -> CRSF_BUF:EA 0C 14 CF 00 63 0B 00 08 02 DC 64 0B 
16:34:03.259 -> CRSF_BUF:EA 0A 08 00 6D 00 01 00 00 12 25 
16:34:03.330 -> CRSF_BUF:EA 0C 14 CF 00 63 0B 00 08 02 DE 64 0B 
16:34:03.501 -> CRSF_BUF:EA 08 1E 00 00 03 BF BC C8 
[... 440 lines cut ...]
16:34:48.300 -> CRSF_BUF:EA 04 07 FF 8A 
16:34:48.300 -> CF_VARIO:EA 04 07 FF 8A 
16:34:48.482 -> CRSF_BUF:EA 0C 14 CC 00 64 0A 00 08 02 DE 64 0A 
16:34:48.659 -> CRSF_BUF:EA 0A 08 00 6C 00 01 00 00 14 24 
16:34:48.691 -> CRSF_BUF:EA 0C 14 CF 00 64 0B 00 08 02 DA 64 0A 
16:34:48.869 -> CRSF_BUF:EA 08 1E FF 1E 02 62 88 49 
16:34:48.902 -> CRSF_BUF:EA 0C 14 CF 00 64 0B 00 08 02 D9 64 0A 
16:34:48.969 -> CRSF_BUF:EA 05 21 4F 4B 00 
16:34:48.969 -> gpsGood:1  gpsfixGood:1  lonGood:1  latGood:1  altGood:1  hdgGood:1  boxhdgGood:0 
16:34:49.003 -> Good flight computer GPS lock
16:34:49.131 -> CRSF_BUF:EA 0C 14 CD 00 63 0A 00 08 02 D6 64 0B 
16:34:49.166 -> CRSF_BUF:EA 04 07 00 0B 
16:34:49.210 -> CF_VARIO:EA 04 07 00 0B 
16:34:49.385 -> CRSF_BUF:EA 0C 14 CD 00 63 0A 00 08 02 CC 64 0A 
16:34:49.538 -> CRSF_BUF:EA 0A 08 00 6C 00 01 00 00 14 24 
16:34:49.681 -> CRSF_BUF:EA 0C 14 C0 00 64 0B 00 08 02 CA 64 0A 
16:34:49.751 -> CRSF_BUF:EA 08 1E 00 00 01 5D 87 77 
16:34:49.865 -> CRSF_BUF:EA 0C 14 C0 00 64 0B 00 08 02 CF 64 0A 
16:34:49.976 -> CRSF_BUF:EA 05 21 4F 4B 00 
16:34:50.088 -> CRSF_BUF:EA 0C 14 C0 00 64 0A 00 08 02 CA 64 0A 
16:34:50.168 -> CRSF_BUF:EA 04 07 00 12 
16:34:50.168 -> CF_VARIO:EA 04 07 00 12 
16:34:50.446 -> CRSF_BUF:EA 0C 14 C0 00 64 0A 00 08 02 D8 64 09 

```

zs6buj, 2024-07-24:
The serial monitor is very slow and adding to the timeout problem. Could you go to your terseCRSF library, to terseCRSF.h and change these debug settings. We don't need to see the buffers anymore

```
#define SHOW_BUFFER
```

Change to this:

```
//#define SHOW_BUFFER
```

Also, be aware, that the CRSF telemetry messages emerging from the transmitter box are pretty slow. Nothing we can do about that unfortunately. We need to get a GPS message on a regular basis, otherwise tracking becomes sporadic.

closed 2025-07-22 by zs6buj