zs6buj/AntTracker #67: Compass flip not working

oldrootbeer, 2024-03-13:
Using a ttgo lora with a bn880 gps & mag.
If I  use flip or non flip alignment in config.h, the heading still rotates opposite to what it should. As if the flip function does not work.

If I physically flip the gps module (mag built in), all is good, but the gps antenna is facing down.

Am I missing something or is there a work around? As I really like the tracker.

v2.20.04

zs6buj, 2024-03-13:
Hmm. That feature was kindly added by one of the contributors. I'll take a look when I get the chance.

zs6buj, 2024-03-13:
Can you confirm which flip you chose:

`CW0_DEG_FLIP, CW90_DEG_FLIP, CW180_DEG_FLIP, CW270_DEG_FLIP`

zs6buj, 2024-03-13:
This is the code. Strangely, the treatment of the FLIP cases is the same as no flip. Line 1610 in the Utilities tab.

```
  float applyCompassAlignment(float hdg , uint8_t rotation) {
    float res = 0.0;
    
    switch (rotation) {
      default:
      case CW0_DEG:
      case ALIGN_DEFAULT:
      case CW180_DEG_FLIP:
          res = hdg;
          break;
      case CW90_DEG:
          res = hdg + 90;
          break;
      case CW180_DEG:
          res = hdg + 180;
          break;
      case CW270_DEG:
          res = hdg + 270;
          break;
      case CW0_DEG_FLIP:
          res = hdg + 180;
          break;
      case CW90_DEG_FLIP:
          res = hdg + 270;
          break;
      case CW270_DEG_FLIP:
          res = hdg + 90;
          break;
    }
    
    if (res > 360.0) {
      res = res - 360.0;
    }

    return  res;
  }
```

oldrootbeer, 2024-03-13:
Sorry. On the road at the moment. I was trying cw90flip. I then tried cw90 and both where identical.

The heading adjusts, just not the flip.

I was attempting to find the references last night but had to be up early today for s long day. I'll look tonight and tomorrow

oldrootbeer, 2024-03-13:
> This is the code. Strangely, the treatment of the FLIP cases is the same as no flip. Line 1610 in the Utilities tab.
>
> ```
>   float applyCompassAlignment(float hdg , uint8_t rotation) {
>     float res [...]

I have made these changes to my local file and I get good results with cw270flip.
I can test the rest over the next day or two when I get the time.

 switch (rotation) {
      default:
      case CW0_DEG:
      case ALIGN_DEFAULT:
      case CW180_DEG_FLIP:
          res = **(360-hdg)**;
          break;
      case CW90_DEG:
          res = hdg + 90;
          break;
      case CW180_DEG:
          res = hdg + 180;
          break;
      case CW270_DEG:
          res = hdg + 270;
          break;
      case CW0_DEG_FLIP:
          res = **(360-hdg)** + 180;
          break;
      case CW90_DEG_FLIP:
          res = **(360-hdg)** + 270;
          break;
      case CW270_DEG_FLIP:
          res = **(360-hdg)** + 90;
          break;
    }

oldrootbeer, 2024-03-14:
This is more correct I think

switch (rotation) {
      default:
      case CW0_DEG:
      case ALIGN_DEFAULT:
      case CW180_DEG_FLIP:
          res = **(360-hdg)+180**;
          break;
      case CW90_DEG:
          res = hdg + 90;
          break;
      case CW180_DEG:
          res = hdg + 180;
          break;
      case CW270_DEG:
          res = hdg + 270;
          break;
      case CW0_DEG_FLIP:
          res = **(360-hdg)**;
          break;
      case CW90_DEG_FLIP:
          res = **(360-hdg) + 90**;
          break;
      case CW270_DEG_FLIP:
          res = **(360-hdg) + 270**;
          break;
    }

zs6buj, 2024-03-14:
This is good. Thank you so far.

zs6buj, 2024-03-21:
Would you like to do a PR?

oldrootbeer, 2024-03-21:
I will do when I figure out how to. I've only done one in the past, but messed it up.

zs6buj, 2024-03-22:
Good job oldrootbeer.

closed 2024-03-22 by zs6buj