zs6buj/AntTracker #68: Feature Request - Accelerometer input for mag tilt compensation

oldrootbeer, 2024-03-14:
As a magnetometer works in 3d, it really needs to ignore the z axis for this project? I guess this can be done by using an accelerometer and maths.
I really can't see how a compass on the tracker is useful unless it's calibrated manually in the field using a compass every time.

I'll look into it, but I'm an (very) amateur programmer.

zs6buj, 2024-03-15:
One other thing. Remember the box compass simply determines the orientation of the box (and antenna at rest) relative to the world. The GPS co-ordinates are used to calculate where to point in the world.

oldrootbeer, 2024-03-15:
I understand your arguments against, as does my friend. I have adjusted the display to show the box heading, which can be mechanically tweeked in the field using a phone compass.
But i'll look into it anyway - for fun.

Thank you for your reply

zs6buj, 2024-03-15:
All good. Enjoy yourself, but beware, coding is addictive lol

closed 2024-03-21 by zs6buj