zs6buj/AntTracker #69: Altitude calculations not working in headingsource = 4 for 2.20.04

oldrootbeer, 2024-03-18:
When in headingsource = 4 (tracker gps & compass), cur.alt_ag is always zero, meaning the elevation servo will not kick in.

On a head scratching inspection, it appears that the variable finalHomeStored is set to true on headingsource 1, 2 and 3, but not 4.
This variable is then used in Mav.ino and Gps.ino as below...

Line 54 gps.ino
```cpp
       if (finalHomeStored) {
          cur.alt_ag = cur.alt - hom.alt;
        } else {
          cur.alt_ag = 0;
        }   
```

Line 412 Mavlink.ino
```cpp
if (finalHomeStored) {
            cur.alt_ag = cur.alt - hom.alt;
          } else {
            cur.alt_ag = 0;
          } 
```

As I'm a beginner, I have modified my code to read...

```cpp
 if ((finalHomeStored) || (headingsource == 4)) {
          cur.alt_ag = cur.alt - hom.alt;
        } else {
          cur.alt_ag = 0;
        }  
```

This seems to work (although needs a good test in the field). Is there is a better way of doing this?

zs6buj, 2024-03-21:
It's not that simple I'm afraid. headdngsource == 4  means Trackerbox_**GPS**_And_Compass.

Since we have a GPS on the box, "home" is assumed to be potentially dynamic, i.e. it can move location in 3D space, including the z axis, or, altitude.  The alt_ag, or altitude above ground now means the altitude of the UAV relative to the altitude of the box, as determined by the box GPS. Of course the box GPS needs unimpeded access to the sky.

zs6buj, 2024-03-21:
If you don't need dynamic home, the I recommend you change to

`#define HEADINGSOURCE  3     // 3=Trackerbox_Compass `

and ignore the box GPS.

I have my HP Elitebook with me, but the screen is rather small so it's not that easy to navigate around the code right now.. Remember that we don't start tracking until minDist > 4m You could perhaps try this debugging macro in config.h

#define DEBUG_AzEl

oldrootbeer, 2024-03-21:
I thought so. I also experience problems with the asl reporting from the gps for hom.alt. Inav reports agl in its telemetry.
But I've modified it for my needs so it gets the relative height.

Thank you

closed 2024-03-21 by oldrootbeer