zs6buj/AntTracker #74: Continuous Rotation Servo 360

HKR1987, 2024-05-22:
Hi,
it is possible to use Continuous Rotation Servo 360 as azimuth servo?
When i try it, it rotate too much

zs6buj, 2024-05-23:
You can not use a continuous rotation azimuth servo turntable unless it is accompanied by electronics and a rotation angle sensor. It has to be smart enough to turn to an angle and stop there. Google this.

closed 2024-05-27 by HKR1987

pki791, 2024-09-03:
@zs6buj this can be done using the mag, maybe it would be possible to implement this? i build my tracker as multiturn and have no way to get a fixed position without mag.

zs6buj, 2024-09-03:
If you switch to stepper motors it solves this problem. Or just 1 stepper to drive the azimuth platter. With steppers I can set an arbitrary zero point, or remember any position and return to it

pki791, 2024-09-05:
Stepper is not a solution for my tracker. For the use of mag and servo it would be:
- read mag via i2c (5883)
- calibration routine (30 seconds of turning in one side)
- simple routine to follow the sensor where about 1500 would mean stop, lower values rotation left, higher values rotation right.