zs6buj/AntTracker #75: esp32 dev module telemetry_in via serial

vshumilov, 2024-05-27:
Hi,
I couldn't get telemetry from ground station to esp32 with serial connection betwe.
Could you tell me please does serial telemetry_in working with esp32 or not?
And how I can get serial connection for telemetry with esp32?
Thanks a lot

vshumilov, 2024-06-06:
Hello, Eric.
Thanks a lot for you answer.
Yes? I use mavlink2 between anttracker and ground station.
I'm try to use next scheme:
```mermaid
flowchart TD
    esp32[esp32 anttracker]<-->|uart|Pi[Orange Pi with mavlink-router]<-->|wfb-ng| drone[Drone]
```
So, anttracker don't get mavlink2 from uart. I seee that Orange Pi send mavlink2 messages to uart tx with analyser.
How to configure esp32 uart telemetry?

zs6buj, 2024-06-06:
Clone latest version V 2.21.06. These are the essential settings in config.h

Build and flash. 

```
//=============================================================================================
//=====================   S E L E C T   E S P   B O A R D   V A R I A N T   ===================
//=============================================================================================
//#define ESP32_Variant     1    //  ESP32 Dev Module - there are several sub-variants that work
//#define ESP32_Variant     4    //  Heltec Wifi Kit 32 
#define ESP32_Variant     5    //  LILYGO® TTGO T-Display ESP32 1.14" ST7789 Colour LCD, IDE board = "ESP32 Dev Module"
//#define ESP32_Variant     6    // LILYGO® TTGO T2 ESP32 OLED Arduino IDE board = "ESP32 Dev Module"
//#define ESP32_Variant     7    // ESP32 Dev Module with ILI9341 2.8" colour TFT SPI 240x320 NOT TESTED _ DON'T USE YET

//=============================================================================================
//====================  I N P U T   M E D I U M    How does telemetPROTOCOLry enter the tracker?
//=============================================================================================
// Choose one only of these input channels 
#define MEDIUM_IN  1    // UART (Serial)       
//#define MEDIUM_IN  2    // WiFi - ESP only
//#define MEDIUM_IN  3    // Bluetooth (Serial) - ESP32 only

//=============================================================================================
//================================  T E L E M E T R Y   P R O T O C O L  ======================
//=============================================================================================
// Select only one telemetry PROTOCOL here
//#define PROTOCOL 0     // AUTO detect protocol
//#define PROTOCOL 1     // Mavlink 1
#define PROTOCOL 2     // Mavlink 2
//#define PROTOCOL 3     // FrSky S.Port
//#define PROTOCOL 4     // FrSky F.Port 1
//#define PROTOCOL 5     // FrSky F.Port 2
//#define PROTOCOL 6     // LTM
//#define PROTOCOL 7     // MSP
//#define PROTOCOL 8     // GPS NMEA
//#define PROTOCOL 9     // CRFS
```

Find the in_rxPin for your esp32 board, and connect rPi UART TX pin to ESP32 UART1 in_rxPin, in this case GPIO27. Also connect ground between the two, obviously.

Find this line in config.h

 `   int8_t in_rxPin =         27;       // uart1 for general serial in`

Also read the wiki.

vshumilov, 2024-06-06:
Eric, thanks a lot.
I'll try 2.21.6 version tomorrow.
It's not work in version 2.20.6 , right?

zs6buj, 2024-07-01:
I have posted V 2.21.07 |(under release_candidate) with fixes and some new features. Please could you check it when you get the chance.

zs6buj, 2024-07-01:
Also check out [https://github.com/zs6buj/CRSF_Utlilties](url) where you will find three utilities for CRSF to send telemetry from uart to UDP or BT, and read UDP. The utilities work with AntTrack.

zs6buj, 2024-07-29:
I guess I can close this issue. Reopen if you need to.

closed 2024-07-29 by zs6buj