zs6buj/AntTracker #76: Feature request. Servo slowdown with no delay

oldrootbeer, 2024-05-31:
I really like your project. It's teaching me tons about Arduino.
I came across this library, which I have integrated quite easily into my own version of this tracker. It slows the servos down, but allowing them to work independently. A must for flying behind the tracker to keep the signal. What do you think?

https://github.com/MicroBahner/MobaTools
I'd do a pull, but too heavily modded into your brilliant project.

zs6buj, 2024-06-01:
Hey Karl, thank you for this valuable feedback. I shall certainly look at this. I still want to add support for stepper motors too, but time is at a premium. I'll feedback to you after.

vshumilov, 2024-06-06:
> add support for stepper motors

It will be great improvement! As well stepper motor and feedback with potentiometer

zs6buj, 2024-06-08:
I already tried this, but Espressif upgrade to core 3.0.0 broke the MobaTools library. I [reported it](https://github.com/MicroBahner/MobaTools/issues/33) and dev Franz-Peter is looking at it. In the meantime, when I get time, I will work with the older ESP32 core.

zs6buj, 2024-08-07:
Ok, I've patched MobaTools to run on ESP32 core 3.0.3. It's in my github /libraries/esp32/core_3.0.3/ folder.

I have also pushed a PR to https://github.com/MicroBahner/MobaTools and maybe soon they will push an update that will arrive in the Arduino IDE library manager.

**You must upgrade to the esp32 core in the Arduino IDE**, or the changes below will not compile

AntTrack V2 21.12 in the Beta folder has been updated to work with MobaTools servo library, and includes support for changing the tracker servos speed. It's a great library and all kudos to dev Franz-Peter.

Adding support for steppers should now be relatively easy, but I want to test with the stepper and driver I have before I publish.

zs6buj, 2024-08-18:
v2 22.00 added support for stepper motors. It's a Beta, so feedback would be appreciated. I used Nem17 motors with TB6600 drivers, but just on the workbench. testMotors() is activated.

[image]

zs6buj, 2024-08-18:
When #defined STEPPERS, there is a phase after startup when the azimuth stepper adjust button is used to centre the az stepper mid-field in front of the operator

closed 2024-10-24 by zs6buj