zs6buj/AntTracker #77: T-Display ESP32 1.14" - Problems

samnikos7, 2024-06-04:
I have various problems with TTGO T-Display ESP32 1.14"

[image]

With v2.21.02 (esp boards v2.0.16 - v2.0.17)
- Unable to connect to my HM-10 bluetooth SLAVE giving its name in #define BT_MODE  1 (Master Mode).
- I get errors compiling the sketch when I try to #define BT_MODE  2 (Slave Mode)

[image]

With the latest version v2.21.04 (updated libraries + esp boards v3.0.0).
- The display is always OFF, but serial monitor is working in IDE.
- Unable to connect to my HM-10 bluetooth SLAVE giving its name in #define BT_MODE  1 (Master Mode).

zs6buj, 2024-06-05:
[samnikos7](https://github.com/samnikos7)

It turns out the new ESP32 core 3.0.0 breaks TFT_eSPI. So please go back to core 2.0.17 for the time being.

[image]

Also, please use tracker v2.21.05 on my git. I changed something in the uart code flow.

samnikos7, 2024-06-05:
Thank you a lot for your quick reply.
- I rolled back to esp32 core v2.0.17, I used tracker v2.21.05 and my TTGO T-Display ESP32 1.14" is working now.
- Bluetooth Module HM-10 is still unable to connect with the ESP. 
- Another Bluetooth Module HC-05 is connecting normally as SLAVE but I would like to use my internal Radiomaster TX16S RF-BM-S02 Bluetooth Module (CC2540 similar to HM-10) as Telemetry. I had no problems using with u360gts.   
https://www.instructables.com/Bluetooth-on-Radiomaster-TX16S/
- When I try to compile with "#define WIFI_PROTOCOL 2    // UDP"  I get errors.
[image]

Is it possible to use Bluetooth as Telemetry IN and Wifi AP as Telemetry MIRROR for other use (like Telemetry Viewer for Android) at the same time?

zs6buj, 2024-06-06:
Thank you again for the feedback. It is invaluable as I am rather time constrained here. 

Unfortunately the HM-10 uses  BLE, Bluetooth 4.0, and I have not yet upgraded the tracker code for BLE support. It's one of the things on the ToDo list. 

Could you please post a copy of your config.h

With the correct settings, [mav2pt](https://github.com/zs6buj/mav2pt/wiki) will switch BT (Classic) to WiFi UDP.

samnikos7, 2024-06-06:
I will try to compile it with multiple combinations (e.g BT as SLAVE, WIFI STA->AP, all protocols one by one, etc) and post the results later.
Here is my current [config.h.txt](https://github.com/user-attachments/files/15613203/config.h.txt) with many complication errors, many Functions from the other *ino files seem to be "not declared in this scope" in the main void loop.

zs6buj, 2024-06-06:
Hold fire for now :) I'm going through all Media, all protocols at present. I'll upload when I'm ready

zs6buj, 2024-06-06:
try v 2.21.06. I have to go fetch my granddaughter and take her to swimming lessons :)

samnikos7, 2024-06-06:
I've just tried v2.21.06 with no complication errors choosing "#define MEDIUM_IN  2    // WiFi - ESP only" choosing every PROTOCOL (1-9) but get bootloops like this.

[image]

When I choose "#define PROTOCOL 0     // AUTO detect protocol" with excact the same settings I get complication errors.

[image]

zs6buj, 2024-06-08:
Ok, thank you

zs6buj, 2024-07-01:
I have posted V 2.21.07 |(under release_candidate) with fixes and some new features. Please could you check it when you get the chance.

zs6buj, 2024-07-01:
Also check out [https://github.com/zs6buj/CRSF_Utlilties](url) where you will find three utilities for CRSF to send telemetry from uart to UDP or BT, and read UDP. The utilities work with AntTrack.

closed 2024-07-29 by zs6buj