zs6buj/AntTracker #78: Bluetooth ble support

sosko71, 2024-07-04:
Hello, I would like to know if there will ever be support for bluetooth ble. I have an rc Frsky Tandem on which I can only get telemetry via bluetooth ble.

zs6buj, 2024-07-05:
Hello, yes there have been a few requests so I will implement BLE.  I don't have any commercial BLE adapters to test with, so I shall use the ESP32s for testing.

I will need the SERVICE and CHARACTERISTIC UUIDs to talk to the adapter, or I will need to scan for nearby devices, services and characteristics, which adds complexity and code. But I guess this is unavoidable. It's on my list.

zs6buj, 2024-07-05:
Can I ask a favour of you? If you have an ESP32 board handy, flash this scan utility onto it, and post the results here. You need to have your Frsky Tandem on at the time.

[BLE_scan.zip](https://github.com/user-attachments/files/16107655/BLE_scan.zip)

sosko71, 2024-07-05:
Hello I am currently at work and will upload the code to the ESP32 when I get home and paste the results

sosko71, 2024-07-05:
OK I managed to get the data. It detected many devices, but my Frsky Tandem is highlighted in bold

Devices found: 13
Scan done!
Advertised Device: Name: , Address: 10:2b:41:35:5c:63, manufacturer data: 7500420401806e102b41355c63122b41355c6201110000000000, rssi: -87 
Advertised Device: Name: , Address: fc:03:9f:e9:17:db, manufacturer data: 75004204018060fc039fe917dbfe039fe917da01a00000000000, rssi: -87 
Advertised Device: Name: , Address: 16:34:42:2f:f1:b8, manufacturer data: 060001092002c7c79e7b1f10e9438cba8ac454544c4819ac842c33a891, rssi: -97 
**Advertised Device: Name: Hello, Address: d0:2e:ab:b1:d1:60, serviceUUID: 0000fff0-0000-1000-8000-00805f9b34fb, txPower: 0, rssi: -63** 
Advertised Device: Name: , Address: 0a:eb:01:e2:8a:53, manufacturer data: 060001092022b34e7db32afb426c987f8fc27708cc9604132af3be243a, rssi: -49 
Advertised Device: Name: , Address: 44:23:7c:8c:3a:76, manufacturer data: 8f030a10020200753a8c7c234481, rssi: -82 
Advertised Device: Name: , Address: 6d:48:1a:40:51:2c, manufacturer data: e000042bca929100, serviceUUID: 0000fe9f-0000-1000-8000-00805f9b34fb, rssi: -80, serviceData:  
Advertised Device: Name: , Address: 3b:bf:16:ba:62:2d, manufacturer data: 06000109202270688636bf23d1efc51f95a40babe0b5f946b7e402a469, rssi: -98 
Advertised Device: Name: , Address: 68:2a:2a:3d:fe:8e, manufacturer data: 4c001006101ad695de98, txPower: 12, rssi: -98 
Advertised Device: Name: , Address: 40:4c:ca:f4:da:72, rssi: -10, serviceData:  
Advertised Device: Name: , Address: 5b:f6:20:7e:2d:27, serviceUUID: 0000fe9f-0000-1000-8000-00805f9b34fb, rssi: -95, serviceData:  
Advertised Device: Name: , Address: fc:e3:93:7a:fb:9f, manufacturer data: 4c0012020003, rssi: -99 
Advertised Device: Name: , Address: 10:b6:1c:66:14:99, rssi: -96

zs6buj, 2024-07-06:
This is exactly what I need. Thank you.

zs6buj, 2024-07-11:
@sosko71:

I've added crsfUART2BLE_server and crsfReadBLE_client to [https://github.com/zs6buj/CRSF_Utlilties](url)

The server reads crsf telemetry and sends it to the client, which decodes the crsf telemetry and prints it.

There is a wrinkle, as BLE4 sends only displayable text and we want to send a structure containing varying data types that can contain 0x00. BLE4 regards 0x00 as end of message and truncates the full message. So I'm using a COBS (google it) routine to remove zeros and then add them back at the other end. It all works fine this way, but of course we do not know what data your Frsky Tandem sends.

There is also a Generic_BLE_Client.ino in the CRSF_Utilities repo. I'm going to suggest that you change the SERVICE_UUID, the CHARACTERISTIC_01_UUID and the bleServerName, build and flash, and try to connect to your Frsky Tandem. Let's see what it prints.

sosko71, 2024-07-11:
Hi 
Maybe a small problem because I don't use the crsf system. I only have frsky receivers with s-port telemetry

zs6buj, 2024-07-12:
Ah, obviously since you have a Frsky Tandem. ok I will have to do a small change for you

zs6buj, 2024-07-12:
You can still try the Generic_BLE_Client.

sosko71, 2024-07-12:
ok i will test it and post the results

sosko71, 2024-07-12:
Here are the data that I get after a successful connection between the esp32 and the rc command, which is connected to the flight controller and receives telemetry

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
Starting BLE Client ...
Scanning for device "Hello"
Device:Name: Hello, Address: d0:2e:ab:b1:d1:60, serviceUUID: 0000fff0-0000-1000-8000-00805f9b34fb, txPower: 0, rssi: -81
our device found, connecting to Hello...
connected to remote BLE server
our service found
our characteristics found
length:22  length:22  fully connected to BLE server Hello
Record:~��a
length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  Record:~
length:22  length:22  length:23  length:22  length:44  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  Record:~
length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  Record:~
length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  Record:~��\
length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  Record:~
length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  Record:~��\
length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  Record:~
length:22  length:22  length:44

zs6buj, 2024-07-13:
I've updated Generic_BLE_Client to print hex and also text in quotes. Let's try that again so we can see what the telemetry looks like in hex.

sosko71, 2024-07-13:
Here are the data that I get

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
Starting BLE Client ...
Scanning for device "Hello"
Device:Name: Hello, Address: d0:2e:ab:b1:d1:60, serviceUUID: 0000fff0-0000-1000-8000-00805f9b34fb, txPower: 0, rssi: -81
our device found, connecting to Hello...
connected to remote BLE server
our service found
our characteristics found
fully connected to BLE server Hello
length:22  Message:len:22:00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  ""
length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:45  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:44  length:22  length:22  length:22  length:22  length:22  length:23  length:22  Message:len:22:7E 00 10 30 52 D3 FF FF FF 5E 7E 7E 00 10 40 52 0C 00 00 00 50 7E 
  "~"
length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:44  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:44  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  Message:len:22:7E 00 10 20 07 63 00 00 00 54 7E 7E 98 10 04 F1 5C 00 00 00 75 7E 
  "~"
length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:44  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  Message:len:22:7E BA 10 10 F0 00 00 00 00 4A 7E 7E 00 10 00 04 2B 4E 00 00 3B 7E 
  "~��"
length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:45  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  Message:len:22:7E 98 10 04 F1 5C 00 00 00 21 7E 7E 00 10 10 02 65 04 00 00 42 7E 
  "~��\"
length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:44  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  Message:len:22:7E 98 10 04 F1 5C 00 00 00 21 7E 7E 00 10 00 06 33 00 00 00 04 7E 
  "~��\"
length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:44  length:23  length:44  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  Message:len:22:7E 00 10 20 07 63 00 00 00 54 7E 7E 00 10 00 01 F6 FF FF FF 4C 7E 
  "~"
length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  Message:len:22:7E 00 10 10 02 67 04 00 00 61 7E 7E 98 10 04 F1 5C 00 00 00 40 7E 
  "~"
length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:44  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:45  length:22  length:22  length:22  length:22  length:44  length:22  length:22  Message:len:22:7E 98 10 01 F1 61 00 00 00 19 7E 7E 00 10 00 06 33 00 00 00 3C 7E 
  "~��a"
length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:45  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:22  length:23  length:22  Message:len:22:7E 98 10 01 F1 61 00 00 00 19 7E 7E 00 10 40 52 0C 00 00 00 17 7E 
  "~��a"
[... cut ...]  length:22  length:22  length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  length:22  Message:len:22:7E 98 10 04 F1 5C 00 00 00 21 7E 7E 00 10 01 04 36 00 00 00 02 7E 
  "~��\"
length:22  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:44  length:22  length:22  length:23  length:22  length:22  length:22  length:22  length:22  Message:len:22:7E 00 10 00 04 3B 75 00 00 5A 7E 7E 98 10 01 F1 61 00 00 00 43 7E 
  "~"

zs6buj, 2024-07-13:
Ok, its regular S.Port. Not Fport. I can go ahead and add the BLE to Tracker for Frsky.

zs6buj, 2024-07-13:
Which Characteristic UUID did you use?

zs6buj, 2024-07-13:
You could try Generic_BLE_Client_Frsky.

sosko71, 2024-07-13:
> Which Characteristic UUID did you use?

CHARACTERISTIC_01_UUID "0000fff6-0000-1000-8000-00805f9b34fb"

zs6buj, 2024-07-14:
My bad. It's in CRSF_Utilities now.

zs6buj, 2024-07-14:
And which SERVICE_UUID did you use?

sosko71, 2024-07-14:
#define SERVICE_UUID "0000fff0-0000-1000-8000-00805f9b34fb"
#define CHARACTERISTIC_01_UUID "0000fff6-0000-1000-8000-00805f9b34fb"
#define bleServerName "Hello"

sosko71, 2024-07-14:
output for Generic_BLE_Client_FrSky

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
Starting BLE Client ...
Scanning for device "Hello"
Device:Name: , Address: 4b:93:d2:f7:98:fa, manufacturer data: 4c001005001820fbe3, txPower: 7, rssi: -74
Device:Name: , Address: 02:ba:19:a8:30:5d, manufacturer data: 060001092022f54335dbac44e653abe321d295966e7061cb64676fe84e, rssi: -33
Device:Name: Hello, Address: d0:2e:ab:b1:d1:60, serviceUUID: 0000fff0-0000-1000-8000-00805f9b34fb, txPower: 0, rssi: -48
our device found, connecting to Hello...
connected to remote BLE server
our service found
our characteristics found
fully connected to BLE server Hello
Message:len:22:01 00 00 00 34 8D 11 40 78 56 AD BA 18 00 00 00 34 12 BA AB 0C 00 
  ""
Message:len:22:7E 98 10 01 F1 61 00 00 00 19 7E 7E 00 10 00 06 60 00 00 00 6F 7E 
  "~��a"
Message:len:22:7E 98 10 01 F1 61 00 00 00 19 7E 7E 00 10 20 07 64 00 00 00 4A 7E 
  "~��a"
Message:len:22:7E 00 10 01 04 31 00 00 00 24 7E 7E 00 10 10 02 65 04 00 00 47 7E 
  "~"
Message:len:22:7E 00 10 40 08 62 02 00 00 38 7E 7E 00 10 30 52 F9 FF 00 00 1E 7E 
  "~��a"
Message:len:22:7E 00 10 00 07 01 00 00 00 16 7E 7E 00 10 10 07 01 00 00 00 10 7E 
  "~"
Message:len:22:7E 00 10 20 07 64 00 00 00 53 7E 7E 00 10 00 01 62 00 00 00 20 7E 
  "~"
Message:len:22:7E 00 10 10 04 00 00 00 00 04 7E 7E 00 10 01 04 31 00 00 00 20 7E 
  "~"
Message:len:22:7E 00 10 00 06 60 00 00 00 76 7E 7E 00 10 40 08 6C 02 00 00 40 7E 
  "~"
Message:len:22:7E 00 10 00 01 51 00 00 00 40 7E 7E 98 10 04 F1 5C 00 00 00 61 7E 
  "~"
Message:len:22:7E 00 10 10 09 77 01 00 00 7F 7E 7E 98 10 04 F1 5C 00 00 00 5E 7E 
  "~"
Message:len:22:7E 98 10 04 F1 5C 00 00 00 21 7E 7E 00 10 30 52 F9 FF FF FF 55 7E 
  "~��\"
Message:len:22:7E 00 10 30 52 F9 FF FF FF 74 7E 7E 98 10 01 F1 61 00 00 00 6D 7E 
  "~"
Message:len:22:7E 00 10 10 07 01 00 00 00 06 7E 7E 98 10 01 F1 61 00 00 00 1F 7E 
  "~"
Message:len:22:7E 00 10 00 01 59 00 00 00 48 7E 7E 98 10 01 F1 61 00 00 00 51 7E 
  "~"
Message:len:22:7E 00 10 01 04 31 00 00 00 24 7E 7E 00 10 10 02 65 04 00 00 47 7E 
  "~"
Message:len:22:7E 00 10 00 06 60 00 00 00 76 7E 7E 00 10 40 08 8A 02 00 00 A6 7E 
  "~"
Message:len:22:7E 98 10 04 F1 5C 00 00 00 21 7E 7E 00 10 20 07 64 00 00 00 72 7E 
  "~��\"
Message:len:22:7E 00 10 10 04 00 00 00 00 04 7E 7E 00 10 01 04 31 00 00 00 20 7E 
  "~"
Message:len:22:7E 98 10 04 F1 5C 00 00 00 21 7E 7E 00 10 10 02 66 04 00 00 41 7E 
  "~��\"
Message:len:22:7E 00 10 00 06 60 00 00 00 76 7E 7E 98 10 04 F1 5C 00 00 00 57 7E 
  "~"
Message:len:22:7E 00 10 30 52 F9 FF FF FF 74 7E 7E 00 10 40 52 05 00 00 00 73 7E 
  "~"
Message:len:22:7E 00 10 20 07 64 00 00 00 53 7E 7E 00 10 00 01 40 00 00 00 02 7E 
  "~"

zs6buj, 2024-07-14:
I have added AntTracker v2.21.08 in the alpha folder. This is VERY preliminary code to support BLE4 for FrSky. Please could you try it with your Frsky Tandem, and past the results when you get the chance. It should print some debugging information for me.

sosko71, 2024-07-15:
Compiled OK but It finds the rc command but does not connect via BLE

Starting AntTrack version:2.21.07
Display support activated: Landscape 
135x240  text_size=2  char_w_px=12  char_h_px=16  scr_h_ch=8  scr_w_ch=20
Target Board = 3  ESP32 / Variant is Expecting BLE In
Selected protocol is S.Port
headingsource = 2 FC Compass
Scanning for device "Hello"
Device:Name: , Address: 32:f9:af:23:bc:4c, manufacturer data: 06000109200227a01e3cbfb0d6f592866591ba929677908484184f638e, rssi: -94
Device:Name: , Address: 17:fd:ab:4c:2a:32, rssi: -92, serviceData: J#QED54g%1b�Q���|�Η��
Device:Name: , Address: 0a:11:68:52:48:4e, manufacturer data: 060001092022b9eabd3d5c22872817a408931c0ee0534d8c9d3e74eb56, rssi: -61
Device:Name: , Address: 67:e5:44:f6:58:a7, manufacturer data: e000042bca929100, serviceUUID: 0000fe9f-0000-1000-8000-00805f9b34fb, rssi: -64, serviceData: 
Device:Name: , Address: 77:63:cd:f1:4b:86, manufacturer data: 4c0010061f1a3448a5ad, txPower: 12, rssi: -93
Device:Name: seeedstudio-m-f4da70, Address: 40:4c:ca:f4:da:72, txPower: 9, rssi: -87, serviceData: 
Device:Name: , Address: 10:2b:41:35:5c:63, manufacturer data: 7500420401806e102b41355c63122b41355c6201110000000000, rssi: -70
Device:Name: Hello, Address: d0:2e:ab:b1:d1:60, serviceUUID: 0000fff0-0000-1000-8000-00805f9b34fb, txPower: 0, rssi: -69
our device found, connecting to Hello...

zs6buj, 2024-07-15:
Could you please review the service and characteristic UUIDs around line 102 in config.h

```
#if (MEDIUM_IN  == 4)
  //#define SERVICE_UUID "91bad492-b950-4226-aa2b-4ede9fa42f59"
  //#define CHARACTERISTIC_01_UUID "cba1d466-344c-4be3-ab3f-189f80dd7518"
  #define SERVICE_UUID "0000fff0-0000-1000-8000-00805f9b34fb"            // use me for Frsky Tandem
  #define CHARACTERISTIC_01_UUID "0000fff6-0000-1000-8000-00805f9b34fb"  // use me for Frsky Tandem
  #define bleServerName "Hello"  // use me for Frsky Tandem
#endif
```

sosko71, 2024-07-15:
Both are OK because if I try with Generic_BLE_Client_FrSky the connection works. The same values ​​are used in both

zs6buj, 2024-07-15:
oops, missed some logic. Could you try v2.21.08b when you can.

sosko71, 2024-07-15:
still not connecting

zs6buj, 2024-07-16:
Please try v2.21.08b. Tested ok here for connect now.

sosko71, 2024-07-16:
The connection is now ok and I get information about the RSSI signal strength on the display, but no information about the voltage and current. Here is part of the data recorded by the serial monitor:

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

Starting antTrack version:2.21.07
Display support activated: Landscape 
135x240  text_size=2  char_w_px=12  char_h_px=16  scr_h_ch=8  scr_w_ch=20
Target Board = 3  ESP32 / Variant is Expecting BLE In
Selected protocol is S.Port
headingsource = 2 FC Compass
Scanning for device "Hello"
our device found, connecting to Hello...
connected to remote BLE server
our service found
our characteristics found
fully connected to BLE server Hello
BLE msgBuf len:22  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 19 7E 
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 19 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 38 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 38 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E BA 10 10 F0 00 00 00 00 4A 7E 
BLE msgBuf len:22  7E BA 10 10 F0 00 00 00 00 4A 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 30 52 D6 FF FF FF 6A 7E 
BLE msgBuf len:22  7E 00 10 30 52 D6 FF FF FF 6A 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 19 7E 
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 19 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 07 08 00 00 00 3E 7E 
BLE msgBuf len:22  7E 00 10 00 07 08 00 00 00 3E 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 07 02 00 00 00 05 7E 
BLE msgBuf len:22  7E 00 10 10 07 02 00 00 00 05 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 01 6D 00 00 00 65 7E 
BLE msgBuf len:22  7E 00 10 00 01 6D 00 00 00 65 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 21 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 21 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 01 04 2E 00 00 00 3F 7E 
BLE msgBuf len:22  7E 00 10 01 04 2E 00 00 00 3F 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 02 63 04 00 00 65 7E 
BLE msgBuf len:22  7E 00 10 10 02 63 04 00 00 65 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 00 7E 
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 00 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 06 61 00 00 00 77 7E 
BLE msgBuf len:22  7E 00 10 00 06 61 00 00 00 77 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 07 07 00 00 00 31 7E 
BLE msgBuf len:22  7E 00 10 00 07 07 00 00 00 31 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 07 03 00 00 00 04 7E 
BLE msgBuf len:22  7E 00 10 10 07 03 00 00 00 04 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 20 07 63 00 00 00 50 7E 
BLE msgBuf len:44  7E 00 10 20 07 63 00 00 00 50 7E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 01 86 00 00 00 97 7E 
BLE msgBuf len:22  7E 00 10 00 01 86 00 00 00 97 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 1A 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 1A 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 02 63 04 00 00 65 7E 
BLE msgBuf len:22  7E 00 10 10 02 63 04 00 00 65 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 00 7E 
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 00 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 06 61 00 00 00 77 7E 
BLE msgBuf len:22  7E 00 10 00 06 61 00 00 00 77 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 30 52 D5 FF FF FF 79 7E 
BLE msgBuf len:22  7E 00 10 30 52 D5 FF FF FF 79 7E 00 00 00 00 00 00 00 00 00 00 00 
[... 6 lines cut ...]
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 3A 7E 
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 3A 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 04 3C 75 00 00 5D 7E 
BLE msgBuf len:22  7E 00 10 00 04 3C 75 00 00 5D 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 25 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 25 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 01 04 2E 00 00 00 3B 7E 
BLE msgBuf len:22  7E 00 10 01 04 2E 00 00 00 3B 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 02 0B 00 00 00 67 7E 
BLE msgBuf len:22  7E 00 10 00 02 0B 00 00 00 67 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 06 61 00 00 00 77 7E 
BLE msgBuf len:44  7E 00 10 00 06 61 00 00 00 77 7E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 40 52 11 00 00 00 4B 7E 
BLE msgBuf len:22  7E 00 10 40 52 11 00 00 00 4B 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 07 03 00 00 00 04 7E 
BLE msgBuf len:22  7E 00 10 10 07 03 00 00 00 04 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 BE 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 BE 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 04 1C 27 00 00 2F 7E 
BLE msgBuf len:22  7E 00 10 00 04 1C 27 00 00 2F 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 01 04 2E 00 00 00 22 7E 
BLE msgBuf len:22  7E 00 10 01 04 2E 00 00 00 22 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 02 63 04 00 00 65 7E 
BLE msgBuf len:22  7E 00 10 10 02 63 04 00 00 65 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 02 0B 00 00 00 67 7E

zs6buj, 2024-07-16:
This is what I want: To see the structure and content of the BLE  messages. Thank you.

sosko71, 2024-07-17:
Here is part of the data recorded by the serial monitor:
There are random Frsky time outs, I am attaching photos of the printouts on the ESP32 oled screen.
[image]
[image]
[image]

Starting antTrack version:2.21.07
Display support activated: Landscape 
135x240  text_size=2  char_w_px=12  char_h_px=16  scr_h_ch=8  scr_w_ch=20
Target Board = 3  ESP32 / Variant is Expecting BLE In
Selected protocol is S.Port
headingsource = 2 FC Compass
Scanning for device "Hello"
our device found, connecting to Hello...
connected to remote BLE server
our service found
our characteristics found
fully connected to BLE server Hello
BLE msgBuf len:22  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  00 00 00 00 00 00 00 00 00 00 00 
BLE msgBuf len:22  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 04 3C 75 00 00 44 7E 
BLE msgBuf len:22  7E 00 10 00 04 3C 75 00 00 44 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 21 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 21 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 02 5B 04 00 00 78 7E 
BLE msgBuf len:22  7E 00 10 10 02 5B 04 00 00 78 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 19 7E 
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 19 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 02 0B 00 00 00 38 7E 
BLE msgBuf len:22  7E 00 10 00 02 0B 00 00 00 38 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 06 60 00 00 00 76 7E 
BLE msgBuf len:22  7E 00 10 00 06 60 00 00 00 76 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 49 7E 
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 49 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 40 52 0F 00 00 00 0D 7E 
BLE msgBuf len:22  7E 00 10 40 52 0F 00 00 00 0D 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 07 03 00 00 00 15 7E 
BLE msgBuf len:22  7E 00 10 10 07 03 00 00 00 15 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 20 07 64 00 00 00 53 7E 
BLE msgBuf len:22  7E 00 10 20 07 64 00 00 00 53 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 A1 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 A1 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 04 1C 27 00 00 2F 7E 
BLE msgBuf len:22  7E 00 10 00 04 1C 27 00 00 2F 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 02 5B 04 00 00 78 7E 
BLE msgBuf len:22  7E 00 10 10 02 5B 04 00 00 78 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 09 73 01 00 00 7B 7E 
BLE msgBuf len:22  7E 00 10 10 09 73 01 00 00 7B 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 6F 7E 
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 6F 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 40 08 80 02 00 00 DA 7E 
BLE msgBuf len:22  7E 00 10 40 08 80 02 00 00 DA 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 07 06 00 00 00 1C 7E 
BLE msgBuf len:22  7E 00 10 00 07 06 00 00 00 1C 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 21 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 21 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 20 07 63 00 00 00 51 7E 
BLE msgBuf len:22  7E 00 10 20 07 63 00 00 00 51 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 01 5F 00 00 00 4E 7E 
BLE msgBuf len:22  7E 00 10 00 01 5F 00 00 00 4E 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 04 00 00 00 00 72 7E 
BLE msgBuf len:22  7E 00 10 10 04 00 00 00 00 72 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 21 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 21 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 09 73 01 00 00 27 7E 
BLE msgBuf len:22  7E 00 10 10 09 73 01 00 00 27 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 19 7E 
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 19 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 57 7E 
[... 11 lines cut ...]
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 04 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 04 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 02 0B 00 00 00 19 7E 
BLE msgBuf len:22  7E 00 10 00 02 0B 00 00 00 19 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 06 60 00 00 00 6F 7E 
BLE msgBuf len:22  7E 00 10 00 06 60 00 00 00 6F 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 40 08 80 02 00 00 DA 7E 
BLE msgBuf len:22  7E 00 10 40 08 80 02 00 00 DA 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 07 06 00 00 00 1C 7E 
BLE msgBuf len:22  7E 00 10 00 07 06 00 00 00 1C 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 07 02 00 00 00 05 7E 
BLE msgBuf len:22  7E 00 10 10 07 02 00 00 00 05 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 01 78 00 00 00 3D 7E 
BLE msgBuf len:22  7E 00 10 00 01 78 00 00 00 3D 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 21 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 21 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 1D 7E 
CRC Good C len:9  10 01 F1 61 00 00 00 1D 7E 
 FrSky F101: RSSI=97
FrSky read good!
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 1D 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 01 04 30 00 00 00 25 7E 
BLE msgBuf len:22  7E 00 10 01 04 30 00 00 00 25 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 09 73 01 00 00 5A 7E 
BLE msgBuf len:22  7E 00 10 10 09 73 01 00 00 5A 7E 00 00 00 00 00 00 00 00 00 00 00

zs6buj, 2024-07-21:
I've not been able to get back to tis yet. My apologies. I'll try get get some desk time today and see if I can simulate your data and test like that.

zs6buj, 2024-07-21:
I'm seeing some weird app IDs I added some new debug code in v2.21.10b

sosko71, 2024-07-21:
Here is part of the data recorded by the serial monitor:

Starting antTrack version:2.21.10
Display support activated: Landscape 
135x240  text_size=2  char_w_px=12  char_h_px=16  scr_h_ch=8  scr_w_ch=20
Target Board = 3  ESP32 / Variant is Expecting BLE In
Selected protocol is S.Port
headingsource = 2 FC Compass
Scanning for device "Hello"
our device found, connecting to Hello...
connected to remote BLE server
our service found
our characteristics found
fully connected to BLE server Hello
BLE msgBuf len:22  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  00 00 00 00 00 00 00 00 00 00 00 
BLE msgBuf len:22  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 04 2B 4E 00 00 68 7E 
BLE msgBuf len:22  7E 00 10 00 04 2B 4E 00 00 68 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 04 00 00 00 00 04 7E 
BLE msgBuf len:22  7E 00 10 10 04 00 00 00 00 04 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 02 4B 04 00 00 6E 7E 
BLE msgBuf len:22  7E 00 10 10 02 4B 04 00 00 6E 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 09 6E 01 00 00 66 7E 
BLE msgBuf len:22  7E 00 10 10 09 6E 01 00 00 66 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 06 59 00 00 00 56 7E 
BLE msgBuf len:22  7E 00 10 00 06 59 00 00 00 56 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 21 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 21 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 40 52 0E 00 00 00 1F 7E 
BLE msgBuf len:22  7E 00 10 40 52 0E 00 00 00 1F 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 07 11 00 00 00 06 7E 
BLE msgBuf len:22  7E 00 10 00 07 11 00 00 00 06 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 20 07 62 00 00 00 50 7E 
BLE msgBuf len:22  7E 00 10 20 07 62 00 00 00 50 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 01 37 00 00 00 26 7E 
BLE msgBuf len:22  7E 00 10 00 01 37 00 00 00 26 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 04 00 00 00 00 5E 7E 
BLE msgBuf len:22  7E 00 10 10 04 00 00 00 00 5E 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 21 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 21 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 09 6E 01 00 00 2B 7E 
BLE msgBuf len:22  7E 00 10 10 09 6E 01 00 00 2B 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 02 0B 00 00 00 19 7E 
BLE msgBuf len:22  7E 00 10 00 02 0B 00 00 00 19 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 67 7E 
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 67 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 30 52 9E FF FF FF 13 7E 
BLE msgBuf len:22  7E 00 10 30 52 9E FF FF FF 13 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 07 11 00 00 00 27 7E 
BLE msgBuf len:22  7E 00 10 00 07 11 00 00 00 27 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 07 02 00 00 00 05 7E 
BLE msgBuf len:22  7E 00 10 10 07 02 00 00 00 05 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 01 69 00 00 00 61 7E 
BLE msgBuf len:22  7E 00 10 00 01 69 00 00 00 61 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 21 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 21 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 1D 7E 
CRC Good C len:9  10 01 F1 61 00 00 00 1D 7E 
appID:F101
 FrSky F101: RSSI=97
FrSky read good!
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 1D 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 01 04 36 00 00 00 23 7E 
BLE msgBuf len:22  7E 00 10 01 04 36 00 00 00 23 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 09 6E 01 00 00 2B 7E 
BLE msgBuf len:22  7E 00 10 10 09 6E 01 00 00 2B 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 02 0B 00 00 00 19 7E 
BLE msgBuf len:22  7E 00 10 00 02 0B 00 00 00 19 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 67 7E 
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 67 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 30 52 9E FF FF FF 13 7E 
BLE msgBuf len:22  7E 00 10 30 52 9E FF FF FF 13 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 07 02 00 00 00 03 7E 
BLE msgBuf len:22  7E 00 10 10 07 02 00 00 00 03 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 20 07 63 00 00 00 54 7E 
BLE msgBuf len:22  7E 00 10 20 07 63 00 00 00 54 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 04 2B 4E 00 00 20 7E 
BLE msgBuf len:22  7E 00 10 00 04 2B 4E 00 00 20 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 04 00 00 00 00 04 7E 
BLE msgBuf len:22  7E 00 10 10 04 00 00 00 00 04 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 02 4B 04 00 00 54 7E 
BLE msgBuf len:22  7E 00 10 10 02 4B 04 00 00 54 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 09 6E 01 00 00 66 7E 
BLE msgBuf len:22  7E 00 10 10 09 6E 01 00 00 66 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 6E 7E 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 6E 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 40 08 AE 88 00 00 7E 7E 
BLE msgBuf len:22  7E 00 10 40 08 AE 88 00 00 7E 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 15 7E 
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 15 7E 00 00 00 00 00 00 00 00 00 00 00

zs6buj, 2024-07-21:
Thanks. progress. I added decode of CRC v2.21.10c

sosko71, 2024-07-21:
Here is part of the data recorded by the serial monitor:

BLE msgBuf len:22  7E 00 10 00 01 84 00 00 00 8C 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 21 7E 
CRC Bad! 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 21 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 61 00 00 00 1D 7E 
CRC Good C len:9  10 01 F1 61 00 00 00 1D 7E 
appID:F101
 FrSky F101: RSSI=97
FrSky read good!
BLE msgBuf len:22  7E 98 10 01 F1 61 00 00 00 1D 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 01 04 32 00 00 00 27 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 01 04 32 00 00 00 27 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 09 6A 01 00 00 43 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 10 09 6A 01 00 00 43 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 40 08 DC 05 00 00 81 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 40 08 DC 05 00 00 81 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 07 FD FF FF FF 85 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 00 07 FD FF FF FF 85 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 20 07 60 00 00 00 57 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 20 07 60 00 00 00 57 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 01 63 00 00 00 25 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 00 01 63 00 00 00 25 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 21 7E 
CRC Bad! 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 21 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 01 04 32 00 00 00 23 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 01 04 32 00 00 00 23 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 02 3F 04 00 00 39 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 10 02 3F 04 00 00 39 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 02 0B 00 00 00 00 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 00 02 0B 00 00 00 00 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 06 59 00 00 00 4F 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 00 06 59 00 00 00 4F 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 30 52 12 00 00 00 E1 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 30 52 12 00 00 00 E1 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 40 52 6D FF FF FF 90 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 40 52 6D FF FF FF 90 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 07 E7 FF FF FF 0A 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 10 07 E7 FF FF FF 0A 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 21 7E 
CRC Bad! 
........
BLE msgBuf len:22  7E 00 10 40 52 B8 FF FF FF 36 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 07 00 00 00 00 17 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 00 07 00 00 00 00 17 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 20 07 62 00 00 00 74 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 20 07 62 00 00 00 74 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 00 01 12 00 00 00 03 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 00 01 12 00 00 00 03 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 55 00 00 00 29 7E 
CRC Good C len:9  10 01 F1 55 00 00 00 29 7E 
appID:F101
[... 46 lines cut ...]
CRC Bad! 
BLE msgBuf len:22  7E 00 10 00 06 4B 00 00 00 5D 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 40 08 44 7A 00 00 3B 7E 
CRC Bad! 
......
BLE msgBuf len:22  7E 00 10 40 08 C4 86 00 00 0C 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 01 F1 48 00 00 00 30 7E 
CRC Bad! 
BLE msgBuf len:22  7E 98 10 01 F1 48 00 00 00 30 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 20 07 13 00 00 00 73 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 20 07 13 00 00 00 73 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 21 7E 
CRC Bad! 
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 21 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 98 10 04 F1 5C 00 00 00 1F 7E 
CRC Good C len:9  10 04 F1 5C 00 00 00 1F 7E 
appID:F104
FrSky read good!
BLE msgBuf len:22  7E 98 10 04 F1 5C 00 00 00 1F 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E 00 10 10 04 00 00 00 00 04 7E 
CRC Bad! 
BLE msgBuf len:22  7E 00 10 10 04 00 00 00 00 04 7E 00 00 00 00 00 00 00 00 00 00 00 
BLE Fr buf:len:11  7E BA 10 10 F0 03 00 00 00 E2 7E 
CRC Bad!

zs6buj, 2024-07-21:
Thank you. The CRC check is failing. A little more progress

zs6buj, 2024-07-23:
Ok, I've temporarily disabled the CRC check for FrSky BLE. Can we see what we get now? v2.21.10d

sosko71, 2024-07-23:
Ok now a little more telemetry data is visible. I am attaching two txt files, one contains the complete printout of the serial monitor, and the other contains details about the ID codes that I can see in the RC command, if that helps.
With GPS coordinates, only Latitude is visible on the display, and Longitude is not visible or read
[telemetry ID.txt](https://github.com/user-attachments/files/16350979/telemetry.ID.txt)
[Serial monitor.txt](https://github.com/user-attachments/files/16350980/Serial.monitor.txt)

zs6buj, 2024-07-23:
Looking at the telemetry in hex form, it looks like it is periodically corrupted. I looked briefly at the Tandem and Ethos, but know very little about it. Are there telemetry protocol option settings, and BLE settings. Remember BT/BLE use the 2.4GHz band, same as WiFi and all sorts of commercial gadgets. If you are in a room with computers and lots of RF noise this could be a problem. However the corruption seems to be periodic. A latitude of -824.6305542 in not possible, but a latitude of 45.5296783 is plausible. Your first test data, a week or so back, comprised two 11 byte messages combined into a 22 byte block, now I'm seeing zeros in the second block. The esp32 documentation mentions a maximum block size of 20 bytes for BLE4. Have you tried the BLE Scan app on your phone? I saw a 7D 5E pair in there, which could be byte stuffing. I'll look at that tomorrow.

sosko71, 2024-07-24:
> Are there telemetry protocol option settings, and BLE settings.

There are no special settings. Only use BLE for telemetry or trainer mode is set.

> If you are in a room with computers and lots of RF noise this could be a problem.

Yes, I have many different devices, so that possibility is likely

> A latitude of -824.6305542 in not possible, but a latitude of 45.5296783 is plausible.

Latitude 45.5296783 is correct, it should extract the longitude from the second number

> Your first test data, a week or so back, comprised two 11 byte messages combined into a 22 byte block, now I'm seeing zeros in the second block.

The first data was captured by Generic_BLE_Client_FrSky, while the data captured by Antena tracker only zeros appear in the second block

> Have you tried the BLE Scan app on your phone?

I use a BLE scanner with which I also obtained the necessary data for the connection. I also tried connecting the tablet to the RC command and using the telemetry program, which works OK

sosko71, 2024-07-25:
Something is not OK

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

Starting antTrack version:2.21.10
Display support activated: Landscape 
135x240  text_size=2  char_w_px=12  char_h_px=16  scr_h_ch=8  scr_w_ch=20
Target Board = 3  ESP32 / Variant is Expecting BLE In
Selected protocol is S.Port
headingsource = 2 FC Compass
Scanning for device "Hello"
our device found, connecting to Hello...
connected to remote BLE server
our service found
our characteristics found
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40090fbc  PS      : 0x00060b30  A0      : 0x800d2460  A1      : 0x3ffd1ae0  
A2      : 0x00000000  A3      : 0x3ffe548c  A4      : 0x00000016  A5      : 0x00000000  
A6      : 0x0110987e  A7      : 0x000049f1  A8      : 0x00000000  A9      : 0x3ffd1b60  
A10     : 0x3ffe540c  A11     : 0x00001800  A12     : 0x3ffe540c  A13     : 0x3ffd1cd8  
A14     : 0x60d1b1ab  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x40090fb8  LEND    : 0x40090fce  LCOUNT  : 0x00000000  

Backtrace: 0x40090fb9:0x3ffd1ae0 0x400d245d:0x3ffd1af0 0x401a5792:0x3ffd1b10 0x400dbc81:0x3ffd1b30 0x400dc589:0x3ffd1b80 0x400dae3d:0x3ffd1ba0 0x400db86d:0x3ffd1c50 0x400f2726:0x3ffd1ca0 0x400f2f15:0x3ffd1cc0 0x4011f4a5:0x3ffd1d10 0x401217ab:0x3ffd1d30

ELF file SHA256: 38e4be1bef3e2dc5

zs6buj, 2024-07-26:
This should be better. Updated v2.21.10e to pass object by reference. And tested with fake data.

sosko71, 2024-07-27:
Still something not ok, getting random reboot
[serial monitor 2.21.10e .txt](https://github.com/user-attachments/files/16400586/serial.monitor.2.21.10e.txt)

zs6buj, 2024-07-28:
I believe this is the clue:

`pD len:23`

There is an extra byte. We expect 2 x 11 bytes. I think FrSky are byte stuffing the data frame. This is not necessary when an entire frame (record) is transmitted and received at a time, like with BLE. It is used to handle serial framing, when a start/stop character is naturally imbedded in the hex frame data. I hope to look at this deeper later today.

Take a look [here](https://www.frsky-rc.com/wp-content/uploads/2017/07/Manual/FRSKY%20TELEMETRY%20PROTOCOL.PDF)

>  11 bytes frame started and ended with 0x7E, with byte stuff.
>  Byte stuffing method:
>  Output
> Byte in frame has value 0x7E is changed into 2 bytes: 0x7D 0x5E
> Byte in frame has value 0x7D is changed into 2 bytes: 0x7D 0x5D
>  Input:
>  When byte 0x7D is received, discard this byte, and the next byte is XORed with 0x20;

zs6buj, 2024-07-28:
Message:len:23:    7E 98 10 01 F1 61 00 00 00 19 7E     7E 00 10 40 08 8C **7D 5D** 00 00 B0 7E

zs6buj, 2024-07-28:
v2.21.11 Unstuffed FrSky byte stuff

sosko71, 2024-07-28:
No more random reset. Serial monitor results
[serial monitor 2.21.11.txt](https://github.com/user-attachments/files/16404196/serial.monitor.2.21.11.txt)

zs6buj, 2024-07-28:
We are getting there:

`Good flight computer GPS lock`

I will remove some of the debugging next for better visibility.

zs6buj, 2024-07-29:
v2.21.11b Removed debug verbosity. The results should be clearer now. You will need to push the "Home" button when requested, then move the 'craft more than 4m away. Servos should move accordingly. Please publish the monitor output.

sosko71, 2024-07-29:
The connection works, when it gets GPS lock I press "Home" and the display shows "tracking active". Then sometimes it works for a few seconds and sometimes it freezes immediately and there is no response. In one case, it worked for a few seconds and it was even possible to detect the movement of one servo, but then it froze again, there were also some resets. I am attaching two records from the serial monitor. 
[serial monitor 2.21.11b 2.txt](https://github.com/user-attachments/files/16414578/serial.monitor.2.21.11b.2.txt)
[serial monitor 2.21.11b 1.txt](https://github.com/user-attachments/files/16414579/serial.monitor.2.21.11b.1.txt)

zs6buj, 2024-07-30:
Thank you for the logs. I removed an unnecessary notify-ble-propery, which should fix that particular crash. The large amount of debug output slows down the loop, so I've further reduced debug verbosity. Now we should see more clearly what's happening with tracking.
v2.21.11c

sosko71, 2024-07-30:
I tried the new version but it still happens that everything stops - freezes. After turning it on, I get a connection with the RC command, it says that the GPS location is OK and I confirm the Home position, and then after a while the serial monitor no longer shows data reception, also there are no changes on the display, after a while the serial monitor says "connected to remote BLE server" and then nothing else happens. 4 logs are attached.
[serial monitor 2.21.11c 4.txt](https://github.com/user-attachments/files/16432448/serial.monitor.2.21.11c.4.txt)
[serial monitor 2.21.11c 3.txt](https://github.com/user-attachments/files/16432449/serial.monitor.2.21.11c.3.txt)
[serial monitor 2.21.11c 2.txt](https://github.com/user-attachments/files/16432450/serial.monitor.2.21.11c.2.txt)
[serial monitor 2.21.11c 1.txt](https://github.com/user-attachments/files/16432451/serial.monitor.2.21.11c.1.txt)

zs6buj, 2024-07-31:
So, each time, after a minute or two we again get "**connected to remote BLE server**", before it stops responding. However it connected previously in this session like this:

> 21:00:26.965 -> Starting BLE Client
> 21:00:26.965 -> Scanning for device "Hello" for 5 mins
> 21:00:27.105 -> our device found, connecting to Hello...
> 21:00:27.519 -> **connected to remote BLE server**
> 21:00:28.076 -> our service found
> 21:00:28.076 -> our characteristics found
> 21:00:28.122 -> fully connected to BLE server Hello

zs6buj, 2024-07-31:
v2.21.11d

zs6buj, 2024-08-02:
I notice it is very, very close to 30 seconds every single time, between the last log output and the spurious 

> connected to remote BLE server

sosko71, 2024-08-02:
yes, it always appears after a certain time, it would be interesting to see what causes the serial monitor to no longer receive data, and also no activity is visible in the operation - neither the display nor the activity of the LED anymore

zs6buj, 2024-08-02:
I'm going to ask you to compile and flag with debug option set in the IDE. Under tools, about halfway down. It will result in quite a lot of monitor output.

zs6buj, 2024-08-02:
Compile and flash

zs6buj, 2024-08-02:
[image]

[37 more replies]

closed 2024-10-24 by zs6buj