zs6buj/AntTracker #79: Alpha 21.21.09 wont compile

toofen, 2024-07-18:
Your latest alpha wont compile for me i get the following errors:

In file included from C:\Users\toofe\Downloads\AntTracker-2.21.09\AntTracker-2.21.09\source\alpha\v2.21.09\antTrack\antTrack.ino:123:
C:\Users\toofe\Downloads\AntTracker-2.21.09\AntTracker-2.21.09\source\alpha\v2.21.09\antTrack\config.h:403:20: error: operator '==' has no left operand
  403 | #if (ESP32_VARIANT == 4) // Heltec Wifi Kit 32 V3 (S3) (thanks to Marc Dornan)
      |                    ^~
C:\Users\toofe\Downloads\AntTracker-2.21.09\AntTracker-2.21.09\source\alpha\v2.21.09\antTrack\config.h:442:22: error: operator '==' has no left operand
  442 |   #if (ESP32_VARIANT == 5)          // LILYGO® TTGO T-Display ESP32 1.14" ST7789 Colour LCD, IDE board = "ESP32 Dev Module"
      |                      ^~
C:\Users\toofe\Downloads\AntTracker-2.21.09\AntTracker-2.21.09\source\alpha\v2.21.09\antTrack\config.h:478:22: error: operator '==' has no left operand
  478 |   #if (ESP32_VARIANT == 6)          // LILYGO® TTGO T2 ESP32 OLED Arduino IDE board = "ESP32 Dev Module"
      |                      ^~
C:\Users\toofe\Downloads\AntTracker-2.21.09\AntTracker-2.21.09\source\alpha\v2.21.09\antTrack\config.h:510:22: error: operator '==' has no left operand
  510 |   #if (ESP32_VARIANT == 7)          // ESP32 Dev Module with ILI9341 2.8" colour TFT SPI 240x320
      |                      ^~
C:\Users\toofe\Downloads\AntTracker-2.21.09\AntTracker-2.21.09\source\alpha\v2.21.09\antTrack\config.h:572:24: error: operator '==' has no left operand
  572 |     #if (ESP32_VARIANT == 6)

toofen, 2024-07-18:
Overall im having problems getting it to compile now, i cant even get the variants that i know has worked in the past to compile now. Something seems to have changed and i dont have any idea what, i dont rule out my own stupidity :)

zs6buj, 2024-07-18:
Hmm. Have you updated Espressif core to 3.x?  That is a big update, and many libraries need to be brought in line. I'm holding at v2.0.17 for now.

[image]

zs6buj, 2024-07-18:
thanks for the feedback. I found a line that was deleted in config.h. Should be ok now.

toofen, 2024-07-20:
Downgraded to boards to 2.0.17 still problems...

In file included from C:\Users\toofe\Downloads\AntTracker-2.21.10\AntTracker-2.21.10\source\alpha\v2.21.10\antTrack\config.h:553,
                 from C:\Users\toofe\Downloads\AntTracker-2.21.10\AntTracker-2.21.10\source\alpha\v2.21.10\antTrack\antTrack.ino:123:
c:\Users\toofe\Documents\Arduino\libraries\Wire\src/Wire.h:44:5: error: 'i2c_t' does not name a type; did you mean 'ino_t'?
     i2c_t * i2c;
     ^~~~~
     ino_t
c:\Users\toofe\Documents\Arduino\libraries\Wire\src/Wire.h:64:5: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?
     i2c_err_t last_error; // @stickBreaker from esp32-hal-i2c.h
     ^~~~~~~~~
     esp_err_t
c:\Users\toofe\Documents\Arduino\libraries\Wire\src/Wire.h:87:5: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?
     i2c_err_t writeTransmission(uint16_t address, uint8_t* buff, uint16_t size, bool sendStop=true);
     ^~~~~~~~~
     esp_err_t
c:\Users\toofe\Documents\Arduino\libraries\Wire\src/Wire.h:88:5: error: 'i2c_err_t' does not name a type; did you mean 'esp_err_t'?
     i2c_err_t readTransmission(uint16_t address, uint8_t* buff, uint16_t size, bool sendStop=true, uint32_t *readCount=NULL);
     ^~~~~~~~~
     esp_err_t
C:\Users\toofe\Downloads\AntTracker-2.21.10\AntTracker-2.21.10\source\alpha\v2.21.10\antTrack\utilities.ino: In function 'uint16_t getNextPacket()':
C:\Users\toofe\Downloads\AntTracker-2.21.10\AntTracker-2.21.10\source\alpha\v2.21.10\antTrack\utilities.ino:205:11: error: 'inlog' was not declared in this scope
       len=inlog.available();     //   wait for more data
           ^~~~~
C:\Users\toofe\Downloads\AntTracker-2.21.10\AntTracker-2.21.10\source\alpha\v2.21.10\antTrack\utilities.ino:205:11: note: suggested alternative: 'log'
       len=inlog.available();     //   wait for more data
           ^~~~~
           log
C:\Users\toofe\Downloads\AntTracker-2.21.10\AntTracker-2.21.10\source\alpha\v2.21.10\antTrack\utilities.ino: In function 'byte nextByte()':
C:\Users\toofe\Downloads\AntTracker-2.21.10\AntTracker-2.21.10\source\alpha\v2.21.10\antTrack\utilities.ino:234:11: error: 'inlog' was not declared in this scope
       b = inlog.read();
           ^~~~~
C:\Users\toofe\Downloads\AntTracker-2.21.10\AntTracker-2.21.10\source\alpha\v2.21.10\antTrack\utilities.ino:234:11: note: suggested alternative: 'log'
       b = inlog.read();
           ^~~~~
           log

zs6buj, 2024-07-21:
I had a look at this again, and it looks like a library issue, but I can't be sure. Could you please post your config.h here for me to look at.

toofen, 2024-07-22:
After a fresh install of Arduino IDE after uninstall and manually deleting the temp files there still are errors. I used the libraries from your source with ESPboard version 2.0.17. The wire.h error seems to be fixed for now but now there are some utilities.ino errors.

C:\Users\toofe\Downloads\AntTracker-2.21.10c\AntTracker-2.21.10c\source\alpha\v2.21.10c\antTrack\utilities.ino: In function 'uint16_t getNextPacket()':
C:\Users\toofe\Downloads\AntTracker-2.21.10c\AntTracker-2.21.10c\source\alpha\v2.21.10c\antTrack\utilities.ino:205:11: error: 'inlog' was not declared in this scope
       len=inlog.available();     //   wait for more data
           ^~~~~
C:\Users\toofe\Downloads\AntTracker-2.21.10c\AntTracker-2.21.10c\source\alpha\v2.21.10c\antTrack\utilities.ino:205:11: note: suggested alternative: 'log'
       len=inlog.available();     //   wait for more data
           ^~~~~
           log
C:\Users\toofe\Downloads\AntTracker-2.21.10c\AntTracker-2.21.10c\source\alpha\v2.21.10c\antTrack\utilities.ino: In function 'byte nextByte()':
C:\Users\toofe\Downloads\AntTracker-2.21.10c\AntTracker-2.21.10c\source\alpha\v2.21.10c\antTrack\utilities.ino:234:11: error: 'inlog' was not declared in this scope
       b = inlog.read();
           ^~~~~
C:\Users\toofe\Downloads\AntTracker-2.21.10c\AntTracker-2.21.10c\source\alpha\v2.21.10c\antTrack\utilities.ino:234:11: note: suggested alternative: 'log'
       b = inlog.read();
           ^~~~~
           log

Link below you can find my config.ino and utilities.ino 
https://github.com/toofen/antracker-tests

zs6buj, 2024-07-22:
It looks like your problem may be related to this library:

`      #include <TFT_eSPI.h>           // Remember to select the T_Display board in   in TFT_eSPI library (135 x 240) `

This is a general library, and must be configured for the exact display. antTrack works for several displays, and I suspect the User_Setup.h and User_Setup_Select.h are not currently setup for the T-Display. In the TFT_eSPI folder (in libraries), do this:

```
1  Delete User_Setup.h and User_Setup_Select.h
2 Copy User_Setup - ST7789.h to User_Setup.h
3 Copy User_Setup_Select - ST7789.h to User_Setup_Select.h
```

toofen, 2024-07-22:
You were right i had forgot to select the display in TFT_eSPI so i did that now. Still recieving the same errors in utilities.ino... I will go back a few versions and see if the errors presist. 

Thank you for your patience, there is a lots to remember and think about :)

zs6buj, 2024-07-22:
Strange. I compiled v2.21.10c(alpha) with your utilities.ino. I'll go back to 2.21.09 and check it out though.

Thanks for the feedback.

closed 2024-07-23 by zs6buj