zs6buj/AntTracker #84: V2.20.06 build failed

zanefond, 2024-09-08:
Hi Eric
I want to compile the latest stable of antenna tracker with bluepill,but can not be done! 
Documents/Arduino/libraries/SoftwareSerial/src/SoftwareSerial.cpp:45:30: fatal error: util/delay_basic.h: No such file or directory
 #include <util/delay_basic.h>
                              ^
compilation terminated.
exit status 1

Compilation error: exit status 1
my options in config.h is as following:
 #define STM32F1xx
#define Telemetry_In  0 
#define PROTOCOL 2     // Mavlink 2
#define BOX_GPS_BAUD 38400

zanefond, 2024-09-10:
/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/bin/arm-none-eabi-g++ -mcpu=cortex-m3 -DVECT_TAB_OFFSET=0x0 -DUSE_HAL_DRIVER -DUSE_FULL_LL_DRIVER -mthumb @/var/folders/cp/vswq7n5108q8dhf0120bnsgc0000gn/T/arduino_build_367807/sketch/build.opt -c -Os -DNDEBUG -w -std=gnu++17 -ffunction-sections -fdata-sections -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit "-I/Users/chenzhanfeng/Downloads/AntTracker-master 2/source/beta/v2.22.04/antTrack/aat/src/antTrack" -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/cores/arduino/avr -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/cores/arduino/stm32 -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/cores/arduino/../../libraries/SrcWrapper/inc -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/cores/arduino/../../libraries/SrcWrapper/inc/LL -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/system/Drivers/STM32F1xx_HAL_Driver/Inc -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/system/Drivers/STM32F1xx_HAL_Driver/Src -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/system/STM32F1xx -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/cores/arduino/../../libraries/USBDevice/inc -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/cores/arduino/../../libraries/VirtIO/inc -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/system/Middlewares/OpenAMP -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/system/Middlewares/OpenAMP/open-amp/lib/include -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/system/Middlewares/OpenAMP/libmetal/lib/include -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/system/Middlewares/OpenAMP/virtual_driver -w -x c++ -E -CC -DSTM32F1xx -DARDUINO=10819 -DARDUINO_BLUEPILL_F103CB -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"BLUEPILL_F103CB\"" "-DVARIANT_H=\"variant_PILL_F103Cx.h\"" -DSTM32F103xB -DHAL_UART_MODULE_ENABLED -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/tools/CMSIS/5.9.0/CMSIS/Core/Include/ -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/ -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/ -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/tools/CMSIS/5.9.0/CMSIS/DSP/Include -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/tools/CMSIS/5.9.0/CMSIS/DSP/PrivateInclude -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/cores/arduino -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/variants/STM32F1xx/F103C8T_F103CB(T-U) -I/Users/chenzhanfeng/Documents/Arduino/libraries/PixhawkArduinoMAVLink-1.0.1/src -I/Users/chenzhanfeng/Documents/Arduino/libraries/MobaTools/src -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/libraries/Wire/src -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/libraries/EEPROM/src -I/Users/chenzhanfeng/Documents/Arduino/libraries/ezButton/src -I/Users/chenzhanfeng/Documents/Arduino/libraries/TinyGPSPlus/src -I/Users/chenzhanfeng/Documents/Arduino/libraries/terseCRSF/src -I/Users/chenzhanfeng/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/libraries/SrcWrapper/src /var/folders/cp/vswq7n5108q8dhf0120bnsgc0000gn/T/arduino_build_367807/sketch/antTrack.ino.cpp -o /var/folders/cp/vswq7n5108q8dhf0120bnsgc0000gn/T/arduino_build_367807/preproc/ctags_target_for_gcc_minus_e.cpp
In file included from /Users/chenzhanfeng/Downloads/AntTracker-master 2/source/beta/v2.22.04/antTrack/aat/src/antTrack/config.h:191,
                 from /Users/chenzhanfeng/Downloads/AntTracker-master 2/source/beta/v2.22.04/antTrack/aat/src/antTrack/antTrack.ino:136:
/Users/chenzhanfeng/Documents/Arduino/libraries/MobaTools/src/MobaTools.h:175:6: error: #error Processor not supported
  175 |     #error Processor not supported
      |      ^~~~~
In file included from /Users/chenzhanfeng/Documents/Arduino/libraries/MobaTools/src/MobaTools.h:199:
/Users/chenzhanfeng/Documents/Arduino/libraries/MobaTools/src/utilities/MoToBase.h:29:6: error: #error "Processor not supported"
   29 |     #error "Processor not supported"
there is also a bug of mobatools,when compiling the bate version on bluepill

zs6buj, 2024-09-10:
[zanefond](https://github.com/zanefond)

My apologies for not replying sooner. I'm presently working on v2.22.04 up to support the stm32f103 again. The Servo and EEPROM libraries present different methods, so it is quite tedious, but I'm making progress.

zs6buj, 2024-09-10:
Can you please confirm which protocol you want to use

zanefond, 2024-09-10:
mavlink is my option ,and the first issue has been solved after updated the libaray of softwareserial,which you uploarded three days ago.

zs6buj, 2024-09-10:
ok, good. I'm making progress. I'll post here when it's working

zs6buj, 2024-09-13:
I would appreciate if you could test v2.22.06 (or later) for me when you get a chance. All the new features are included for STM32F103, including CRSF support and OLED display, but since there were many changes it needs thorough testing. I am unfortunately unable to do field testing at present. I guess you know the Arduino IDE settings:

[image]

zanefond, 2024-09-14:
yes! I have downloaded this version,and it compiled successfully. I will do the filed test in the coming holidays ! 
have a nice weekend ,thank you for developing such a wonderful project!

zanefond, 2024-10-11:
"Hi Eric! 
I've conducted a field test, but it's not working. When I set the home position, the telemetry stream is received by the AAT, but it just doesn't track. When I moved my plane around  5 meters away, and the situation LED work as the instruction. By the way, I'm confused about the compass type; it defines whether it's for ground use or air use."
my config is as below:
#define STM32_VARIANT     1      // STM32 Blue Pill F103C8 with 128k
#define MEDIUM_IN  1    // UART (Serial)   Teensy, STM32F103, ESP8266 and ESP32  
#define PROTOCOL 0     // AUTO detect protocol
#define HEADINGSOURCE  2     // 2=Flight Computer Compass
#define QMC5883L
#define Compass_Declination -4.55
  //#define REVERSEAZIMUTH          // my azimuth servo has a reversed action
  //#define REVERSEELEVATION
  #if defined STM32F1xx      // my STM32 based tracker has different servos
    const uint16_t minAzPWM = 525;   // right (because mine is reversed)
    const uint16_t maxAzPWM = 2475;  // left   
    const uint16_t minElPWM = 525;   // front 
    const uint16_t maxElPWM = 2475;  // back

the pic is shown the hardware setup
[image]
[image]

zs6buj, 2024-10-11:
[zanefond](https://github.com/zanefond)  "just doesn't track. When I moved my plane around 5 meters away"

A while back I added a minimum altitude (above field) before tracking starts, and this is proving to be a problem for many people while testing. in globals.h. 

```
    uint8_t minDist = 4;  // dist from home where tracking starts AND
    uint8_t minAltAg = 4; // alt above ground where tracking starts
```

I have today changed this in **v2.22.13** on GitHub. 

    uint8_t minDist = 4;  // dist from home where tracking starts AND
    uint8_t minAltAg = 0; // alt above ground where tracking starts

If you still have a problem could you please post the monitor log

zanefond, 2024-10-11:
thank you again for your great work! I will do the test as soon as possible! and give a feedback.

zanefond, 2024-10-16:
hello eric !
my aat basing on bluepill works,after changed the minalt parameter. but  the wrong direction when tacking.anyway it works now!i will do more test when time is available. thank you for you kindly instruction! may you have a nice day!

zanefond, 2024-10-19:
I've tried adjusting those two parameters multiple times but still can't track properly. It might be due to incorrect PWM values for my servo. I will keep trying, thank you for your help!

zs6buj, 2025-03-18:
Please reopen if you need to

closed 2025-03-18 by zs6buj